OpenXiangShan/NEMU #53: isa_mmu_check: Assertion `0' failed

chenglimin, 2022-02-16:
root@xiangshan-06:/root# ./squeezenet 000005.png
[src/isa/riscv64/system/mmu.c,209,isa_mmu_check] addr misaligned happened: vaddr:b6f23 len:8 type:1 pc:20000117d4
riscv64-nemu-interpreter: src/isa/riscv64/system/mmu.c:210: isa_mmu_check: Assertion `0' failed.
Aborted

Lemover, 2022-02-23:
这里是出现问题的地方：
https://github.com/OpenXiangShan/NEMU/blob/master/src/isa/riscv64/system/mmu.c#L207

粗暴的解决方式：
您可以直接将这行代码和下面的assert(0); 注释掉。

一些说明：
有些的硬件并不支持非对齐访问，因此nemu默认会直接assert，来提示代码有问题。不过nemu也可以通过报例外的方式，让例外处理函数“模拟”非对齐访存。
后面我们会添加一些配置选项，在编译之前配置这个功能。您现在可以先直接注释掉这里。

Lemover, 2022-02-23:
非对齐访存会触发例外，因此会大幅影响程序的性能。您可以留意一下这个例外的发生频率，如果比较频繁，可能需要判断例外发生在系统代码中，还是用户程序中。如果发生在用户程序中，就需要重新编译一下来避免了（比如在链接脚本中，指定偏移量之类的）。

chenguokai, 2022-02-25:
> busybox制作的文件系统里执行应用程序时不会出现这个问题的

NEMU内执行的软件bug除非您可以确定属于NEMU的行为出错，一般情况下不属于NEMU能够修复的问题。