OpenXiangShan/NEMU #55: simpoint profile issue

prodemen, 2022-03-23:
执行
./build/riscv64-nemu-interpreter $AM_HOME/apps/microbench/build/microbench-riscv64-noop.bin -D outputs -w microbench -C simpoint_profile --simpoint-profile --interval=1000000
时，没有输出bbv文件，只是打印了执行信息，请问可能的原因是什么？
[image]

matchchen, 2022-03-23:
加-b用batch mode运行

poemonsense, 2022-03-29:
用的是cpt-bk分支吗？checkpoint需要使用cpt-bk分支，参考https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/simpoint/

erdevonmatteo11, 2022-04-13:
> 用的是cpt-bk分支吗？checkpoint需要使用cpt-bk分支，参考https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/simpoint/

在cpt-bk分支无法使用make riscv64-xs_defconfig命令编译nemu。是不是只能使用主分支编译的nemu来执行Profiling步骤？

poemonsense, 2022-04-13:
cpt-bk 分支的编译方法是 make ISA=riscv64 XIANGSHAN=1

closed 2022-09-27 by shinezyy