OpenXiangShan/NEMU #87: NEMU cpt-bk分支实现checkpoint功能问题

YangQi1234, 2022-06-20:
[image]
[image]
目前在设置NEMU checkpoint模式时，用汇编改写a0寄存器时遇到这个错误
（At present, this error is encountered when rewriting the a0 register with assembly when setting the checkpoint mode of NEMU）
[image]
想了解一下这个错误当中寄存器范围怎么修改呢？谢谢！
(How to modify the register range in this error? thanks!)

Lemover, 2022-06-20:
两个方面的问题：
1. address map是如何设置的
2. 为什么会出问题，要怎么修改

1. address map 的设置和使用
nemu 访问物理地址，会先判断是 ddr 还是 外设：https://github.com/OpenXiangShan/NEMU/blob/master/src/memory/paddr.c#L96

Lemover, 2022-06-20:
两个方面的问题：
1. address map是如何设置的
2. 为什么会出问题，要怎么修改

1. address map 的设置和使用
nemu 访问物理地址，会先判断是 ddr 还是 外设：https://github.com/OpenXiangShan/NEMU/blob/master/src/memory/paddr.c#L96
https://github.com/OpenXiangShan/NEMU/blob/master/src/memory/paddr.c#L126
ddr和外设是不同的处理方式，您如果有兴趣可以继续追下去。
函数 `in_pmem` 负责这个判断，可以简单理解为 `0x80000000` 以上的为 ddr，以下的为 外设（具体判断更复杂一些）。

2. 为什么会出错，要怎么修改
这个报错信息在 外设地址检查函数，如果地址不在任何一个外设的地址范围，就会报错。
根据报错信息，地址为 `0xffffff00` ，应该在 ddr 里面，但是报错了。说明 ddr 的地址空间没有覆盖 `0xffffff00` ，因此被分配到 外设空间了。错误应该在于 nemu 的配置中，内存空间分配不够：https://github.com/OpenXiangShan/NEMU/blob/cpt-bk/include/memory/paddr.h#L10。
但是默认大小已经是 8GB 了，覆盖了出错的地址。所以您在使用哪个分支呢，是否修改了nemu的配置呢，能否提供更多的日志信息呢？

YangQi1234, 2022-06-20:
你好，我这个是cpt-bk分支，好像没有找到外设内存基地址，主内存基地已找到
[image]
我单步调试看到确实是进入到了外设内存地址中，
[image]
但是不明白外设内存分布是怎么样的，在哪可以查询到呢。我找了好久，没找到，谢谢。

YangQi1234, 2022-06-21:
我们是用cpt-bk分支
[image]
ddr设置的也是8G空间，我们觉得是命令用错了，我们想了解下如果只是单纯配置a0寄存器为101打开checkpoint模式该用什么命令呢？我们目前用./build/riscv64-nemu-interpreter  rv64uc-p-rvc.bin -D outputs -w build -C simpoint_profile --simpoint-profile --interval=1000000和./build/riscv64-nemu-interpreter -b  rv64uc-p-rvc.bin都不行（rv64uc-p-rvc.bin是修改寄存器的汇编代码可执行文件），谢谢！

Lemover, 2022-06-21:
打checkpoint分为三部，你的命令是符合第一步的格式的（但是少了-b参数，batch mode）。

cpt-bk分支，现在不支持在M态生成checkpoint。我们现在使用checkpoint的场景是，在linux之上，生成spec程序或其他程序的checkpoint，在启动spec程序之前，运行自定义程序，进入checkpoint模式。

需要提醒，手册中忘记说明的细节：
checkpoint模式下，nemu会加载一段程序 `gcpt.bin` ，到内存中，地址空间为：`(0x80000000,0xa0000)` 。gcpt.bin 负责 checkpoint 的恢复，这一段程序也会成为 checkpoint 文件的一部分。如果您自己想要生成自己的程序的checkpoint，需要把起始地址调到 `0x800a0000` 。

上面的信息可能不足以解决你的问题，您可以先尝试一下。

YangQi1234, 2022-06-23:
你好，我已经生成checkpoint了的.gz文件了，
[image]
想了解一下，下一步该如何调用verilator emu来对它进行仿真呢？谢谢

YangQi1234, 2022-06-23:
[image]
[image]
想再补充请教下，根据解析进行一次profiling是要先关闭时间中断，然后开启profiling,最后nemu trap吗，是要求汇编代码先将a0寄存器置为0x100，然后再置为0x101,最后再置为0x000？我的理解对吗？谢谢

Lemover, 2022-06-23:
> 

nemu-trap指令的使用方式是：
1. 置a0寄存器        ：li a0 0x100
2. nemu-trap指令  ：0x6b
两条指令完成一个功能。类似于函数调用+传参数。时间中断和profiling相当于是两次函数调用，需要执行上面的指令两次。

Lemover, 2022-06-23:
> 你好，我已经生成checkpoint了的.gz文件了， [image] 想了解一下，下一步该如何调用verilator emu来对它进行仿真呢？谢谢

对于香山的emu，checkpoint文件可以当成是普通的workload来用，香山会处理。
如果您需要在自己的emu上运行，可以参考香山的处理方式：https://github.com/OpenXiangShan/difftest/blob/master/src/test/csrc/common/ram.cpp#L153

YangQi1234, 2022-06-23:
[image]
你好，我们已经按照方法实现了checkpoint，但是最后生成这么多切片，每一个切片的大小是22M，但是我们并行跑第一个和第二个切片的时候发现，好像执行的代码是一样的
[image]
我一直以为是切完片后把程序分段了，然后再并行执行这些分段的代码计时，我想问一下这些分完片的内容都相同吗？我只用执行其中一个分片就行了？

Lemover, 2022-06-23:
checkpoint 是对系统的执行状态进行保存，所以静态代码是一样的，但是处于不同的执行阶段。
并行执行所有的checkpoint（注意需要限制每个片段的执行时间，即22M），“相当于是”执行了整个程序。（注意需要限制每个片段的执行时间，即22M）。对于某个切片，如果不限制执行时间（指令数），它会一直执行下去。

您可以跑一下其他的checkpoint，看看执行阶段是不是一样的，比如最后一个片段。

YangQi1234, 2022-06-23:
1、[image]
你好，我分别执行了第三个和最后一个checkpoint的代码，打印结果如下，我还是不太明白，他们为什么又是会出现相同的打印，按道理说它们分别属于不同的执行阶段，不应该会有相同的打印。
[image]
按照打checkpoint原理来说，每个点都是属于不同的执行阶段，为什么会有相关的打印执行，就算是有之前的打印缓存被保留下来了为什么没有，第一个切片的hello world打印？我这边不太明白。
2、那我该如何通过执行这些切片来最后还原我的spec2006的分数呢？通过计算每条指令的执行时间然后最后进行估算总的spec2006的执行时间吗？你们有具体的估算spec2006的方法吗？谢谢！

Lemover, 2022-06-23:
可以检查一下nemu跑worload时，行为是否是正确的。
另外还有生成checkpoint时的log，是否是正常的。

YangQi1234, 2022-06-24:
（1）[image]
想请教下，如果我的“hello world!"打印在_200000_.gz这个切片中，我在执行其它切片gz文件的时候是不是应该不会有这个打印信息，这个打印信息只存在_200000_.gz切片信息中，是吗？这是我checkpoint的打印信息log。
[香山切片log.txt](https://github.com/OpenXiangShan/NEMU/files/8973561/log.txt)
（2）
[image]
你所指的限制每个片段执行时间是指--checkpoint-interval这个参数吗？还是有其它命令指定，谢谢。

Lemover, 2022-06-24:
1. 是的。
2. 每个切片对应于workload的一段执行流程，但其实际上是系统状态的快照，在执行时需要手动限制执行周期。在香山中，通过 `-I` 参数来指定执行指令书，如 `-I 22M` 。具体时间长度，由打checkpoint时的设计决定，也就是 --checkpoint-interval。

突然想到一个细节没有写明：每个checkpoint文件的时间点，并不是严格的对应于checkpoint的时间点，而是会提前一个周期（interval），目的是为了预热硬件中的cache、bpu等内容。

YangQi1234, 2022-06-24:
请教一下你们之前测试香山内核spec2006的记录在哪里可以找到呢？谢谢

Lemover, 2022-06-29:
我们测试 SPEC06，一方面是通过 NEMU 生成 checkpoint，运行 checkpoint，并和 NEMU 做 diff-test 对比。另一方面，在FPGA上运行完整的 SPEC06 程序。

Lemover, 2022-06-30:
根据反馈，更新了香山文档中关于 Simpoint 的介绍：https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/simpoint/

YangQi1234, 2022-06-30:
想问一下，为什么我最后生成的切片，每一个切片大小比之前没有经过切片的大小还要大呢？这个正常吗？按道理说，切片是原目标文件的采样部分，为什么切片后单个切片还比原目标文件大？搞不明白，最后单个切片在emu上跑的时候时间也是很长，我用的也是你们的香山的emu，想请教下，你们跑spec2006的429测试用例单个切片时间一般是跑多久呢？谢谢。

closed 2022-07-04 by Lemover

bersarker, 2022-07-14:
> Checkpoint 文件是对系统状态的保存，包括体系结构状态和内存数据等等，文件大小大于程序本身的大小是正常的。 单个 checkpoint 在emu上的执行时间和 1）emu的仿真速度 2）checkpoint 的区间大小有关。 如：一个 checkpoint 的区间大小为 20，000，000条指令，测试程序的IPC为 2，emu仿真速度为 2000/s。最终需要仿真 20000000 * 2 / 2 / 2000 = 10000 s = 2.7h。 对于香山这么大规模的处理器核，仿真速度跑到2000/s也是理想的，不同程序的不同片段的IPC也有差别。我们跑 Checkpoint 的时间一般接近这个时间。
1、[image]
乘以2是指的预热的20000000条指令吗？
2、你们是怎么从每个切片的指令周期数估算整个workload总的指令周期数(cycles)呢？还是不太明白，毕竟每个切片才20000000条指令，是每个切片的指令周期数(cycles)的加权和，再乘以有多少个interval吗？毕竟一个切片是代表一类interval区间的数据吧，你把所有切片的所执行的指令周期数(cycles)加权和加到一起也不能一个workload全部的执行指令周期数(cycles)吧?

Lemover, 2022-07-14:
> Checkpoint 文件是对系统状态的保存，包括体系结构状态和内存数据等等，文件大小大于程序本身的大小是正常的。 单个 checkpoint 在emu上的执行时间和 1）emu的仿真速度 2）checkpoint 的区间大小有关。 如：一个 checkpoint 的区间大小为 20，000，000条指令，测试程序的IPC为 2，emu仿真速度为 2000/s。最终需要仿真 20000000 [...]

您可以参考香山手册：https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/simpoint/

YangQi1234, 2022-07-15:
是这样的我们目前执行emu的运行指令是。/emu -i  XXX.gz -I 10000000 -W 9900000 。当中-W我们看了下是预热的指令数，为什么我们执行完了后最后统计的指令数和指令周期数依然将预热部分算进去了，我们查看了源码，emu最后运行时应该是先执行预热部分，预热执行完成后再进行相关寄存器的重置，好像最后结果看相关寄存器没有重置成功？我们最后的统计结果如图，谢谢。
[image]
[image]

YangQi1234, 2022-07-15:
是这样的我们目前执行emu的运行指令是。/emu -i XXX.gz -I 19900000 -W 9900000 。当中-W我们看了下是预热的指令数，为什么我们执行完了后最后统计的指令数和指令周期数依然将预热部分算进去了，我们查看了源码，emu最后运行时应该是先执行预热部分，预热执行完成后再进行相关寄存器的重置，好像最后结果看相关寄存器没有重置成功？我们最后的统计结果如图，谢谢。
[image]
[image]