OpenXiangShan/NEMU #90: NEMU支持PMP

runninglinuxkernel, 2022-06-25:
目前NEMU不支持PMP，运行支持PMP的程序，NEMU会crash。

```
Welcome to riscv64-NEMU!
For help, type "help"
	                                            ___   ___
    //   ) )                    //   ) )  //   ) )    / /
   //___/ /   ___       __     ((        //___/ /    / /
  / __  (   //___) ) //   ) )    \      / __  (     / /
 //    ) ) //       //   / /       ) ) //    ) )   / /
//____/ / ((____   //   / / ((___ / / //____/ / __/ /___
sbi_set_pmp: start: 0x0 order 64 prot 0x7
sbi_set_pmp: pmpaddr: 0xffffffffffffffff  pmpcfg 0x1f, cfs_csr 0x3a0 addr_csr 0x3b0
sbi_trap_error: trap handler failed (error -5)
mcause: 0000000000000002  mtval: 0000000000000000 
mepc: 0000000080000cdc mstatus : 0000000a00001800
 gp : 0000000000000000 tp : 0000000080000208 t0 : 0000000000001000
 t1 : 0000000000000000 t2 : 0000000000000000 t3 : 0000000080003f60
 s1 : 0000000000000000 a0 : 00000000000003b0 a1 : ffffffffffffffff
 a2 : ffffffffffffffff a3 : 0000000000000010 a4 : ffffffffffffe9c8
 a5 : ffffffffffffffff a6 : 0000000000000000 a7 : 0000000000000000
 s2 : 0000000000000000 s3 : 0000000000000000 s4 : 0000000000000000
 s5 : 0000000000000000 s6 : 0000000000000000 s7 : 0000000000000000
 s8 : 0000000000000000 s9 : 0000000000000000 s10: 0000000000000000
 s11: 0000000000000000 t3 : 0000000000000000 t4: 0000000000000000
 t5 : 0000000000000000 t6 : 0000000000000000 sp: 0000000080003ef0
 ra: 0000000080000218
sbi panic
```

Lemover, 2022-06-25:
nemu 对pmp的支持还不完全，你可以提供一下操作流程么？我们复现一下。主要是nemu和workload的编译流程和选项，谢谢🙏

runninglinuxkernel, 2022-06-25:
在BenOS实验里，主要设置小于8个pmpcfg range。
例如：

pmp0cfg：0x0000 0000 – 0xffff fffff ffff ffff ，属性为可读/可写/可执行
pmp1cfg：0x8000 0000 – 0x8004 0000，属性为可读/可写/可执行

+       /*
+        * 配置PMP
+        * 所有地址空间都可以访问
+        */
+       sbi_set_pmp(0, 0, -1UL, PMP_RWX);
+       sbi_set_pmp(1, 0x80000000, 0x40000, PMP_RWX);

NEMU里可以先不需要支持64个pmpcfg range，可以先支持8个。

sashimi-yzh, 2022-06-25:
在menuconfig里面选中下面选项即可
```
ISA-dependent Options for riscv64
[*] Enable PMP CSR Read & Write
[*]   Enable PMP Check
```

closed 2022-06-25 by sashimi-yzh

runninglinuxkernel, 2022-06-25:
good