OpenXiangShan/NEMU #94: NEMU cpt-bk分支运行中，参数选择的问题

bersarker, 2022-07-04:
按照 香山官方文档中，cpt-bk分支运行方法 ，执行了其中的profiling、cluster、checkpointing三条指令。
分别能够得到 simpoint_bbv、simpoints0 & weights0、xxxxx.gz切片文件。

当前问题是：
1.  .gz切片文件过大，导致使用emu执行时间太久，请问一下如何降低切片文件的大小？

2.运行cpt-bk分支时，如何选择其中参数？比如指令间隔大小应该怎样选择，以100,000还是1,000,000为区间比较好？或者是依照程序总指令数，按照一定比例划分指令区间？还有maxK的大小应该怎么取？例如我在maxK的值分别取30和100，得到了分类（切片）数量分别为23与29，应该如何选择？

以下为我执行的过程中具体切片数据大小：
我在运行的程序是SPEC INT 2006 的 429.mcf，三次不同指令间隔的运行结果如下：
指令间隔 100,000:  bbv文件：140.33MB  切片数量：21    .gz切片：大多数是280~330MB,2个20+MB，剩余170~220MB
指令间隔 1,000,000:  bbv文件：51.90MB  切片数量：26    .gz切片：大多数是280~330MB,1个29.57MB,
指令间隔 10,000,000:  bbv文件：7.1MB  切片数量：14    .gz切片：大多数是280MB以上
*其中.gz切片中295.*MB的文件最多

lico614257, 2022-07-04:
您好，我最近在做和你同样的事情，但是在nemu起linux上运行spec2006的过程中，十分耗时，想请问一下您的SPEC INT 2006 的 429.mcf这个case在nemu上第一步profile过程中运行了多长时间？

bersarker, 2022-07-04:
> 

可能会需要12-18小时吧。我是用脚本连续运行3条指令，确认过cluster过程很短（应该是2小时以内），三条指令全程大概24-36小时。请问您的切片的指令间隔设定的多少啊？我也想参考一下

shinezyy, 2022-07-04:
1.gz切片文件过大，导致使用emu执行时间太久，请问一下如何降低切片文件的大小？

我也没办法

2.运行cpt-bk分支时，如何选择其中参数？比如指令间隔大小应该怎样选择，以100,000还是1,000,000为区间比较好？或者是依照程序总指令数，按照一定比例划分指令区间？还有maxK的大小应该怎么取？例如我在maxK的值分别取30和100，得到了分类（切片）数量分别为23与29，应该如何选择？

建议读SimPoint的paper和文档，他们有非常详尽的解释。这不是NEMU或者香山相关的topic，我们只是借用SimPoint这个工具而已。

3.还有如下的三个阶段的指令中，除了 maxK和指令间隔，还有什么可以调节的参数？

同上

4.想问一下香山执行checkpoint执行SPEC INT 2006 的 429.mcf，大概多长时间？

100M指令最长可能需要跑2周

YangQi1234, 2022-07-04:
> 我运行的case是471，第一步profile过程中运行了7个小时，这个和host主机可能有主要的关系。我的切片的指令间隔设定的10000000。
老哥这是我的邮箱544915662@qq.com，连给我发个联系方式吗？我们也在研究这个东西，一起进步啊，哈哈哈。

bersarker, 2022-07-04:
> > 
> 
> 关于耗时较长的问题，可以尝试用tracing分支，tracing分支带性能优化

想问一下，您的切片大小是多少的？

> 1.gz切片文件过大，导致使用emu执行时间太久，请问一下如何降低切片文件的大小？
>
> 我也没办法 [...]

想问一下，您最后生成的切片大小每个大概多大呢？

shinezyy, 2022-07-04:
> 想问一下，您的切片大小是多少的？
>
>
> 想问一下，您最后生成的切片大小每个大概多大呢？

从几十M到几百M不等。

YangQi1234, 2022-07-05:
> > 
> 
> 关于耗时较长的问题，可以尝试用tracing分支，tracing分支带性能优化

[image]
老哥用你tracing的分支显示内存调用不足啊。需要改哪里呢？谢谢

shinezyy, 2022-07-05:
> > > 
> > 
> > 
> > 关于耗时较长的问题，可以尝试用tracing分支，tracing分支带性能优化
> 
> [image] 老哥用你tracing的分支显示内存调用不足啊。需要改哪里呢？谢谢

跑了多久出现的？可能是有bug。

bersarker, 2022-07-08:
> 从几十M到几百M不等。

您好，还想问一下，您的spec cpu 2006 所有测试用例用checkpoint跑完需要多久？同时用了多少台设备？

shinezyy, 2022-09-27:
> 您好，还想问一下，您的spec cpu 2006 所有测试用例用checkpoint跑完需要多久？同时用了多少台设备？

这取决于服务器的配置、微结构。这个问题在不同版本的xiangshan里也是不同的答案，我的建议是您先跑100k指令，算算仿真速度 (cycle per second)，然后再根据总指令数做一下除法来算运行时间。

closed 2024-01-25 by shinezyy