OpenXiangShan/NEMU #95: SPEC CPU 2006 分数 切片跑分 估值 问题

KeithPiTsui, 2022-07-19:
[image]
我们这边用这环境切片跑了429测试用例。

IntervalSize = 1M，共有 283226 个 Interval (在 Profiling时，BBV的行数)，即429用例约共执行了 283226 * 1M = 283,226,000,000 (+/-) 1M 条指令。

然后，我用以下公式去推算你们结果，发现IPC只有0.21左右。请指教下，是在计算的过程中，哪里出现了问题吗？

(给定：429测试用例得分为 标准用时除以实际用时，其中如上图，用例得分为14.52，标准用时为 9120s，emu模拟CPU的主频为 2.0GHz，总执行指令数为 283,226,000,000)

429测试用例得分 =  标准用时 / 实际用时        (求得 实际用时 为 9120 / 14.52 = 628.0991735537191)
实际用时 = 总执行时钟数 / CPU主频               (求得 总执行时钟数 为 628.0991735537191 * 2G = 1348832704528.9258 )
总执行时钟数 = 总执行指令数 / 加权平均IPC  (求得 加权平均IPC  为 283,226,000,000 / 1348832704528.9258 = 0.21 )

因此，把给定的参数代入，得出的 IPC 约为 0.21。

shinezyy, 2022-07-25:
可能没有哪里出了问题，429就是这么慢

KeithPiTsui, 2022-07-25:
所以得分的计算方式是对的，就是这emu跑429慢。我这理解对吗？

shinezyy, 2022-07-25:
> 429测试用例得分 = 标准用时 / 实际用时 (求得 实际用时 为 9120 / 14.52 = 628.0991735537191) 
> 实际用时 = 总执行时钟数 / CPU主频 (求得 总执行时钟数 为 628.0991735537191 * 2G = 1348832704528.9258 ) 
上面两步没有问题

> 总执行时钟数 = 总执行指令数 / 加权平均IPC (求得 加权平均IPC 为 283,226,000,000 / 1348832704528.9258 = 0.21 )
这一步有一点小细节需要注意一下，SimPoint的权重是以指令计算的，所以应该算加权平均CPI然后取倒数得到IPC。

shinezyy, 2022-07-25:
或者这么算：
总执行时钟数 = Sigma(每个片段所代表的cluster的时钟周期数)
一个片段所代表的cluster的时钟周期数 = 这个片段所代表的cluster的指令数 / 这个片段的CPI
这个片段所代表的cluster的指令数 = 总指令数 * 权重

这个计算方法化简之后和上面说的算加权平均CPI的算法是一样的。

shinezyy, 2022-07-28:
因为逻辑变动，包括修时序、改feature，可能会导致性能变化。
跟服务器关系不大。

KeithPiTsui, 2022-07-28:
这样的差异是属于合理的误差吗？还有我们这边使用的gcc版本可能跟你们跑分的gcc版本不一样，这也会导致结果差异吗？

shinezyy, 2022-07-28:
> 这样的差异是属于合理的误差吗？还有我们这边使用的gcc版本可能跟你们跑分的gcc版本不一样，这也会导致结果差异吗？

- gcc版本会导致结果差异
- 因为你没有给我具体的差异情况（子项、数值），所以我无法判断是否合理……

closed 2024-01-25 by shinezyy

qqli579, 2024-03-05:
> 或者这么算： 总执行时钟数 = Sigma(每个片段所代表的cluster的时钟周期数) 一个片段所代表的cluster的时钟周期数 = 这个片段所代表的cluster的指令数 / 这个片段的CPI 这个片段所代表的cluster的指令数 = 总指令数 * 权重
> 
> 这个计算方法化简之后和上面说的算加权平均CPI的算法是一样的。
这里的总指令数是NEMU采集simpoint的时候log里打印出来total guest instructions吗？
```
[src/cpu/cpu-exec.c:95,monitor_statistic] total guest instructions = 254,652,097,553
```
还有你们给的specCPU2006的分数，在采集simpoint的时候用的--cpt-interval是多少？

shinezyy, 2024-03-05:
> ```
> [src/cpu/cpu-exec.c:95,monitor_statistic] total guest instructions = 254,652,097,553
> ```
>
> 还有你们给的specCPU2006的分数，在采集simpoint的时候用的--cpt-interval是多少？

20M