OpenXiangShan/NEMU #100: Tracing分支的切片在香山的 emu 上运行出错

KeithPiTsui, 2022-08-04:
我用 Tracing 分支，已经成功完成 profiling 生成 bbv 文件，然后通过 simpoint 程序，输出 simpoints 和 weights 文件，然后在 NEMU上生成切片。可是，那些切片在 emu 上运行不起来。另外，我把在Tracing分支输出的 simpoints 和 weights 拿到 cpt-bk分支的NEMU上，可以成功切片和在emu上运行。所以，我判断问题出现在Tracing分支的切片功能。下面是，Tracing分支切片使用的命令，以及在emu上的报错信息。请问，如何解决呢？

```bash
1 #!/usr/bin/sh                                                                                                                                                                                                    
2 ../build/riscv64-nemu-interpreter \
3     -b /riscv/xs-env/riscv-pk/build/bbl.bin \
4     -D . \
5     -w bbl \
6     -C take_cpt \
7     -S take_cpt \
8     --cpt-interval 1000000 \
9     -r /riscv/xs-env/NEMU/resource/gcpt_restore/build/gcpt.bin
```
```bash
$> emu -i take_cpt/bbl/20/_536240_0.072592_.gz -I 1000000
Emu compiled at Mar 21 2022, 20:50:05
The image is take_cpt/bbl/20/_536240_0.072592_.gz
Using simulated 16384MB RAM
Gzip file detected and loading image from extracted gz file
--diff is not given, try to use $(NEMU_HOME)/build/riscv64-nemu-interpreter-so by default
NemuProxy using /riscv/xs-env/NEMU/build/riscv64-nemu-interpreter-so
[src/memory/paddr.c:81,init_mem] mmap memory to anonymous file
The first instruction of core 0 has commited. Difftest enabled. 
ERROR: invalid mem read from paddr 0x10000000, NEMU raise illegal inst exception
sh: spike-dasm????????????

============== Commit Group Trace (Core 0) ==============
commit group [00]: pc 0010000000 cmtcnt 1 <--
commit group [01]: pc 0000000000 cmtcnt 0
commit group [02]: pc 0000000000 cmtcnt 0
commit group [03]: pc 0000000000 cmtcnt 0
commit group [04]: pc 0000000000 cmtcnt 0
commit group [05]: pc 0000000000 cmtcnt 0
commit group [06]: pc 0000000000 cmtcnt 0
commit group [07]: pc 0000000000 cmtcnt 0
commit group [08]: pc 0000000000 cmtcnt 0
commit group [09]: pc 0000000000 cmtcnt 0
commit group [10]: pc 0000000000 cmtcnt 0
commit group [11]: pc 0000000000 cmtcnt 0
commit group [12]: pc 0000000000 cmtcnt 0
commit group [13]: pc 0000000000 cmtcnt 0
commit group [14]: pc 0000000000 cmtcnt 0
commit group [15]: pc 0000000000 cmtcnt 0

============== Commit Instr Trace ==============
commit inst [00]: pc 0010000000 inst 0010029b wen 1 dst 00000005 data 0000000000000001 <--
commit inst [01]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [02]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [03]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [04]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [05]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [06]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [07]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [08]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [09]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [10]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [11]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [12]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [13]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [14]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [15]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [16]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [17]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [18]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [19]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [20]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [21]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [22]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [23]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [24]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [25]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [26]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [27]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [28]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [29]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [30]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [31]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000

==============  REF Regs  ==============
  $0: 0x0000000000000000   ra: 0x88b0117316fdf463   sp: 0x8f2fde55abec1f2c   gp: 0x2b4b239c6259e1a0 
  tp: 0x96d46bf23bffa86b   t0: 0x0000000000000001   t1: 0xc24c1caf1b8e6823   t2: 0x5aa7032eca00dd23 
  s0: 0x3f626b247aae3353   s1: 0x5ba53723b34756c0   a0: 0x380cdb07a5a58caf   a1: 0x377ed3564e95cc4d 
  a2: 0xd411fef010dade06   a3: 0x1ba6d3fb1cec189a   a4: 0x6f018a24fc544269   a5: 0x1bf43813a7fbb7ac 
  a6: 0xfe77eca18ab07b80   a7: 0xa1b66f5f3f3cd109   s2: 0x545a1831dfe58105   s3: 0x2a325a3ecb635806 
  s4: 0xedd2db71c4214781   s5: 0x1b98f2cef9c3e7da   s6: 0x77462083ed0f71cc   s7: 0x84a8e2d1d810ca56 
  s8: 0x85337553e885ffd8   s9: 0x533998b77e546d91  s10: 0x4b876709ca0f486e  s11: 0x406cce6a1327b057 
  t3: 0x741d9a8403c08905   t4: 0xca81c3af810f34d6   t5: 0x1fc3c62f86cce9b7   t6: 0x4ebf40bcd97af7c0 
 ft0: 0x865b3aabd7d43852  ft1: 0x4897fb25bb1d7953  ft2: 0x3f660f30658f5273  ft3: 0x092a4a5c1a02f1fb 
 ft4: 0x22803dca6532c49f  ft5: 0xf3dc71c8f2311850  ft6: 0x8e3552dbc8174309  ft7: 0xd3cac1431d6005d0 
 fs0: 0x3e56a9c5d0725871  fs1: 0x5afd5e76867d5733  fa0: 0x72077dd16195a6c7  fa1: 0x2ee785f650d0a8bd 
 fa2: 0x9779c5ecf5cf4a00  fa3: 0x5ee50155d0597c45  fa4: 0xfbb8c7a71ecf70ca  fa5: 0x6a48b8ccf3393d70 
 fa6: 0x1613954ae0e66c87  fa7: 0x2f16489d4394d006  fs2: 0x335e038d2fb37a58  fs3: 0x5b8fbd26f0940e89 
 fs4: 0xa8ce72642fdb0fcb  fs5: 0xce2bf90ee812f68c  fs6: 0x64380a3a0d36250e  fs7: 0x702d1e2d0f1d0dec 
 fs8: 0xfe19b4edb1c1cae4  fs9: 0xe2951460b4c3737d fs10: 0x8fb12a5de2ed0158 fs11: 0x05b3a50d5fd64c27 
 ft8: 0x3ffff7aacee2f812  ft9: 0x4a04aa0facff9f79 ft10: 0xa3d49d7740a62609 ft11: 0x970d4b40f4c483cf 
pc: 0x0000000000000000 mstatus: 0x0000000a00001800 mcause: 0x0000000000000002 mepc: 0x0000000010000000
                       sstatus: 0x0000000200000000 scause: 0x0000000000000000 sepc: 0x0000000000000000
satp: 0x0000000000000000
mip: 0x0000000000000000 mie: 0x0000000000000000 mscratch: 0x0000000000000000 sscratch: 0x0000000000000000
mideleg: 0x0000000000000000 medeleg: 0x0000000000000000
mtval: 0x0000000000000000 stval: 0x000000104ee9bfbd mtvec: 0x0000000000000000 stvec: 0x0000000000000000
privilege mode:3  pmp: below
 0: cfg:0x00 addr:0x0000000000000000| 1: cfg:0x00 addr:0x0000000000000000
 2: cfg:0x00 addr:0x0000000000000000| 3: cfg:0x00 addr:0x0000000000000000
 4: cfg:0x00 addr:0x0000000000000000| 5: cfg:0x00 addr:0x0000000000000000
 6: cfg:0x00 addr:0x0000000000000000| 7: cfg:0x00 addr:0x0000000000000000
 8: cfg:0x00 addr:0x0000000000000000| 9: cfg:0x00 addr:0x0000000000000000
10: cfg:0x00 addr:0x0000000000000000|11: cfg:0x00 addr:0x0000000000000000
12: cfg:0x00 addr:0x0000000000000000|13: cfg:0x00 addr:0x0000000000000000
14: cfg:0x00 addr:0x0000000000000000|15: cfg:0x00 addr:0x0000000000000000
priviledgeMode: 3
mstatus different at pc = 0x0010000000, right= 0x0000000a00001800, wrong = 0x0000000a00000000
 mcause different at pc = 0x0010000000, right= 0x0000000000000002, wrong = 0x0000000000000000
   mepc different at pc = 0x0010000000, right= 0x0000000010000000, wrong = 0xe636dbfe88cc0ed0
Core 0: ABORT at pc = 0x321db953d7
total guest instructions = 1
instrCnt = 1, cycleCnt = 558, IPC = 0.001792
Saving snapshots to file system. Please wait.
Please remove unused snapshots manually
Seed=0 Guest cycle spent: 559 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 3,089ms
```

shinezyy, 2022-09-14:
编译riscv64-nemu-interpreter-so用的是什么config？
你的这个报错应该是difftest用的NEMU的ref版本不对，可能是分支用错了，也可能是config用错了。
另外，我们不用tracing分支做difftest。

HUA-FENG1995, 2022-09-22:
> 编译riscv64-nemu-interpreter-so用的是什么config？ 你的这个报错应该是difftest用的NEMU的ref版本不对，可能是分支用错了，也可能是config用错了。 另外，我们不用tracing分支做difftest。

我们用master分支跟emu进行差分测试，也遇到了相同的问题，请问这个问题怎么解决？

shinezyy, 2022-09-22:
> > 编译riscv64-nemu-interpreter-so用的是什么config？ 你的这个报错应该是difftest用的NEMU的ref版本不对，可能是分支用错了，也可能是config用错了。 另外，我们不用tracing分支做difftest。
> 
> 我们用master分支跟emu进行差分测试，也遇到了相同的问题，请问这个问题怎么解决？

编译riscv64-nemu-interpreter-so用的是什么config？

如果方便的话，可以把运行的命令贴出来

HUA-FENG1995, 2022-09-22:
make riscv64-xs-ref_defconfig
make -j
编译完成后，在NEMU目录下生成目录build，生成riscv64-nemu-interpreter-so可执行文件，用于emu进行差分测试。

HUA-FENG1995, 2022-09-22:
> > > 编译riscv64-nemu-interpreter-so用的是什么config？ 你的这个报错应该是difftest用的NEMU的ref版本不对，可能是分支用错了，也可能是config用错了。 另外，我们不用tracing分支做difftest。
> > 
> > 
> > 我们用master分支跟emu进行差分测试，也遇到了相同的问题，请问这个问题怎么解决？
> 
> 编译riscv64-nemu-interpreter-so用的是什么config？
> 
> 如果方便的话，可以把运行的命令贴出来

make riscv64-xs-ref_defconfig
make -j
编译完成后，在NEMU目录下生成目录build，生成riscv64-nemu-interpreter-so可执行文件，用于emu进行差分测试。
用的是最新版NEMU

HUA-FENG1995, 2022-09-26:
> make riscv64-xs-ref_defconfig make -j 编译完成后，在NEMU目录下生成目录build，生成riscv64-nemu-interpreter-so可执行文件，用于emu进行差分测试。 用的是最新版NEMU

小哥哥，这个问题应该怎么解决呀？

HUA-FENG1995, 2022-09-26:
> make riscv64-xs-ref_defconfig make -j 编译完成后，在NEMU目录下生成目录build，生成riscv64-nemu-interpreter-so可执行文件，用于emu进行差分测试。 用的是最新版NEMU

请问我使用的运行命令对吗？还是需要修改riscv64-xs-ref_defconfig里面的内容。如果是ref版本不对，那需要怎么修改呀

AugustusWillisWang, 2022-09-26:
我们看到 log 里, NEMU 有这样的报错:
```
ERROR: invalid mem read from paddr 0x10000000, NEMU raise illegal inst exception
```

这里的地址问题在最新的 NEMU master config 上已经调整过:

https://github.com/OpenXiangShan/NEMU/blob/b590a5827087cbde639fb3b6cc3afbba3782e954/configs/riscv64-xs-ref_defconfig#L71

您可以确认下 difftest 使用的 NEMU commit id 吗? 可以确认下它是否是 master 分支上最新的版本?

shinezyy, 2022-09-27:
> CONFIG_HAS_FLASH=y

那您看看[configs/riscv64-xs-ref_defconfig](https://github.com/OpenXiangShan/NEMU/blob/b590a5827087cbde639fb3b6cc3afbba3782e954/configs/riscv64-xs-ref_defconfig#L71)是否有这一行：
CONFIG_HAS_FLASH=y

HUA-FENG1995, 2022-09-27:
> > CONFIG_HAS_FLASH=y
> 
> 那您看看[configs/riscv64-xs-ref_defconfig](https://github.com/OpenXiangShan/NEMU/blob/b590a5827087cbde639fb3b6cc3afbba3782e954/configs/riscv64-xs-ref_defconfig#L71)是否有这一行： CONFIG_HAS_FLASH=y

有的，这行代码代表什么意思呀？

shinezyy, 2022-09-27:
> > > CONFIG_HAS_FLASH=y
> > 
> > 
> > 那您看看[configs/riscv64-xs-ref_defconfig](https://github.com/OpenXiangShan/NEMU/blob/b590a5827087cbde639fb3b6cc3afbba3782e954/configs/riscv64-xs-ref_defconfig#L71)是否有这一行： CONFIG_HAS_FLASH=y
> 
> 有的，这行代码代表什么意思呀？

那就很奇怪了，你确定你们遇到的问题和上面贴的log完全一样吗？你们方便贴一下你们的log吗

shinezyy, 2022-09-27:
> > > CONFIG_HAS_FLASH=y
> > 
> > 
> > 那您看看[configs/riscv64-xs-ref_defconfig](https://github.com/OpenXiangShan/NEMU/blob/b590a5827087cbde639fb3b6cc3afbba3782e954/configs/riscv64-xs-ref_defconfig#L71)是否有这一行： CONFIG_HAS_FLASH=y
> 
> 有的，这行代码代表什么意思呀？

这行配置是设置NEMU的地址空间的，如果没有开启，那就会报主楼贴的那种错误。但是你的有这一行还报这个错就不是很合理了，我们无法重现这个错误，也就无法debug。

HUA-FENG1995, 2022-09-27:
> > > > CONFIG_HAS_FLASH=y
> > > 
> > > 
> > > 那您看看[configs/riscv64-xs-ref_defconfig](https://github.com/OpenXiangShan/NEMU/blob/b590a5827087cbde639fb3b6cc3afbba3782e954/configs/riscv64-xs-ref_defconfig#L71)是否有这一行： CONFIG_HAS_FLASH=y
> > 
> > 
> > 有的，这行代码代表什么意思呀？
> 
> 那就很奇怪了，你确定你们遇到的问题和上面贴的log完全一样吗？你们方便贴一下你们的log吗

> 这行配置是设置NEMU的地址空间的，如果没有开启，那就会报主楼贴的那种错误。但是你的有这一行还报这个错就不是很合理了，我们无法重现这个错误，也就无法debug。

`Emu compiled at Sep  2 2022, 15:47:18
The image is /riscv/penghuafeng/XS/riscv-tests/isa/build/rv64hf/p-rorw.bin
Using simulated 8192MB RAM
--diff is not given, try to use $(NEMU_HOME)/build/riscv64-nemu-interpreter-so by default
NemuProxy using /riscv/penghuafeng/NEMU/build/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled. 
ERROR: invalid mem read from paddr 0x0000000010000000, NEMU raise access exception
sh: spike-dasm: command not found

============== Commit Group Trace (Core 0) ==============
commit group [00]: pc 0010000000 cmtcnt 1 <--
commit group [01]: pc 0000000000 cmtcnt 0
commit group [02]: pc 0000000000 cmtcnt 0
commit group [03]: pc 0000000000 cmtcnt 0
commit group [04]: pc 0000000000 cmtcnt 0
commit group [05]: pc 0000000000 cmtcnt 0
commit group [06]: pc 0000000000 cmtcnt 0
commit group [07]: pc 0000000000 cmtcnt 0
commit group [08]: pc 0000000000 cmtcnt 0
commit group [09]: pc 0000000000 cmtcnt 0
commit group [10]: pc 0000000000 cmtcnt 0
commit group [11]: pc 0000000000 cmtcnt 0
commit group [12]: pc 0000000000 cmtcnt 0
commit group [13]: pc 0000000000 cmtcnt 0
commit group [14]: pc 0000000000 cmtcnt 0
commit group [15]: pc 0000000000 cmtcnt 0

============== Commit Instr Trace ==============
commit inst [00]: pc 0010000000 inst 0010029b wen 1 dst 00000005 data 0000000000000001 <--
commit inst [01]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [02]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [03]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [04]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [05]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [06]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [07]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [08]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [09]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [10]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [11]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [12]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [13]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [14]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [15]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [16]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [17]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [18]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [19]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [20]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [21]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [22]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [23]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [24]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [25]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [26]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [27]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [28]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
commit inst [29]: pc 0000000000 inst 00000000 wen 0 dst 00000000 data 0000000000000000
[... 20 lines cut ...]
pc: 0x0000000000000000 mstatus: 0x0000000a00001800 mcause: 0x0000000000000001 mepc: 0x0000000010000000
                       sstatus: 0x0000000200000000 scause: 0x0000000000000000 sepc: 0x0000000000000000
satp: 0x0000000000000000
mip: 0x0000000000000000 mie: 0x0000000000000000 mscratch: 0x0000000000000000 sscratch: 0x0000000000000000
mideleg: 0x0000000000000000 medeleg: 0x0000000000000000
mtval: 0x0000000010000000 stval: 0x0000000000000000 mtvec: 0x0000000000000000 stvec: 0x0000000000000000
privilege mode:3  pmp: below
 0: cfg:0x00 addr:0x0000000000000000| 1: cfg:0x00 addr:0x0000000000000000
 2: cfg:0x00 addr:0x0000000000000000| 3: cfg:0x00 addr:0x0000000000000000
 4: cfg:0x00 addr:0x0000000000000000| 5: cfg:0x00 addr:0x0000000000000000
 6: cfg:0x00 addr:0x0000000000000000| 7: cfg:0x00 addr:0x0000000000000000
 8: cfg:0x00 addr:0x0000000000000000| 9: cfg:0x00 addr:0x0000000000000000
10: cfg:0x00 addr:0x0000000000000000|11: cfg:0x00 addr:0x0000000000000000
12: cfg:0x00 addr:0x0000000000000000|13: cfg:0x00 addr:0x0000000000000000
14: cfg:0x00 addr:0x0000000000000000|15: cfg:0x00 addr:0x0000000000000000
priviledgeMode: 3
[... cut ...] 00
 mcause different at pc = 0x0010000000, right= 0x0000000000000001, wrong = 0x0000000000000000
   mepc different at pc = 0x0010000000, right= 0x0000000010000000, wrong = 0x0000000000000000
  mtval different at pc = 0x0010000000, right= 0x0000000010000000, wrong = 0x0000000000000000
Core 0: ABORT at pc = 0x0
total guest instructions = 1
instrCnt = 1, cycleCnt = 555, IPC = 0.001802
Seed=0 Guest cycle spent: 556 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 88ms`

shinezyy, 2022-09-27:
我用最新的NEMU无法重现这个错误：
我使用的NEMU的commit是b56b00f80a436b04cf2901f493d06e05974b9fb9

我运行的命令是：

NEMU ref编译：
export NEMU_HOME=`pwd`
rm build -rf
make riscv64-xs-ref_defconfig
make menuconfig  --> Save
make -j30

emu运行：
emu -i checkpoint.gz --diff=../xs-ref-nemu/build/riscv64-nemu-interpreter-so

我不确定是不是你编译NEMU或者运行emu的时候NEMU_HOME指向了错误的地址，或者其他原因

HUA-FENG1995, 2022-09-27:
> 我用最新的NEMU无法重现这个错误： 我使用的NEMU的commit是b56b00f80a436b04cf2901f493d06e05974b9fb9
>
> 我运行的命令是：
>
> NEMU ref编译： export NEMU_HOME=`pwd` rm build -rf make riscv64-xs-ref_defconfig make menuconfig --> Save make [...]

NEMU_HOME应该需要指向什么地址呀？我根据你的命令执行了一次，还是不行。

NEMU ref编译命令和你一样
emu运行命令是：
./build/emu -i ../riscv-tests/isa/build/rv64hf/p-rorw.bin

HUA-FENG1995, 2022-10-17:
> NEMU_HOME 指向 commit 为 [b56b00f](https://github.com/OpenXiangShan/NEMU/commit/b56b00f80a436b04cf2901f493d06e05974b9fb9) 的 NEMU
> 
> ```
> 我在NEMU目录下输入git log命令查看，报了个fatal，
> fatal: Not a git repository (or any parent up to mount point /riscv)
> Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
> ```
> 
> 这样的话看不出来 NEMU 版本，请单独 clone 一个最新的 NEMU：`https://github.com/OpenXiangShan/NEMU`

我们在linux下重新下载了NEMU，是b56b00f的 NEMU，还是单独能运行，差分会报这个地址错误。指向NEMU_HOME的地址是没问题的，请问还有没有其他方向可以解决这个问题？

Jishuaishuai123, 2022-10-19:
> NEMU_HOME 指向 commit 为 [b56b00f](https://github.com/OpenXiangShan/NEMU/commit/b56b00f80a436b04cf2901f493d06e05974b9fb9) 的 NEMU
>
> ```
> 我在NEMU目录下输入git log命令查看，报了个fatal，
> fatal: Not a git repository (or [...]

Hi 您好：我在用最新版NEMU与EMU进行差分时也遇到了相同的问题：ERROR:invalid mem read from paddr 0x00000000010000000,NEMU raise access exception，根据上诉线索进行尝试，未解决；另外，我用上诉提供的编译方法来编译旧版NEMU也是可以成功差分的。请问这个问题有没有可能是因为EMU版本的问题导致呢？期待您的回复，谢谢。

Jishuaishuai123, 2022-10-25:
> 这行配置是设置NEMU的地址空间的，如果没有开启，那就会报主楼贴的那种错误。但是你的有这一行还报这个错就不是很合理了，我们无法重现这个错误，也就无法debug。

您好：如您所说，使用相同的版本NEMU，我们这边差分报错（ERROR:invalid mem read from paddr 0x00000000010000000,NEMU raise access exception），这个问题和库的缺失有关系吗

Jishuaishuai123, 2022-10-25:
> 这行配置是设置NEMU的地址空间的，如果没有开启，那就会报主楼贴的那种错误。但是你的有这一行还报这个错就不是很合理了，我们无法重现这个错误，也就无法debug。
您好：使用相同的版本NEMU，你们差分OK，我们这边差分报错（ERROR:invalid mem read from paddr 0x00000000010000000,NEMU raise access exception），这个问题和库的缺失有关系吗