OpenXiangShan/NEMU #101: tracing分支 在生成最后的切片不会停止

bersarker, 2022-08-06:
Tracing分支的NEMU切完最后一片后，会停着不动，没有正常退出，如下图所示。生成完最后一片后，会接着生成一片为0000.gz，然后就停着不动了，不能正常退出，请问是什么问题呢？
[image]

shinezyy, 2022-09-14:
SPEC跑完之后有没有执行 after workload？（一个elf，里面调用了NEMU的伪指令来退出执行）

RichardRo, 2023-04-06:
用tracing分支也遇到这个问题，同样是生成所有checkpoint之后还会继续生成一个_0_0.000000_.gz，然后卡住。
用了after workload。
cpb-bk分支没这个问题。

大概比了一下tracing和cpt-bk代码，感觉是不是tracing分支退出checkpoint的判断有点问题？
cbt-bk分支在serializer.notify_taken(icount); 里判断simpoint2Weights都处理完为空了，就直接exit(0); 退出了程序，见下面的比对：
[image]

而tracing分支在notify_taken没有退出，退出的判断留给了 if (serializer.shouldTakeCpt(icount))，shouldTakeCpt里即使 simpoint2Weights.empty()为真，还需要 && !checkpoint_taking，这时候checkpoint_taking应该为true吧，所以最终结果是false，无法退出，继续进行checkpoint？
[image]

shinezyy, 2023-04-07:
This issue should be finished in #121

closed 2024-01-25 by shinezyy