OpenXiangShan/NEMU #189: 关于使用NEMU-tracing生成baremetal app checkpoint

Bwoah-Kimi, 2023-09-22:
在(https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/simpoint/)文档中，默认流程是构建一个linux内核并在linux中运行benchmark，再打包成bbl.bin让NEMU运行。目前我想要直接让NEMU运行bin文件，例如简单的hello world，请问是可行的吗？

Bwoah-Kimi, 2023-09-22:
目前已经将gcpt_restore的内存地址地址空出，且已经在运行目标程序之前完成关闭时钟中断 和 进入 Simpoint Profiling 模式这两个操作。
profiling结果如下所示：
[profiling.txt](https://github.com/OpenXiangShan/NEMU/files/12696497/profiling.txt)
clustering结果如下所示：
[cluster.txt](https://github.com/OpenXiangShan/NEMU/files/12696499/cluster.txt)
但是运行checkpoint时目标文件夹没有输出，运行结果如下所示：
[checkpoint.txt](https://github.com/OpenXiangShan/NEMU/files/12696503/checkpoint.txt)

恳请解答！

Bwoah-Kimi, 2023-09-24:
大概知道问题出现在什么地方了。我让NEMU-tracing运行的是nexus-am生成的coremark二进制文件，没有放在linux里面，NEMU进入的是Machine Mode，在不使用--cpt-mmode情况下是不会生成checkpoint的。
请问m mode生成checkpoint可能会在什么情况下出现问题？如果执行的是可信代码的话，是否--cpt-mmode也能够得到理想的结果，并且能够让GEM5和emu运行呢？

shinezyy, 2024-01-25:
> 大概知道问题出现在什么地方了。我让NEMU-tracing运行的是nexus-am生成的coremark二进制文件，没有放在linux里面，NEMU进入的是Machine Mode，在不使用--cpt-mmode情况下是不会生成checkpoint的。 请问m mode生成checkpoint可能会在什么情况下出现问题？如果执行的是可信代码的话，是否--cpt-mmode也能够得到理想的结果，并且能够让GEM5和emu运行呢？

可以参考这个issue：
https://github.com/OpenXiangShan/NEMU/issues/54

closed 2025-12-16 by cebarobot