OpenXiangShan/NEMU #238: 关于NEMU配置(单独运行)模式遇到的问题

Jeacy, 2023-12-22:
根据文档https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/nemu/，

cebarobot, 2023-12-22:
这可能是 gcc 版本的问题，能否请您提供一下 `gcc --version` 的运行结果？
另外，我们目前只能保证香山相关工具在 Ubuntu 20.04 下能够正常运行，还是推荐您在 Ubuntu 20.04 环境下运行 NEMU。

This might be an issue with the gcc version. Could you please provide the output of gcc --version?
Additionally, currently we can only guarantee that the XiangShan-related tools run properly on Ubuntu 20.04. Therefore, we recommend running NEMU in an Ubuntu 20.04 environment.

Jeacy, 2023-12-24:
gcc的版本是9.3。抱歉目前的开发环境没法切换Ubuntu。

cebarobot, 2023-12-25:
您可以尝试按照 #140 中提供的办法，向 `NEMU/resource/softfloat/repo/build/Linux-x86_64-GCC/Makefile 46:48` 添加 `-fPIC` 看看能否解决问题。

RetHat 系发行版 CentOS 使用的软件包版本和 debian 系发行版 Ubuntu 使用的软件包版本差异非常巨大，这可能带来无穷无尽的奇怪问题，我们没有能力和精力来协助您处理发行版引发相关问题。
我们仍强烈建议您使用 Ubuntu 20.04 作为开发环境，您可以考虑使用虚拟机或者 Docker 来部署新的环境。

You could try adding `-fPIC` to `NEMU/resource/softfloat/repo/build/Linux-x86_64-GCC/Makefile 46:48` as suggested in #140 and see if it resolves the issue?

The differences in package versions between Red Hat-based distributions like CentOS and Debian-based distributions like Ubuntu can lead to countless unexpected issues. We lack the capability and resources to assist with distribution-related problems.
We strongly recommend using Ubuntu 20.04 as your development environment. You may consider deploying a new environment using a virtual machine or Docker.

Jeacy, 2023-12-25:
尝试用了以上方法还是报错。若更新最新的NEMU代码是否能解决问题？

cebarobot, 2023-12-25:
> 尝试用了以上方法还是报错。若更新最新的NEMU代码是否能解决问题？

您可以更新 NEMU 代码再作尝试。但是，我无法复现您 626af2 的编译问题，我这里编译一切正常。这个编译问题是有关与 softfload 库的，更新 NEMU 代码不太可能修复这一问题。

另外，根据您提供的标错记录，您可以尝试将前面提到的 `-fPIC` 换成 `-fPIE` 再试试，很有可能解决您的问题。

Jeacy, 2023-12-25:
-fPIC 换成 -fPIE还是不行。我目前正在尝试使用Ubuntu重新走一遍流程

closed 2024-01-04 by Jeacy