OpenXiangShan/NEMU #252: NEMU做profiling时出现错误，Failed to execute /init (error -2)

DimanChauncey, 2024-01-17:
自行编译了riscv的coremark，使用的是https://github.com/riscv-boom/riscv-coremark，加入了nemu_signal，做了bbl.bin，NEMU做profiling时出现错误，Failed to execute /init (error -2)，想问下该如何解决

详细输出：
Welcome to riscv64-NEMU!
For help, type "help"
bbl loader
freq-mhz = 500
CLINT: set frequency to 500 MHz
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.18.0-14486-g655055af981b-dirty (root@73c87a4ba4b3) (gcc version 10.5.0 (Ubuntu 10.5.0-1ubuntu1~20.04)) #26 Wed Jan 17 16:26:46 CST 2024
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x(____ptrval____) (20480 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Cannot allocate SWIOTLB buffer
[    0.000000] elf_hwcap is 0x112d
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 7575
[    0.000000] Kernel command line: root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon
[    0.000000] Dentry cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 29100K/30720K available (713K kernel code, 76K rwdata, 102K rodata, 84K init, 98K bss, 1620K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000000] console [hvc0] enabled
[    0.000000] console [hvc0] enabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.010000] pid_max: default: 4096 minimum: 301
[    0.010000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.010000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.030000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.030000] futex hash table entries: 16 (order: -4, 384 bytes)
[src/isa/riscv64/system/mmu.c:216,isa_mmu_check] addr misaligned happened: vaddr:ffffffff800c78b4 len:8 type:1 pc:ffffffff80015810
[    0.050000] clocksource: Switched to clocksource riscv_clocksource
[    0.050000] Unpacking initramfs...
[    0.110000] workingset: timestamp_bits=62 max_order=13 bucket_order=0
[    0.130000] random: get_random_bytes called from 0xffffffff80018cd0 with crng_init=0
[    0.130000] Freeing unused kernel memory: 84K
[    0.130000] This architecture does not have kernel memory protection.
[    0.130000] Failed to execute /init (error -2)
[    0.130000] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[    0.130000] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]---

DimanChauncey, 2024-01-17:
我设置了，file /init ${RISCV_ROOTFS_HOME}/rootfsimg/coremark 755 0 0

cebarobot, 2024-01-18:
那请您检查 coremark 是否正确编译，文件是否完整，环境变量 `${RISCV_ROOTFS_HOME}` 有没有正确设置。

DimanChauncey, 2024-01-23:
官方的coremark需要做什么修改，才能在nemu上进行profiling然后打checkpoint么？

cebarobot, 2024-01-23:
请参考香山文档中有关simpoint的内容，以及我们以往的tutorial

shinezyy, 2024-01-23:
这个问题分为两部分：
1. 官方的coremark需要做什么修改，才能在nemu上运行？
2. 官方的coremark需要做什么修改，才能在nemu上进行profiling然后打checkpoint？

这里的问题和profiling没关系。

建议提问者先不考虑profiling的事情，先把coremark跑起来。

DimanChauncey, 2024-01-24:
感谢回答，是这样的，我用qemu运行了同样的coremark程序，是可以run起来的。以下是这个编译好的程序的链接：https://drive.google.com/file/d/1RmAHQePZSP9OCWp_uJZNRO551Vu8JLxD/view?usp=sharing

按照教程制作成bbl.bin以后，NEMU profiling报错是/init返回-2，同样的制作bbl流程，hello程序和用的另一个benchmark可以正常运行。bbl.bin文件是以下链接：https://drive.google.com/file/d/1RbbMMifxAnIoeN9RyjzjCNagPCb-SjF4/view?usp=sharing

riscv-linux/riscv-pk的使用都和教程一样，coremark使用的工具链是riscv64-unknown-linux-gnu，pk使用的工具链是riscv64-unknown-elf，打包之前也执行了make clean.

riscv-rootfs的initramfs-emu.txt文件如下所示：
dir /dev 755 0 0
nod /dev/console 644 0 0 c 5 1
nod /dev/null 644 0 0 c 1 3
file /init ${RISCV_ROOTFS_HOME}/rootfsimg/coremark 755 0 0

其中coremark软链接了以上提到的google drive链接中程序coremark.riscv。

shinezyy, 2024-01-24:
麻烦提供一下你用的分支，git commit hash 和 NEMU用的config

shinezyy, 2024-01-24:
请提供一下NEMU的分支、commit hash id和编译时用的config文件

DimanChauncey, 2024-01-24:
NEMU: tracing分支  5429318e2b83

NEMU: 编译时的命令是：make riscv64-xs-simpoint_defconfig，这个文件我没有更改，是该hash id的源文件

NEMU profiling命令：$NEMU_HOME/build/riscv64-nemu-interpreter $RISCV_PK_HOME/build/bbl.bin -D /home/test -w coremark -C profiling -b --simpoint-profile --cpt-interval 100000-r $NEMU_HOME/resource/gcpt_restore/build/gcpt.bin

shinezyy, 2024-01-24:
> NEMU: tracing分支 [5429318](https://github.com/OpenXiangShan/NEMU/commit/5429318e2b83521cc862d28f2389dd5c899ffe32)
> 
> NEMU: 编译时的命令是：make riscv64-xs-simpoint_defconfig，这个文件我没有更改，是该hash id的源文件
> 
> NEMU profiling命令：$NEMU_HOME/build/riscv64-nemu-interpreter $RISCV_PK_HOME/build/bbl.bin -D /home/test -w coremark -C profiling -b --simpoint-profile --cpt-interval 100000-r $NEMU_HOME/resource/gcpt_restore/build/gcpt.bin

可以用master分支试一试

DimanChauncey, 2024-01-24:
还是同样的问题，运行如下：

[src/monitor/monitor.c:181,parse_args] Doing Simpoint Profiling
[src/checkpoint/path_manager.cpp:54,init] Cpt id: -1
[src/checkpoint/path_manager.cpp:72,setSimpointProfilingOutputDir] Created /home/zxk/nemu_result/profiling/coremark/

[src/checkpoint/simpoint.cpp:83,init] Doing simpoint profiling with interval 100000
[src/checkpoint/serializer.cpp:300,next_index] set next index 0
[src/checkpoint/path_manager.cpp:91,setCheckpointingOutputDir] donot set checkpoint path without Checkpoint mode
[src/isa/riscv64/init.c:135,init_isa] NEMU will start from pc 0x80000000
[src/monitor/monitor.c:337,init_monitor] You are providing a gcpt restorer when doing simpoing profiling, If you didn't link the program correctly, this will corrupt your memory/program.
[src/monitor/image_loader.c:74,load_img] Loading Gcpt restorer form cmdline: /root/NEMU/resource/gcpt_restore/build/gcpt.bin

[src/monitor/image_loader.c:99,load_img] Warning: size is larger than img_size(upper limit), please check if code is missing. size:1100 img_size:f00
[src/monitor/image_loader.c:119,load_img] Read 3840 bytes from file /root/NEMU/resource/gcpt_restore/build/gcpt.bin to 0x80000000
[src/monitor/image_loader.c:74,load_img] Loading image (bbl/bare metal app) from cmdline: /root/riscv-pk/build/bbl.bin

[src/monitor/image_loader.c:119,load_img] Read 2580068 bytes from file /root/riscv-pk/build/bbl.bin to 0x800a0000
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'uartlite' at [0x00000000000003f8, 0x0000000000000404]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'rtc' at [0x0000000000000048, 0x000000000000004f]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'screen' at [0x0000000000000100, 0x0000000000000107]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'keyboard' at [0x0000000000000060, 0x0000000000000063]
[src/device/sdcard.c:137,init_sdcard] Can not find sdcard image: 
[src/monitor/monitor.c:52,welcome] Debug: OFF
[src/monitor/monitor.c:57,welcome] Build time: 21:11:32, Jan 24 2024
Welcome to riscv64-NEMU!
For help, type "help"
bbl loader
freq-mhz = 500
CLINT: set frequency to 500 MHz
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.18.0-14486-g655055af981b-dirty (root@73c87a4ba4b3) (gcc version 10.5.0 (Ubuntu 10.5.0-1ubuntu1~20.04)) #56 Wed Jan 24 20:10:06 CST 2024
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x(____ptrval____) (54784 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Cannot allocate SWIOTLB buffer
[    0.000000] elf_hwcap is 0x112d
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 7575
[    0.000000] Kernel command line: root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon
[    0.000000] Dentry cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 29068K/30720K available (713K kernel code, 76K rwdata, 102K rodata, 116K init, 98K bss, 1652K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000000] console [hvc0] enabled
[    0.000000] console [hvc0] enabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.000000] pid_max: default: 4096 minimum: 301
[    0.000000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.020000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.020000] futex hash table entries: 16 (order: -4, 384 bytes)
[    0.040000] clocksource: Switched to clocksource riscv_clocksource
[    0.040000] Unpacking initramfs...
[    0.140000] workingset: timestamp_bits=62 max_order=13 bucket_order=0
[    0.170000] random: get_random_bytes called from 0xffffffff80020cd0 with crng_init=0
[    0.170000] Freeing unused kernel memory: 116K
[    0.170000] This architecture does not have kernel memory protection.
[    0.170000] Failed to execute /init (error -2)
[    0.170000] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[    0.170000] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]---

xyyy1420, 2024-01-25:
https://github.com/riscv-software-src/riscv-pk/commit/4ae5a8876fc2c31776b1777405ab14f764cc0f36 可以给pk添加这个补丁之后再发一份bbl.bin嘛？

DimanChauncey, 2024-01-25:
按照您提供的commit，加了代码以后，做成的bbl.bin、pk、vmlinux、coremark.riscv放在这个链接了：https://drive.google.com/file/d/1vXnUemh-DyXHXtpKULmTudsBeGaFKpta/view?usp=sharing

其中coremark和vmlinux采用riscv64-linux-gnu-编译而成，pk采用riscv64-unknown-elf-编译而成。
非常感谢各位！

xyyy1420, 2024-01-25:
在qemu中使用参数qemu-system-riscv64 -machine nemu -nographic -m 8G -smp1 -bios bbl.bin -cpu rv64,v=false,h=false运行得到了一样和NEMU一样的结果，基本可以排除NEMU实现上的问题；

后续建议参考rootfsimg目录下的文件，链接init为busybox，这样就可以在启动后执行脚本来检查自己打包的rootfs是否有问题；

截图参考下文，另外运行的是gcptl.bin是我打包了gcpt.bin和bbl.bin，和NEMU的实际执行是一致的。

[image]

xiaokamikami, 2024-01-25:
[image]

使用nemu(master分支)与spike做difftest，运行到Kernel panic后内部程序卡死，没有触发difftest报错，应该不是nemu的问题，建议检查一下Kernel 编译选项是否设置正确了嘛