OpenXiangShan/NEMU #267: Compilation failed when using ref_defconfig after other arch

cebarobot, 2024-02-22:
#### Description
If one have previously executed `make <some_defconfig> && make -j`, attempting `make <some-ref_defconfig> && make -j` will result in an error about softfloat.

#### Reproduce
```
git clone https://github.com/OpenXiangShan/NEMU.git
cd NEMU
export NEMU_HOME=$(pwd)
make riscv64-xs_defconfig
make -j
make riscv64-xs-ref_defconfig
make -j
```
And then got a Compilation error:
```
+ g++ ~/NEMU/build/riscv64-nemu-interpreter-so
/usr/bin/ld: resource/softfloat/build/softfloat.a(f16_sqrt.o): relocation R_X86_64_PC32 against symbol `softfloat_approxRecipSqrt_1k1s' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
```

## Environment
* Ubuntu 20.04.6 LTS
* gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
* NEMU dc3492f68e0aca5d3e1839c1c558bf0c71dc682f

## Analysis
When building stand-alone NEMU, Makefile won't pass `-fPIC` to softflow. Then building shared NEMU, softflow won't not be recompiled.

## One solution
Just make shared NEMU first (`make riscv64-xs-ref_defconfig && make -j`) when cloning a new repo.
Or, `rm -rf resource/softfloat/repo && rm -rf resource/softfloat/build` then build again.

poemonsense, 2024-02-22:
`make clean-all`

poemonsense, 2024-03-20:
Close this after #278. Open it again if `make clean` does not work.

closed 2024-03-20 by poemonsense

qiao-huan, 2025-01-14:
> `make clean-all`

Thank you，that works, although I don't know why it works. ^_^