OpenXiangShan/NEMU #269: checkpoint版本的bbl构建出现错误

YudeFang, 2024-02-24:
[image]
使用的版本 
riscv-pk noop分支 
riscv-roots checkpoint分支
riscv-linux nanshan分支

config.log详细信息
riscv64-unknown-linux-gnu-gcc (GCC) 11.1.0
Thread model: single
103 gcc version 9.3.0 () 
104 configure:2357: $? = 0
105 configure:2346: riscv64-unknown-elf-gcc -V >&5
106 riscv64-unknown-elf-gcc: error: unrecognized command line option '-V'
107 riscv64-unknown-elf-gcc: fatal error: no input files                                  
108 compilation terminated.
109 configure:2357: $? = 1
110 configure:2346: riscv64-unknown-elf-gcc -qversion >&5
111 riscv64-unknown-elf-gcc: error: unrecognized command line option '-qversion'; did you     mean '--version'?
112 riscv64-unknown-elf-gcc: fatal error: no input files
113 compilation terminated.
114 configure:2357: $? = 1
115 configure:2377: checking whether the C compiler works
116 configure:2399: riscv64-unknown-elf-gcc    conftest.c  >&5
117 /usr/lib/riscv64-unknown-elf/bin/ld: cannot find crt0.o: No such file or directory
118 /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lc
119 /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lgloss
120 collect2: error: ld returned 1 exit status
121 configure:2403: $? = 1
122 configure:2441: result: no
123 configure: failed program was:
124 | /* confdefs.h */
125 | #define PACKAGE_NAME "RISC-V Proxy Kernel"
126 | #define PACKAGE_TARNAME "riscv-pk"
127 | #define PACKAGE_VERSION "1.0.0"
128 | #define PACKAGE_STRING "RISC-V Proxy Kernel 1.0.0"
129 | #define PACKAGE_BUGREPORT "Andrew Waterman"
130 | #define PACKAGE_URL ""
131 | /* end confdefs.h.  */
132 |
133 | int
134 | main ()
135 | {
136 |
137 |   ;
138 |   return 0;
139 | }
140 configure:2446: error: in `/home/ray/other-disks/diskd/data-fyd/xiangshan-env/xs-env/riscv-pk/build':
41 configure:2448: error: C compiler cannot create executables
142 See `config.log' for more details

xyyy1420, 2024-02-26:
请根据需求自行修改initramfs-spec.txt

YudeFang, 2024-02-26:
> 请根据需求自行修改initramfs-spec.txt
[image]

config.log详细错误信息
112 compilation terminated.113 configure:2357: $? = 1114 configure:2377: checking whether the C compiler works115 configure:2399: riscv64-unknown-elf-gcc    conftest.c  >&5116 /usr/lib/riscv64-unknown-elf/bin/ld: cannot find crt0.o: No such file or directory 117 /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lc118 /usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lgloss119 collect2: error: ld returned 1 exit status
[image]

closed 2024-02-27 by xyyy1420