OpenXiangShan/NEMU #270: 构建spec2006 checkpoint workload出现问题，

YudeFang, 2024-02-27:
I delete this line in initramfs-spec.txt, 
**file /spec/astar_base ${SPEC_BUILD_PATH}/astar 755 0 0** 
from(https://github.com/OpenXiangShan/riscv-rootfs/blob/checkpoint/rootfsimg/initramfs-spec.txt)
and the **riscv-pk** is compiled successfully to generate bbl.bin.the following is my initramfs-spec.txt
[image]

But it seems that it can not be executed by NEMU correctly ,so where is wrong?

And at the same time , it can't be profiling by simpoint ,here is my bash 
[image]

the following is my initramfs-spec.txt and my SPEC2006 is compiled in riscv64 successfully

shinezyy, 2024-02-27:
> But it seems that it can not be executed by NEMU correctly ,so where is wrong?

You should provide your error log of NEMU

YudeFang, 2024-02-27:
INSTRUCTION SETS WANT TO BE FREE
invalid opcode(PC = 0x0000000080162020):
        02 90 97 88 01 00 93 88 ...
        88979002 88930001...
There are two cases which will trigger this unexpected exception:
1. The instruction at PC = 0x0000000080162020 is not implemented.
2. Something is implemented incorrectly.
Find this PC(0x0000000080162020) in the disassembling result to distinguish which case it is.

If it is the first case, see
       _                         __  __                         _ 
      (_)                       |  \/  |                       | |
  _ __ _ ___  ___ ________   __ | \  / | __ _ _ __  _   _  __ _| |
 | '__| / __|/ __|______\ \ / / | |\/| |/ _` | '_ \| | | |/ _` | |
 | |  | \__ \ (__        \ V /  | |  | | (_| | | | | |_| | (_| | |
 |_|  |_|___/\___|        \_/   |_|  |_|\__,_|_| |_|\__,_|\__,_|_|

for more details.

If it is the second case, remember:
* The machine is always right!
* Every line of untested code is always wrong!

[src/cpu/cpu-exec.c:708,cpu_exec] nemu: ABORT at pc = 0x0000000080162020
[src/cpu/cpu-exec.c:714,cpu_exec] trap code:-1
[src/cpu/cpu-exec.c:93,monitor_statistic] host time spent = 5895 us
[src/cpu/cpu-exec.c:95,monitor_statistic] total guest instructions = 193237
[src/cpu/cpu-exec.c:97,monitor_statistic] simulation frequency = 32779813 instr/s
[src/utils/state.c:28,is_exit_status_bad] NEMU exit with bad state: 3, halt ret: -1
ray@raymond:~/other-disks/diskd/data-fyd/xiangshan-env/xs-env/NEMU$ ./build/riscv64-nemu-interpreter -b ../../../bbl.bin > nemu.log
bbl loader
freq-mhz = 500
CLINT: set frequency to 500 MHz
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
ray@raymond:~/other-disks/diskd/data-fyd/xiangshan-env/xs-env/NEMU$ ls
build                      Kconfig      modify_pointdir_structure.py  ready-to-run  tools
checkpoint_example_result  lib-include  nemu.log                      resource
configs                    LICENSE      onesimpoint.py                scripts
include                    Makefile     README.md                     src
ray@raymond:~/other-disks/diskd/data-fyd/xiangshan-env/xs-env/NEMU$ vim nemu.log
ray@raymond:~/other-disks/diskd/data-fyd/xiangshan-env/xs-env/NEMU$ ./build/riscv64-nemu-interpreter -b ../../../bbl.bin 
[src/isa/riscv64/init.c:135,init_isa] NEMU will start from pc 0x80000000
[src/monitor/image_loader.c:203,load_img] Loading image (bbl/bare metal app) from cmdline: ../../../bbl.bin

[src/monitor/image_loader.c:257,load_img] Read 10135012 bytes from file ../../../bbl.bin to 0x80000000
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'uartlite' at [0x00000000000003f8, 0x0000000000000404]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'rtc' at [0x0000000000000048, 0x000000000000004f]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'screen' at [0x0000000000000100, 0x0000000000000107]
[... 35 lines cut ...]
There are two cases which will trigger this unexpected exception:
1. The instruction at PC = 0x0000000080162020 is not implemented.
2. Something is implemented incorrectly.
Find this PC(0x0000000080162020) in the disassembling result to distinguish which case it is.

If it is the first case, see
       _                         __  __                         _ 
      (_)                       |  \/  |                       | |
  _ __ _ ___  ___ ________   __ | \  / | __ _ _ __  _   _  __ _| |
 | '__| / __|/ __|______\ \ / / | |\/| |/ _` | '_ \| | | |/ _` | |
 | |  | \__ \ (__        \ V /  | |  | | (_| | | | | |_| | (_| | |
 |_|  |_|___/\___|        \_/   |_|  |_|\__,_|_| |_|\__,_|\__,_|_|

for more details.

If it is the second case, remember:
* The machine is always right!
* Every line of untested code is always wrong!

[src/cpu/cpu-exec.c:708,cpu_exec] nemu: ABORT at pc = 0x0000000080162020
[src/cpu/cpu-exec.c:714,cpu_exec] trap code:-1
[src/cpu/cpu-exec.c:93,monitor_statistic] host time spent = 5215 us
[src/cpu/cpu-exec.c:95,monitor_statistic] total guest instructions = 193237
[src/cpu/cpu-exec.c:97,monitor_statistic] simulation frequency = 37054074 instr/s
[src/utils/state.c:28,is_exit_status_bad] NEMU exit with bad state: 3, halt ret: -1

YudeFang, 2024-02-27:
> > But it seems that it can not be executed by NEMU correctly ,so where is wrong?
> 
> You should provide your error log of NEMU

INSTRUCTION SETS WANT TO BE FREE
invalid opcode(PC = 0x0000000080162020):
        02 90 97 88 01 00 93 88 ...
        88979002 88930001...
There are two cases which will trigger this unexpected exception:
1. The instruction at PC = 0x0000000080162020 is not implemented.
2. Something is implemented incorrectly.
Find this PC(0x0000000080162020) in the disassembling result to distinguish which case it is.

If it is the first case, see
       _                         __  __                         _ 
      (_)                       |  \/  |                       | |
  _ __ _ ___  ___ ________   __ | \  / | __ _ _ __  _   _  __ _| |
 | '__| / __|/ __|______\ \ / / | |\/| |/ _` | '_ \| | | |/ _` | |
 | |  | \__ \ (__        \ V /  | |  | | (_| | | | | |_| | (_| | |
 |_|  |_|___/\___|        \_/   |_|  |_|\__,_|_| |_|\__,_|\__,_|_|

for more details.

If it is the second case, remember:
* The machine is always right!
* Every line of untested code is always wrong!

[src/cpu/cpu-exec.c:708,cpu_exec] nemu: ABORT at pc = 0x0000000080162020
[src/cpu/cpu-exec.c:714,cpu_exec] trap code:-1
[src/cpu/cpu-exec.c:93,monitor_statistic] host time spent = 5215 us
[src/cpu/cpu-exec.c:95,monitor_statistic] total guest instructions = 193237
[src/cpu/cpu-exec.c:97,monitor_statistic] simulation frequency = 37054074 instr/s
[src/utils/state.c:28,is_exit_status_bad] NEMU exit with bad state: 3, halt ret: -1

shinezyy, 2024-02-27:
please provide full log

YudeFang, 2024-02-27:
[src/isa/riscv64/init.c:135,init_isa] NEMU will start from pc 0x80000000
[src/monitor/image_loader.c:203,load_img] Loading image (bbl/bare metal app) from cmdline: ../../../bbl.bin

[src/monitor/image_loader.c:257,load_img] Read 1017 bytes from file ../../../bbl.bin to 0x80000000
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'uartlite' at [0x00000000000003f8, 0x0000000000000404]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'rtc' at [0x0000000000000048, 0x000000000000004f]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'screen' at [0x0000000000000100, 0x0000000000000107]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'keyboard' at [0x0000000000000060, 0x0000000000000063]
[src/device/sdcard.c:137,init_sdcard] Can not find sdcard image: 
[src/monitor/monitor.c:54,welcome] Debug: OFF
[src/monitor/monitor.c:59,welcome] Build time: 00:40:45, Feb 25 2024
[src/cpu/cpu-exec.c:708,cpu_exec] nemu: ABORT at pc = 0x0000000080000000
[src/cpu/cpu-exec.c:714,cpu_exec] trap code:-1
[src/cpu/cpu-exec.c:93,monitor_statistic] host time spent = 434 us
[src/cpu/cpu-exec.c:95,monitor_statistic] total guest instructions = 65536
[src/cpu/cpu-exec.c:97,monitor_statistic] simulation frequency = 151004608 instr/s
[src/utils/state.c:28,is_exit_status_bad] NEMU exit with bad state: 3, halt ret: -1
 
it seems went wrong when NEMU just boot

shinezyy, 2024-02-27:
Which config is used

YudeFang, 2024-02-27:
./build/riscv64-nemu-interpreter --log=nemu.log  ../..
/../bbl.bin 
this one

xyyy1420, 2024-02-27:
[image]

If you're using this paragraph of the document as a reference, pay attention to where I've labeled it

xyyy1420, 2024-02-27:
by the way, if SimPoint profiling has been failed, please provide full log( include stderr and stdout )

YudeFang, 2024-02-27:
> by the way, if SimPoint profiling has been failed, please provide full log( include stderr and stdout )
err log
riscv64-nemu-interpreter: src/utils/filename.c:34: is_gz_file: Assertion `0' failed.   
&&out log is blank

xyyy1420, 2024-02-27:
[image]

xyyy1420, 2024-02-27:
If there are no other questions,  I'll close this issue?

YudeFang, 2024-02-27:
thank you very much ,it's ok

closed 2024-02-27 by xyyy1420