OpenXiangShan/NEMU #271: SPEC2006 profiling is failed

YudeFang, 2024-02-27:
After I modify the script, the profiling is still failed
[image]

[src/monitor/monitor.c:184,parse_args] Doing Simpoint Profiling
[src/checkpoint/path_manager.cpp:54,init] Cpt id: -1
[src/checkpoint/path_manager.cpp:72,setSimpointProfilingOutputDir] Created /home/ray/other-disks/diskd/data-fyd/xiangshan-env/xs-env/NEMU/checkpoint_example_result/simpoint-profiling/bbl/

[src/checkpoint/simpoint.cpp:83,init] Doing simpoint profiling with interval 20000000
[src/checkpoint/serializer.cpp:344,next_index] set next index 0
[src/checkpoint/path_manager.cpp:91,setCheckpointingOutputDir] donot set checkpoint path without Checkpoint mode
[src/isa/riscv64/init.c:135,init_isa] NEMU will start from pc 0x80000000
[src/monitor/monitor.c:350,init_monitor] You are providing a gcpt restorer when doing simpoing profiling, If you didn't link the program correctly, this will corrupt your memory/program.
[src/monitor/image_loader.c:203,load_img] Loading Gcpt restorer form cmdline: /home/ray/other-disks/diskd/data-fyd/xiangshan-env/xs-env/NEMU/resource/gcpt_restore/build/gcpt.bin

[src/monitor/image_loader.c:237,load_img] Warning: size is larger than img_size(upper limit), please check if code is missing. size:1100 img_size:f00
[src/monitor/image_loader.c:257,load_img] Read 3840 bytes from file /home/ray/other-disks/diskd/data-fyd/xiangshan-env/xs-env/NEMU/resource/gcpt_restore/build/gcpt.bin to 0x80000000
[src/monitor/image_loader.c:203,load_img] Loading image (bbl/bare metal app) from cmdline: /home/ray/other-disks/diskd/data-fyd/bbl.bin

[src/monitor/image_loader.c:257,load_img] Read 1017 bytes from file /home/ray/other-disks/diskd/data-fyd/bbl.bin to 0x800a0000
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'uartlite' at [0x00000000000003f8, 0x0000000000000404]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'rtc' at [0x0000000000000048, 0x000000000000004f]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'screen' at [0x0000000000000100, 0x0000000000000107]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'keyboard' at [0x0000000000000060, 0x0000000000000063]
[src/device/sdcard.c:137,init_sdcard] Can not find sdcard image: 
[src/monitor/monitor.c:54,welcome] Debug: OFF
[src/monitor/monitor.c:59,welcome] Build time: 00:40:45, Feb 25 2024
Welcome to riscv64-NEMU!
For help, type "help"
invalid opcode(PC = 0x00000000800a0000):
        1b 5b 31 3b 33 34 6d 5b ...
        3b315b1b 5b6d3433...
There are two cases which will trigger this unexpected exception:
1. The instruction at PC = 0x00000000800a0000 is not implemented.
2. Something is implemented incorrectly.
Find this PC(0x00000000800a0000) in the disassembling result to distinguish which case it is.
[src/cpu/cpu-exec.c:708,cpu_exec] nemu: ABORT at pc = 0x00000000800a0000
[src/cpu/cpu-exec.c:714,cpu_exec] trap code:-1
[src/cpu/cpu-exec.c:93,monitor_statistic] host time spent = 343 us
[src/cpu/cpu-exec.c:95,monitor_statistic] total guest instructions = 9
[src/cpu/cpu-exec.c:97,monitor_statistic] simulation frequency = 26239 instr/s
[src/utils/state.c:28,is_exit_status_bad] NEMU exit with bad state: 3, halt ret: -1

YudeFang, 2024-02-28:
here is my bbl.bin
[bbl.bin.zip](https://github.com/OpenXiangShan/NEMU/files/14427673/bbl.bin.zip)

xiaokamikami, 2024-02-28:
The error address should be the bbl entry point after loading gcpt restor. You should check the start address of the bbl disassembly. When using checkpoint recovery, you should offset the start address of bbl by a0000

YudeFang, 2024-02-28:
[image]
it seems wrong when I rebuild my bbl.bin,i didn't see this before. I search my ./riscv-rootf/rootfsimg/build,it seems other documents was all correct except busy box
[image]

xyyy1420, 2024-02-28:
please RTFM

YudeFang, 2024-02-28:
what do you mean RTFM?please specific,thank

xyyy1420, 2024-02-28:
read the friendly manual

YudeFang, 2024-02-29:
INSTRUCTION SETS WANT TO BE FREE

[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000

[    0.000000] Linux version 4.18.0-14486-g655055af981b-dirty (ray@raymond) (gcc version 11.1.0 (GCC)) #8 Thu Feb 29 11:23:18 CST 2024

[    0.000000] bootconsole [early0] enabled

[    0.000000] Initial ramdisk at: 0x(____ptrval____) (2343730 bytes)

[    0.000000] Zone ranges:

[    0.000000]   DMA32    empty

[    0.000000]   Normal   [mem 0x0000000080200000-0x00000000ffffffff]

[    0.000000] Movable zone start for each node

[    0.000000] Early memory node ranges

[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]

[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]

[    0.000000] software IO TLB [mem 0xfa3fe000-0xfe3fe000] (64MB) mapped at [(____ptrval____)-(____ptrval____)]

[    0.000000] elf_hwcap is 0x112d

[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615

[    0.000000] Kernel command line: root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon

[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)

[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)

[    0.000000] Sorting __ex_table...

[    0.000000] Memory: 1989836K/2095104K available (3188K kernel code, 214K rwdata, 914K rodata, 2420K init, 773K bss, 105268K reserved, 0K cma-reserved)

[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1

[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0

[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns

[    0.000000] Console: colour dummy device 80x25

[    0.000000] console [hvc0] enabled

[    0.000000] console [hvc0] enabled

[    0.000000] bootconsole [early0] disabled

[    0.000000] bootconsole [early0] disabled

[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)

[    0.000000] pid_max: default: 32768 minimum: 301

[    0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)

[    0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)

[    0.010000] devtmpfs: initialized

[    0.010000] random: get_random_u32 called from bucket_table_alloc+0x6e/0x1a2 with crng_init=0

[    0.010000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns

[    0.010000] futex hash table entries: 256 (order: 0, 6144 bytes)

[    0.020000] NET: Registered protocol family 16

[    0.020000] SCSI subsystem initialized

[    0.030000] clocksource: Switched to clocksource riscv_clocksource

[    0.210000] NET: Registered protocol family 2

[    0.210000] tcp_listen_portaddr_hash hash table entries: 1024 (order: 2, 16384 bytes)

[    0.210000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)

[    0.210000] TCP bind hash table entries: 16384 (order: 5, 131072 bytes)

[    0.210000] TCP: Hash tables configured (established 16384 bind 16384)

[    0.210000] UDP hash table entries: 1024 (order: 3, 32768 bytes)

[    0.210000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)

[    0.210000] NET: Registered protocol family 1

[    0.310000] RPC: Registered named UNIX socket transport module.

[    0.310000] RPC: Registered udp transport module.

[    0.310000] RPC: Registered tcp transport module.

[    0.310000] RPC: Registered tcp NFSv4.1 backchannel transport module.

[    3.410000] Unpacking initramfs...

[    6.760000] workingset: timestamp_bits=62 max_order=19 bucket_order=0

[    6.940000] NFS: Registering the id_resolver key type

[    6.940000] Key type id_resolver registered

[    6.940000] Key type id_legacy registered

[    6.940000] nfs4filelayout_init: NFSv4 File Layout Driver Registering...

[    7.000000] NET: Registered protocol family 38

[    7.000000] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)

[    7.000000] io scheduler noop registered

[    7.000000] io scheduler deadline registered

[    7.000000] io scheduler cfq registered (default)

[    7.000000] io scheduler mq-deadline registered

[    7.000000] io scheduler kyber registered

[    8.400000] loop: module loaded

[    8.400000] NET: Registered protocol family 17

[    8.400000] Key type dns_resolver registered

[    8.400000] Freeing unused kernel memory: 2420K

[    8.400000] This architecture does not have kernel memory protection.

[    8.400000] Failed to execute /init (error -2)

[    8.400000] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.

[    8.400000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.18.0-14486-g655055af981b-dirty #8

[    8.400000] Call Trace:

[    8.400000] [<ffffffe00025ee7e>] walk_stackframe+0x0/0xa6

[    8.400000] [<ffffffe00025eff4>] show_stack+0x2a/0x34

[    8.400000] [<ffffffe0005659fa>] dump_stack+0x20/0x28

[    8.400000] [<ffffffe000262e04>] panic+0xc0/0x1d0

[    8.400000] [<ffffffe000576786>] kernel_init+0xdc/0xe8

[    8.400000] [<ffffffe00025dde2>] ret_from_exception+0x0/0xc

[    8.400000] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]---
it's my profiling error message, it seems NEMU can't find init documents to start a init process

xiaokamikami, 2024-02-29:
> ```
>    INSTRUCTION SETS WANT TO BE FREE
> ```
> 
> [ 0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
> 
> [ 0.000000] Linux version 4.18.0-14486-g655055af981b-dirty (ray@raymond) (gcc version 11.1.0 (GCC)) #8 Thu Feb 29 11:23:18 CST 2024
> 
> [ 0.000000] bootconsole [early0] enabled
> 
> [ 0.000000] Initial ramdisk at: 0x(**ptrval**) (2343730 bytes)
> 
> [ 0.000000] Zone ranges:
> 
> [ 0.000000] DMA32 empty
> 
> [ 0.000000] Normal [mem 0x0000000080200000-0x00000000ffffffff]
> 
> [ 0.000000] Movable zone start for each node
> 
> [ 0.000000] Early memory node ranges
> 
> [ 0.000000] node 0: [mem 0x0000000080200000-0x00000000ffffffff]
> 
> [ 0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
> 
> [ 0.000000] software IO TLB [mem 0xfa3fe000-0xfe3fe000] (64MB) mapped at [(**ptrval**)-(**ptrval**)]
> 
> [ 0.000000] elf_hwcap is 0x112d
> 
> [ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 516615
> 
> [ 0.000000] Kernel command line: root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon
> 
> [ 0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
> 
> [ 0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
> 
> [ 0.000000] Sorting __ex_table...
> 
> [ 0.000000] Memory: 1989836K/2095104K available (3188K kernel code, 214K rwdata, 914K rodata, 2420K init, 773K bss, 105268K reserved, 0K cma-reserved)
> 
> [ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
> 
> [ 0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
> 
> [ 0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
> 
> [ 0.000000] Console: colour dummy device 80x25
> 
> [ 0.000000] console [hvc0] enabled
> 
> [ 0.000000] console [hvc0] enabled
> 
> [ 0.000000] bootconsole [early0] disabled
> 
> [ 0.000000] bootconsole [early0] disabled
> 
> [ 0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
> 
> [ 0.000000] pid_max: default: 32768 minimum: 301
> 
> [ 0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
> 
> [ 0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
> 
> [ 0.010000] devtmpfs: initialized
> 
> [ 0.010000] random: get_random_u32 called from bucket_table_alloc+0x6e/0x1a2 with crng_init=0
> 
> [ 0.010000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
> 
> [ 0.010000] futex hash table entries: 256 (order: 0, 6144 bytes)
> 
> [ 0.020000] NET: Registered protocol family 16
> 
> [ 0.020000] SCSI subsystem initialized
> 
> [ 0.030000] clocksource: Switched to clocksource riscv_clocksource
> 
> [ 0.210000] NET: Registered protocol family 2
> 
> [ 0.210000] tcp_listen_portaddr_hash hash table entries: 1024 (order: 2, 16384 bytes)
> 
> [ 0.210000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
> 
> [ 0.210000] TCP bind hash table entries: 16384 (order: 5, 131072 bytes)
> 
> [ 0.210000] TCP: Hash tables configured (established 16384 bind 16384)
> 
> [ 0.210000] UDP hash table entries: 1024 (order: 3, 32768 bytes)
> 
> [ 0.210000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
> 
> [ 0.210000] NET: Registered protocol family 1
> 
> [ 0.310000] RPC: Registered named UNIX socket transport module.
> 
> [ 0.310000] RPC: Registered udp transport module.
> 
> [ 0.310000] RPC: Registered tcp transport module.
> 
> [ 0.310000] RPC: Registered tcp NFSv4.1 backchannel transport module.
> 
> [ 3.410000] Unpacking initramfs...
> 
> [ 6.760000] workingset: timestamp_bits=62 max_order=19 bucket_order=0
> 
> [ 6.940000] NFS: Registering the id_resolver key type
> 
> [ 6.940000] Key type id_resolver registered
> 
> [ 6.940000] Key type id_legacy registered
> 
> [ 6.940000] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
> 
> [ 7.000000] NET: Registered protocol family 38
> 
> [ 7.000000] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
> 
> [ 7.000000] io scheduler noop registered
> 
> [ 7.000000] io scheduler deadline registered
> 
> [ 7.000000] io scheduler cfq registered (default)
> 
> [ 7.000000] io scheduler mq-deadline registered
> 
> [ 7.000000] io scheduler kyber registered
> 
> [ 8.400000] loop: module loaded
> 
> [ 8.400000] NET: Registered protocol family 17
> 
> [ 8.400000] Key type dns_resolver registered
> 
> [ 8.400000] Freeing unused kernel memory: 2420K
> 
> [ 8.400000] This architecture does not have kernel memory protection.
> 
> [ 8.400000] Failed to execute /init (error -2)
> 
> [ 8.400000] Kernel panic - not syncing: No working init found. Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
> 
> [ 8.400000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.18.0-14486-g655055af981b-dirty #8
> 
> [ 8.400000] Call Trace:
> 
> [ 8.400000] [] walk_stackframe+0x0/0xa6
> 
> [ 8.400000] [] show_stack+0x2a/0x34
> 
> [ 8.400000] [] dump_stack+0x20/0x28
> 
> [ 8.400000] [] panic+0xc0/0x1d0
> 
> [ 8.400000] [] kernel_init+0xdc/0xe8
> 
> [ 8.400000] [] ret_from_exception+0x0/0xc
> 
> [ 8.400000] ---[ end Kernel panic - not syncing: No working init found. Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]--- it's my profiling error message, it seems NEMU can't find init documents to start a init process
I think this is also a liunx compilation problem, reference
https://github.com/OpenXiangShan/NEMU/issues/252

YudeFang, 2024-02-29:
I have update my riscv-rootfs to the newest checkpoint branch release ,but meets a new issue it seems /init can't find libm.so.6: 
but in initramfs-spec.txt ，I really set the correct path for libm.so.6 
[image]

INSTRUCTION SETS WANT TO BE FREE

[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000

[    0.000000] Linux version 4.18.0-14486-g655055af981b-dirty (ray@raymond) (gcc version 11.1.0 (GCC)) #9 Thu Feb 29 15:10:37 CST 2024

[    0.000000] bootconsole [early0] enabled

[    0.000000] Initial ramdisk at: 0x(____ptrval____) (6595063 bytes)

[    0.000000] Zone ranges:

[    0.000000]   DMA32    empty

[    0.000000]   Normal   [mem 0x0000000080200000-0x00000000ffffffff]

[    0.000000] Movable zone start for each node

[    0.000000] Early memory node ranges

[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]

[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]

[    0.000000] software IO TLB [mem 0xfa3fe000-0xfe3fe000] (64MB) mapped at [(____ptrval____)-(____ptrval____)]

[    0.000000] elf_hwcap is 0x112d

[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615

[    0.000000] Kernel command line: root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon

[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)

[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)

[    0.000000] Sorting __ex_table...

[    0.000000] Memory: 1985684K/2095104K available (3188K kernel code, 218K rwdata, 914K rodata, 6568K init, 773K bss, 109420K reserved, 0K cma-reserved)

[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1

[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0

[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns

[    0.000000] Console: colour dummy device 80x25

[    0.000000] console [hvc0] enabled

[    0.000000] console [hvc0] enabled

[    0.000000] bootconsole [early0] disabled

[    0.000000] bootconsole [early0] disabled

[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)

[    0.000000] pid_max: default: 32768 minimum: 301

[    0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)

[    0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)

[    0.010000] devtmpfs: initialized

[    0.010000] random: get_random_u32 called from bucket_table_alloc+0x6e/0x1a2 with crng_init=0

[    0.010000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns

[    0.010000] futex hash table entries: 256 (order: 0, 6144 bytes)

[    0.020000] NET: Registered protocol family 16

[    0.030000] SCSI subsystem initialized

[    0.040000] clocksource: Switched to clocksource riscv_clocksource

[    0.070000] NET: Registered protocol family 2

[    0.270000] tcp_listen_portaddr_hash hash table entries: 1024 (order: 2, 16384 bytes)

[    0.270000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)

[    0.270000] TCP bind hash table entries: 16384 (order: 5, 131072 bytes)

[    0.270000] TCP: Hash tables configured (established 16384 bind 16384)

[    0.270000] UDP hash table entries: 1024 (order: 3, 32768 bytes)

[    0.270000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)

[    0.270000] NET: Registered protocol family 1

[    0.320000] RPC: Registered named UNIX socket transport module.

[    0.320000] RPC: Registered udp transport module.

[    0.320000] RPC: Registered tcp transport module.

[    0.320000] RPC: Registered tcp NFSv4.1 backchannel transport module.

[    8.310000] Unpacking initramfs...

[   16.680000] workingset: timestamp_bits=62 max_order=19 bucket_order=0

[   16.860000] NFS: Registering the id_resolver key type

[   16.860000] Key type id_resolver registered

[   16.860000] Key type id_legacy registered

[   16.860000] nfs4filelayout_init: NFSv4 File Layout Driver Registering...

[   16.920000] NET: Registered protocol family 38

[   16.920000] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)

[   16.920000] io scheduler noop registered

[   16.920000] io scheduler deadline registered

[   16.920000] io scheduler cfq registered (default)

[   16.920000] io scheduler mq-deadline registered

[   16.920000] io scheduler kyber registered

[   18.310000] loop: module loaded

[   18.310000] NET: Registered protocol family 17

[   18.310000] Key type dns_resolver registered

[   18.310000] Freeing unused kernel memory: 6568K

[   18.310000] This architecture does not have kernel memory protection.

/init: error while loading shared libraries: libm.so.6: cannot open shared object file: No such file or directory

[   18.470000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00

[   18.470000] 

[   18.470000] CPU: 0 PID: 1 Comm: init Not tainted 4.18.0-14486-g655055af981b-dirty #9

[   18.470000] Call Trace:

[   18.470000] [<ffffffe00066be7e>] walk_stackframe+0x0/0xa6

[   18.470000] [<ffffffe00066bff4>] show_stack+0x2a/0x34

[   18.470000] [<ffffffe0009729fa>] dump_stack+0x20/0x28

[   18.470000] [<ffffffe00066fe04>] panic+0xc0/0x1d0

[   18.470000] [<ffffffe000670fa2>] do_exit+0x5c6/0x62a

[   18.470000] [<ffffffe000671820>] do_group_exit+0x22/0x5e

[   18.470000] [<ffffffe000671874>] __wake_up_parent+0x0/0x22

[   18.470000] [<ffffffe00066add4>] ret_from_syscall+0x0/0xe

[   18.470000] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00

[   18.470000]  ]---

shinezyy, 2024-03-01:
> /init: error while loading shared libraries: libm.so.6: cannot open shared object file: No such file or directory

这个报错是因为没找到libm这个库。要么把它放到init ramfs里，要么静态链接你的应用避免使用动态库。

YudeFang, 2024-03-01:
file /lib/libm.so.6 ${RISCV}/sysroot/lib64/lp64d/libm.so.6 755 0 0 这句不是已经放在initramfs里了么 路径也没问题

YudeFang, 2024-03-07:
问题解决了，run.sh脚本中加入export LD_LIBRARY_PATH=/lib