OpenXiangShan/NEMU #273: NEMU master分支编译找不到zstd.h和SDL2/SDL.h

qqli579, 2024-03-06:
参考文档https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/nemu/编译master分支，报如下错误
```
src/device/device.c:21:10: fatal error: SDL2/SDL.h: No such file or directory
   21 | #include <SDL2/SDL.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [/xiangshangfine/NEMU/scripts/build.mk:52: /xiangshangfine/NEMU/build/obj-riscv64-nemu-interpreter/src/device/device.o] Error 1
make: *** Waiting for unfinished jobs....
src/device/vga.c:18:10: fatal error: SDL2/SDL.h: No such file or directory
   18 | #include <SDL2/SDL.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [/xiangshangfine/NEMU/scripts/build.mk:52: /xiangshangfine/NEMU/build/obj-riscv64-nemu-interpreter/src/device/vga.o] Error 1
src/device/keyboard.c:18:10: fatal error: SDL2/SDL.h: No such file or directory
   18 | #include <SDL2/SDL.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [/xiangshangfine/NEMU/scripts/build.mk:52: /xiangshangfine/NEMU/build/obj-riscv64-nemu-interpreter/src/device/keyboard.o] Error 1
src/monitor/ui.c:28:10: fatal error: readline/readline.h: No such file or directory
   28 | #include <readline/readline.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [/xiangshangfine/NEMU/scripts/build.mk:52: /xiangshangfine/NEMU/build/obj-riscv64-nemu-interpreter/src/monitor/ui.o] Error 1
src/monitor/image_loader.c:28:10: fatal error: zstd.h: No such file or directory
   28 | #include <zstd.h>
      |          ^~~~~~~~
compilation terminated.
make: *** [/xiangshangfine/NEMU/scripts/build.mk:52: /xiangshangfine/NEMU/build/obj-riscv64-nemu-interpreter/src/monitor/image_loader.o] Error 1
src/checkpoint/serializer.cpp:39:10: fatal error: zstd.h: No such file or directory
   39 | #include <zstd.h>
      |          ^~~~~~~~
compilation terminated.
make: *** [/xiangshangfine/NEMU/scripts/build.mk:59: /xiangshangfine/NEMU/build/obj-riscv64-nemu-interpreter/src/checkpoint/serializer.opp] Error 1

```

shinezyy, 2024-03-06:
装库都不会，建议转行哈

shinezyy, 2024-03-06:
因为用户采用的发行版不一样，不同的发行版安装的命令不同。比如ubuntu，可以按照如下方法安装：

https://www.google.com/search?q=how+to+install+sdl2+on+ubuntu
https://www.google.com/search?q=how+to+install+libreadline+dev+on+ubuntu
https://www.google.com/search?q=how+to+install+zstd+on+ubuntu

我们后续会把这些命令更新到文档里，我们感谢您对文档的间接贡献。

同时，我们建议您遇到此类问题先进行搜索。如果这种级别的问题都无法独立解决，是不可能正常进行香山/NEMU/XS-gem5相关开发的，可以考虑换个别的项目或者工作。

qqli579, 2024-03-06:
香山这么烂的项目就不要开源了，拉低了中国开源的水平。

closed 2024-03-11 by shinezyy