OpenXiangShan/NEMU #274: spec2006在ubuntu20.04交叉编译生成的可执行文件无法进行simpoint

YudeFang, 2024-03-08:
如图，编译spec的时候我正确修改了gcc,g++,gfortran的路径，也能正常将spec编译成功
[image]
但是当我使用astar下面的可执行文件时，NEMU总是报错说无法找到配置文件，应该是spec编译的问题但我不清楚具体为啥报错了，我拿正确的astar程序可以正常simpoint
[image]

YudeFang, 2024-03-08:
我生成的可执行文件是这两个，我都试了 都不太行
[image]
还有就是我的可执行文件比正确的aster可执行文件貌似小很多，所以确实不知道哪一步错了

shinezyy, 2024-03-09:
额，你用的是unknown-elf-gcc，

你可以看看我们这个项目的readme，我们给出的编译speccpu06示例的时候用的是riscv64-unknown-linux-gnu-：
https://github.com/OpenXiangShan/CPU2006LiteWrapper

unknown-elf-gcc和unknown-linux-gnu的区别可以参考toolchain的readme：
https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md

一个可能不太准确的概括：你的SPECCPU是在Linux环境下运行的，你应该用riscv64-unknown-linux-gnu前缀的工具链。

YudeFang, 2024-03-09:
这两个我都试过了 elf和gnu。都不太行，编译出来的可执行文件大小跟正确的相比都小很多

发自我的iPhone

 

 
 https://github.com/OpenXiangShan/CPU2006LiteWrapper
 
 https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md

shinezyy, 2024-03-09:
你参考一下我们的示例项目里选项：
https://github.com/OpenXiangShan/CPU2006LiteWrapper/blob/33c53413dbfe604867e297201bfbb07f3abac005/Makefile.apps#L76

其中比较关键的是：
- 使用riscv64-unknown-linux-gnu工具链
- 静态链接SPECCPU：-static 

你说的 “编译出来的可执行文件大小跟正确的相比都小很多”，那是因为你没有静态链接

YudeFang, 2024-03-11:
> 你参考一下我们的示例项目里选项： https://github.com/OpenXiangShan/CPU2006LiteWrapper/blob/33c53413dbfe604867e297201bfbb07f3abac005/Makefile.apps#L76
> 
> 其中比较关键的是：
> 
> * 使用riscv64-unknown-linux-gnu工具链
> * 静态链接SPECCPU：-static
> 
> 你说的 “编译出来的可执行文件大小跟正确的相比都小很多”，那是因为你没有静态链接

已解决

shinezyy, 2024-03-11:
> > 你参考一下我们的示例项目里选项： https://github.com/OpenXiangShan/CPU2006LiteWrapper/blob/33c53413dbfe604867e297201bfbb07f3abac005/Makefile.apps#L76
> > 其中比较关键的是：
> > 
> > * 使用riscv64-unknown-linux-gnu工具链
> > * 静态链接SPECCPU：-static
> > 
> > 你说的 “编译出来的可执行文件大小跟正确的相比都小很多”，那是因为你没有静态链接
> 
> 已解决

是通过静态编译解决了吗？你的答案对社区很重要 ：）

YudeFang, 2024-03-12:
> > > 你参考一下我们的示例项目里选项： https://github.com/OpenXiangShan/CPU2006LiteWrapper/blob/33c53413dbfe604867e297201bfbb07f3abac005/Makefile.apps#L76
> > > 其中比较关键的是：
> > > 
> > > * 使用riscv64-unknown-linux-gnu工具链
> > > * 静态链接SPECCPU：-static
> > > 
> > > 你说的 “编译出来的可执行文件大小跟正确的相比都小很多”，那是因为你没有静态链接
> > 
> > 
> > 已解决
> 
> 是通过静态编译解决了吗？你的答案对社区很重要 ：）

已解决 以下是我的spec编译配置文件
####################################################################
# RISC-V Config file for SPEC CPU2006
# Last Update: 2015 Jan 12

# -------------------------------------------------------------

# This file is an agglomeration from the provided CPU2006 example configuration
# files (Example-simple.cfg and Example-linux64-amd64-gcc43+.cfg).

# Note: I make no promises as to its suitability for use in official
# submissions.
 
# Please see http://www.spec.org/cpu2006/Docs/config.html (also available in
# the Docs directory of your SPEC tree) for details on config file setup.  The
# config.html page has a list of all of the fields required for a full
# publication ofresults.
 
#####################################################################
# System Under Test (SUT) Section

# If it affects performance, you need to describe it, either in the
# pre-defined fields or by adding it to the notes section. Replace the
# setting below with the ones that match your machine.

#######################################################################
  
# Tester Description
test_sponsor       = Test Sponsor (Optional, defaults to hw_vendor)
tester             = (Optional, defaults to hw_vendor)

# System Description
hw_model           = Unknown HW Model
hw_memory          = 4 GB
hw_disk            = 1 1TB Mystery Disk
hw_vendor          = Berkeley Architecture Research
hw_other           = None
hw_avail           = Dec-9999

# CPU description
# See http://www.spec.org/cpu2006/Docs/runrules.html#cpucount
# for a discussion of these fields

hw_cpu_name        = Unknown RISC-V CPU 
hw_cpu_char        =
hw_cpu_mhz         = 1000
hw_fpu             = Integrated
hw_nchips          = 1
hw_ncores          = number of cores enabled
hw_ncoresperchip   = number of cores manufactured into each chip
hw_nthreadspercore = number of threads enabled per core
hw_ncpuorder       = 1,2 chips

# Cache description

hw_pcache          = 9999 MB I + 9999 MB D on chip per chip
hw_scache          = 9999 MB I+D on chip per chip
[... 225 lines cut ...]
exemd5=8d89493e0d4249c0a1c1daa87df9ad5a

473.astar=base=riscv=default:
# Last updated Sat Mar  9 13:28:38 2024
optmd5=ce29a646a94e8431d7e7f426443f946c
baggage=
compile_options=\
@eNq1UdFqgzAUffcrLnktmVBKH0pb0Jh12awJq0K7F7Ghk4ySDKNjn79oB1U65l52n8K9J+fecw7h\
W7EAv7GVf1Tar5SVH/MZPivdfOJSN7icTABLwAbMe62MtncGcLQTlOREZO6ZRDTMNr1eHrM0jWlO\
k4gFCQBgPgX8qg22daVkjYuzKqzSJbhGUSsJLab3XcxncKmlNU0lT2uPLIDs9yv0+6HoG8fDRy7S\
FRrcjTwn1PHfx8Fm52a3GjpESBPykN+AhqKQxy+LRMq27IU65JjGjlzw5zQImeM6DMmdZOTFLHka\
zeJPdv7sJizN8e0ka7uGa10Nas2Lo3GP/0N6t5pnbWa9wL4Ain3Faw==
exemd5=67984679c12303561373390a316632fb

999.specrand=base=riscv=default:
# Last updated Sat Mar  9 13:30:06 2024
optmd5=30f388ad0376e1a871dbfe87c88f3df6
baggage=
compile_options=\
@eNqtUdFqwjAUfe9XXPJ+LQzxQVSwaSeZ0YTZPmwvUoOWiCSjSWWfv7Rs1DFc97BDHsLNvSfnnEvF\
Rk4hblwdH7SJa+3UdTLGizbNO1amwUopwHAs2DevrXEjC5juZEb3VBbhuk2zpFhBAIoHwJOx6Hyt\
lcfyokunTQWhUHqt4BP9+J7LyfirDDNnm1odFxGdAqVz8rsq0rWJ5EnIfE6+SSRR8BToH/lytQtv\
P+WSSLTDMmcb9pqFliHpHaUUz/kyYZzlL7esnQsScbZdD0b5p5TuBDSzh/NRebeAHr3pNg6eDqf2\
/8a7j0XRbuFmBR9er7LR
exemd5=45653f5163121ec32b8c94848fffc546