OpenXiangShan/NEMU #276: 编译riscv64-nemu-interpreter支持pie问题

wuchangsheng, 2024-03-09:
编译步骤如下
make riscv64-xs_defconfig
make menuconfig
make -j

会报链接重定向错误，有两种方法修改，一是编译增加 -PIE 支持 pie，二是去掉pie编译选项。请问哪个修改更为合适，NEMU需要pie特性吗？

一，编译增加 -PIE 支持 pie
resource/softfloat/repo/build/Linux-x86_64-GCC/Makefile : COMPILE_C增加 -fPIE 编译选项
Makefile : CFLAGS_BUILD增加 -fPIE 编译选项

二，去掉pie编译选项
Makefile :82 LDFLAGS去掉-pie链接选项

xyyy1420, 2024-03-11:
尝试不修改，make clean-all之后重新编译

poemonsense, 2024-03-12:
> 尝试不修改，make clean-all之后重新编译

Maybe we can just rename clean-all to clean, clean to clean-shallow? Then outsider users won't experience these errors. To save time, the user can make clean-shallow only.

xyyy1420, 2024-03-12:
> > 尝试不修改，make clean-all之后重新编译
> 
> Maybe we can just rename clean-all to clean, clean to clean-shallow? Then outsider users won't experience these errors. To save time, the user can make clean-shallow only.

Or we just make `make clean` could delete softfloat?

poemonsense, 2024-03-12:
> > > 尝试不修改，make clean-all之后重新编译
> > 
> > 
> > Maybe we can just rename clean-all to clean, clean to clean-shallow? Then outsider users won't experience these errors. To save time, the user can make clean-shallow only.
> 
> Or we just make `make clean` could delete softfloat?

Good idea

closed 2024-03-15 by poemonsense