OpenXiangShan/NEMU #286: Potential memory access out of bound problem when printing $ISA-manual logo

Ruivolution-recruit, 2024-03-20:
In `src/isa/$ISA/logo.c` ($ISA is riscv32,riscv64 or mips32),`isa_logo[]` doesn't have a termination character to indicate the end of a string.

[src/isa/riscv64/logo.c](https://github.com/OpenXiangShan/NEMU/blob/master/src/isa/riscv64/logo.c)
[src/isa/riscv32/logo.c](https://github.com/OpenXiangShan/NEMU/blob/master/src/isa/riscv64/logo.c)
[src/isa/mips32/logo.c](https://github.com/OpenXiangShan/NEMU/blob/master/src/isa/riscv64/logo.c)

Meanwhile,in `src/engine/interpreter/hostcall.c`, there is a statement which access this `isa_logo[]` as a string at line 52:
```
 printf("\33[1;31mIf it is the first case, see\n%s\nfor more details.\n\nIf it is the second case, remember:\n"
      "* The machine is always right!\n"
      "* Every line of untested code is always wrong!\33[0m\n\n", isa_logo);
```
In the absence of '\0' at the end of `isa_logo`, memory access is not assured to be within legal bound. Hence, the program is under the risk of `global heap overflow` issue.

(There is  [a similar issue in NJU-ProjectN/NEMU](https://github.com/NJU-ProjectN/nemu/issues/18) and my English presentation and translation are both not very good, so I "quoted" a lot of statement from that one to express my thought)

cebarobot, 2024-03-20:
Yes, that seems to be a problem. You can open a pull request to fix it.

cebarobot, 2024-03-20:
fix by #287

closed 2024-03-20 by cebarobot