OpenXiangShan/NEMU #312: 关于根据cpu.pc取指疑问

wuchangsheng, 2024-04-17:
初始化时 cpu.pc = RESET_VECTOR; 设置为 0x80000000
实际运行是pc是很小的值，如Have taken checkpoint on pc 0x19490
根据pc取指流程也没有看见+baseaddr操作
请问从pc=0x19490是从虚拟地址多少的地方取指的

cebarobot, 2024-04-17:
请提供更详细的信息，例如您的工作负载、运行截图、使用的nemu提交编号等

wuchangsheng, 2024-04-18:
> 请提供更详细的信息，例如您的工作负载、运行截图、使用的nemu提交编号等

感谢回复，运行使用都没有问题，我是在学习nemu的代码流程，对pc值的使用有疑问

工作负载是对speccpu2017的541.leela_r进行切片
NEMU的commitid是fe61fbfb1b681fc11d3539b2cb8984344173093b。
日志截图如下：
[image]

cebarobot, 2024-04-18:
那么，0x19490 这个 PC 就是虚拟地址，Linux 上用户态程序的虚拟地址。

wuchangsheng, 2024-04-18:
> 那么，0x19490 这个 PC 就是虚拟地址，Linux 上用户态程序的虚拟地址。

制作切片是dump保存了0x100000000开始的8G内存，0x19490 地址的指令流不就没有保存，restore的时候如何根据 0x19490 找到执行指令的呢
另外一般 0x19490 地址，不是用户态能够访问的，没有做偏移处理嘛

cebarobot, 2024-04-18:
> 制作切片是dump保存了0x100000000开始的8G内存，0x19490 地址的指令流不就没有保存，restore的时候如何根据 0x19490 找到执行指令的呢

制作检查点是将 **物理地址 `0x80000000`** 开始的内存全部 dump 出来。内存（RAM）的**物理地址**总是从 `0x80000000` 开始的，大小根据 NEMU 的配置决定（默认是 8 GiB）。

`0x19490` 是**虚拟地址**，它对应的**物理地址**一定是在 `0x80000000` 开始的 8 GiB 之内。

> 另外一般 0x19490 地址，不是用户态能够访问的，没有做偏移处理嘛

**虚拟地址**到**物理地址**的转化过程由页表、TLB 等机制完成。您可参考 RISC-V 特权指令集手册、网络百科全书或相关的操作系统/计算机体系结构教科书。

closed 2024-04-24 by wuchangsheng