OpenXiangShan/NEMU #330: 对于vsstatus寄存器，在执行完浮点指令需要修改fs位域时(dirty_fs)，NEMU中没有做对应修改

prjmoriarty, 2024-05-13:
在使用NEMU作为difftest的模拟器进行仿真验证时，发现对于vsstatus寄存器中fs位域的更新逻辑，Xiangshan中进行了dirty_fs的处理，而NEMU中没有对应处理，导致vsstatus的值对比出错

prjmoriarty, 2024-05-13:
这是对应的仿真log
[image]
[image]
这是对应的指令流
[image]

prjmoriarty, 2024-05-13:
> 感谢您的报告，能否麻烦您提供一下香山和 NEMU 的 commit 信息，以及 workload 信息，以方便我们定位问题？

您好 对应的仿真log与指令段已提交评论，麻烦帮忙看一下 感谢

cebarobot, 2024-05-13:
请问这个 workload 是随机指令测试吗？还是实际的应用？

prjmoriarty, 2024-05-13:
> 请问这个 workload 是随机指令测试吗？还是实际的应用？

这个只是用开源工具RISCV-DV生成的随即指令测试，并不是真实的应用

closed 2024-05-13 by pxk27

cebarobot, 2024-05-13:
@prjmoriarty 我们应该修复了这个问题，您可以再测试看看。如果还有问题，请随时重开这个issue