OpenXiangShan/NEMU #353: When Excuting WFI, SPIKE and NEMU Lose Synchronization

cebarobot, 2024-06-03:
#### Description
When excuting `wfi` instruction, SPIKE and NEMU will lose synchoronization and cause difftest error like below:

```
[NEMU/include/cpu/difftest.h:48,difftest_check_reg] a1 is different after executing instruction at pc = 0x0000000080000004, right = 0x0000000000000000, wrong = 0x0000000020240000
[NEMU/include/cpu/difftest.h:48,difftest_check_reg] pc is different after executing instruction at pc = 0x0000000080000004, right = 0x0000000080000004, wrong = 0x0000000080000008
```

We could find that, NEMU has excuted `0x80000004` inst, but SPIKE haven't.

#### Reproduce
There is a simple assembly program to reproduce this problem.
```
wfi
li a1, 0x20240603
.word 0x6b # Hit Good Trap
```

Compile this program with:
```
riscv64-linux-gnu-gcc -c test.S
riscv64-linux-gnu-objcopy -O binary test.o test.bin
```
Then run using `riscv64-rvh-diff-spike_defconfig` config and difftest with `./ready-to-run/riscv64-spike-H-ext-so`:
```
make riscv64-rvh-diff-spike_defconfig
make -j
./build/riscv64-nemu-interpreter -d ./ready-to-run/riscv64-spike-H-ext-so -b ./test.bin
```

And NEMU will report the above error.

#### Expected Behaviour
The above program should simply hit good trap without any difftest error.

#### Additional Information
We found that, this issue was introduced by commit bee4a1c6, which bumped the spike-so in ready-to-run. It seems that spike has changed something, which led to this issue.

The spike commit should be:
* before this bump: [dc253ddb458b230a7cb855bfeea956d0f7399f6a](https://github.com/OpenXiangShan/riscv-isa-sim/commit/dc253ddb458b230a7cb855bfeea956d0f7399f6a)
* after this bump: [70ac58881f89a052c95e36f9312a9c223cc53609](https://github.com/OpenXiangShan/riscv-isa-sim/commit/70ac58881f89a052c95e36f9312a9c223cc53609)

cebarobot, 2024-06-03:
@poemonsense Do you have any idea about this?

poemonsense, 2024-06-03:
I have never tested the nemu-diff-with-spike. However, I have a general concern of the position of `wfi`. Can you have a try moving `wfi` to the second instruction?

For example,

```
nop
wfi
li a1, 0x20240603
.word 0x6b # Hit Good Trap
```

cebarobot, 2024-06-04:
I tried. But there is no difference.

This small program was extraced from a large program (xvisor). I think it is representative enough.

cebarobot, 2024-06-04:
It seems that I found the problem.

The previous workable spike-so is built from [pxk27/riscv-isa-sim](https://github.com/pxk27/riscv-isa-sim) and it removed some wfi logic at [this commit](https://github.com/pxk27/riscv-isa-sim/commit/b4eac2cab4d20fada37503a7aa48dc75f70390a2).

https://github.com/pxk27/riscv-isa-sim/blob/45436744fdbcc411bae7b66db424c90dfaf9555e/riscv/execute.cc#L274-L278

But current spike-so is built from [OpenXiangShan/riscv-isa-sim](https://github.com/OpenXiangShan/riscv-isa-sim), which keeps the wfi logic. 

This wfi code would throw an exception to skip pc+1 and quit step(), in order to handle interrupts outside. But for difftest ref so, there is no handler outside the step().

Now we have two way to solve this problem:
* let dut (XiangShan and NEMU) require to skip wfi instruction on ref side.
* remove wfi logic from spike step().

poemonsense, 2024-06-04:
> It seems that I found the problem.
> 
> The previous workable spike-so is built from [pxk27/riscv-isa-sim](https://github.com/pxk27/riscv-isa-sim) and it removed some wfi logic at [this commit](https://github.com/pxk27/riscv-isa-sim/commit/b4eac2cab4d20fada37503a7aa48dc75f70390a2).
> 
> ```
>       // debug mode wfis must nop
>       if (unlikely(in_wfi && !state.debug_mode)) {
>         throw wait_for_interrupt_t();
>       }
> ```
> 
> But current spike-so is built from [OpenXiangShan/riscv-isa-sim](https://github.com/OpenXiangShan/riscv-isa-sim), which keeps the wfi logic.
> 
> This wfi code would throw an exception, skip pc+1 then quit step(), in order to handle interrupts outside. But for difftest ref so, there is no handler outside the step().
> 
> Now we have two way to solve this problem:
> 
> * let dut (XiangShan and NEMU) require to skip wfi instruction on ref side.
> * remove wfi logic from spike step().

Use something like `#ifdef CONFIG_DIFFTEST`?

cebarobot, 2024-06-04:
> Use something like `#ifdef CONFIG_DIFFTEST`?

I agree. I will open a PR at spike repo.

closed 2024-06-04 by cebarobot