OpenXiangShan/NEMU #452: Error when make

Jeff-Smith-debug, 2024-08-15:
【Info】
when exec make under NEMU/, Error:
src/base/branch_trace.cpp:93:92: error: invalid conversion from 'const char*' to 'char;

cebarobot, 2024-08-16:
Please give more information. Such as the version of NEMU, the defconfig you used and the operation of compiling NEMU, etc.

According the information you provided, I cannot tell you what happened.

Jeff-Smith-debug, 2024-08-16:
Thanks for your kind reply. Sorry for my missing error info;
【config】
gcc version:  11.4.0
i git clone the repo yesterday;
【Error info】
Error info is below:
src/base/branch_trace.cpp:93:92: error: invalid conversion from 'const char*' to 'char*'
                    argv[i] = args[i].c_str()

【Solve】
I solve the error by modifying as belows:  and make is OK.
 //char *argv[128] = {nullptr};
  const char *argv[128];

//void init_monitor(int, char*[]);
void init_monitor(int, const char*[]);
     
I dont know that whether i am right?

cebarobot, 2024-08-16:
That's really strange. Currently, the master branch of NEMU does not have an file `src/base/branch_trace.cpp` and we cannot reproduce your problem. 

Could you please provide the branch and commit id of the NEMU you used?

Jeff-Smith-debug, 2024-08-16:
branch: master
I type the cmd:  git log. the most recent commit is 
commit: 66a92eed67e5fcb7c31f063521775ac7685c2625 (HEAD -> master ,origin/master ,origin/HEAS)
date : Mon July 22

is the infor which i give is right what you want.

cebarobot, 2024-08-16:
This is not a commmit of OpenXiangShan/NEMU repo. 

Which url did you use when cloning the NEMU repo? You may open the issue in an wrong place.

cebarobot, 2024-08-16:
Are you using https://github.com/yaozhicheng/NEMU ? This is a fork of OpenXiangShan/NEMU. You may open an issue there.

Jeff-Smith-debug, 2024-08-16:
yes,yes,yes, you are right. i git clone from above.
Sorry for my carefulless. I do a miss.

cebarobot, 2024-08-16:
That's ok. I will close this issue as the problem is solved.

closed 2024-08-16 by cebarobot