OpenXiangShan/NEMU #457: `make menuconfig` error

Donkey-Hao, 2024-08-17:
Hi all, I followed the `README.md` file to run `make menuconfig`, but I got this:
```bash
(base) ➜  NEMU git:(3033a69f) make menuconfig
/home/xs-env/NEMU/scripts/config.mk:20: Warning: .config does not exists!
/home/xs-env/NEMU/scripts/config.mk:21: To build the project, first run 'make menuconfig'.
+ CC confdata.c
make[1]: cc: No such file or directory
make[1]: *** [/home/xs-env/NEMU/scripts/build.mk:48: /home/xs-env/NEMU/tools/kconfig/build/obj-mconf/confdata.o] Error 127
make: *** [/home/xs-env/NEMU/scripts/config.mk:39: /home/xs-env/NEMU/tools/kconfig/build/mconf] Error 2
```

I have checked the `/home/xs-env/NEMU/Makefile`, I didn't find the `menuconfig` rule. How can I fix this problem to run XiangShan in NEMU?

cebarobot, 2024-08-17:
menuconfig is included in scripts/config.mk, which is included by the "main" Makefile file.

Also, Please make sure your repo is cloned compeletely and properly. There should be a file tools/kconfig/confdata.c.

Donkey-Hao, 2024-08-17:
Hi @cebarobot, thank you for replying. I found the `menuconfig` in `scripts/config.mk`. I  have the file `tools/kconfig/confdata.c`:
```bash
(base) ➜  NEMU git:(3033a69f) ls -l tools/kconfig
total 244
drwxr-xr-x 3 root root  4096 Aug 17 11:00 build
-rw-r--r-- 1 root root 16112 Aug 17 10:39 conf.c
-rw-r--r-- 1 root root 27837 Aug 17 10:39 confdata.c
-rw-r--r-- 1 root root 30726 Aug 17 10:39 expr.c
-rw-r--r-- 1 root root  9873 Aug 17 10:39 expr.h
-rw-r--r-- 1 root root  9320 Aug 17 10:39 lexer.l
-rw-r--r-- 1 root root  3749 Aug 17 10:39 list.h
-rw-r--r-- 1 root root  4413 Aug 17 10:39 lkc.h
-rw-r--r-- 1 root root  1929 Aug 17 10:39 lkc_proto.h
drwxr-xr-x 2 root root  4096 Aug 17 10:39 lxdialog
-rw-r--r-- 1 root root   706 Aug 17 10:39 Makefile
-rw-r--r-- 1 root root 28006 Aug 17 10:39 mconf.c
-rw-r--r-- 1 root root 22753 Aug 17 10:39 menu.c
-rw-r--r-- 1 root root 15694 Aug 17 10:39 parser.y
-rw-r--r-- 1 root root 11254 Aug 17 10:39 preprocess.c
-rw-r--r-- 1 root root 29322 Aug 17 10:39 symbol.c
-rw-r--r-- 1 root root  2210 Aug 17 10:39 util.c
```
Have I cloned completely and properly? I have run `git submodule update --init --recursive` in `xs-env`. I really have no idea about this mistake. In case I didn't clone the repo completely and properly, I cloned it again and got the same error.

cebarobot, 2024-08-17:
Could you please provide more information? 
- Which OS are you using? 
- Have you installed gcc?
- Could you just run `cc --version` and `gcc --version` and paste the output?

Donkey-Hao, 2024-08-17:
Hi @cebarobot, thanks for your help, I have figured it out. Before that, the GCC version is 9:
```
(base) ➜  XiangShan git:(8fae59bba) gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.5.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-5Q4PKF/gcc-9-9.5.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
gcc version 9.5.0 (Ubuntu 9.5.0-1ubuntu1~22.04)
```
I run `apt install build-essential`  updated it to 11. I run `make  menuconfig` again and every thing goes well (succeeded!). 
I am not sure whether the version of GCC concerns of this mistake, because of:
```
The following NEW packages will be installed:
  build-essential g++ g++-11 gcc gcc-11
```
So I am not sure which packages works. Anyway, thank you very much for help me.

Hao

closed 2024-08-17 by Donkey-Hao

cebarobot, 2024-08-18:
I think the gcc version is just the key to the issue. Before, you just have gcc-9 installed and there is no alias of `cc` (acctually the default c compiler), which is used by make menuconfig. After, you installed build-essential (the default compilation toolchains of Ubuntu 22.04), so there is an alias of `cc`.