OpenXiangShan/NEMU #482: NEMU载入的BBL过大时出现的out of memory问题

FoolgryGamer, 2024-08-30:
```
[  475.490000] Node 0 active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:1113780kB isolated(anon):0kB isolated(file):0kB mapped:0kB dirty:0kB writeback:0kB shmem:0kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no^M
[  475.490000] Normal free:4152kB min:4236kB low:5356kB high:6476kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:1113780kB writepending:0kB present:2095104kB managed:1122420kB mlocked:0kB kernel_stack:144kB pagetables:0kB bounce:0kB free_pcp:216kB local_pcp:216kB free_cma:0kB^M
[  475.490000] lowmem_reserve[]: 0 0 0^M
[  475.490000] Normal: 2*4kB (H) 0*8kB 1*16kB (H) 1*32kB (H) 2*64kB (H) 1*128kB (H) 1*256kB (H) 3*512kB (UH) 2*1024kB (H) 0*2048kB 0*4096kB = 4152kB^M
[  475.490000] 278452 total pagecache pages^M
[  475.490000] 0 pages in swap cache^M
[  475.490000] Swap cache stats: add 0, delete 0, find 0/0^M
[  475.490000] Free swap  = 0kB^M
[  475.490000] Total swap = 0kB^M
[  475.490000] 523776 pages RAM^M
[  475.490000] 0 pages HighMem/MovableOnly^M
[  475.490000] 243171 pages reserved^M
[  475.490000] [ pid ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name^M
[  475.490000] Kernel panic - not syncing: Out of memory and no killable processes...^M
```
报错的上下文如上所示：采用的BBL大小为1.7G，NEMU里默认的RAM大小为2G，按原理来说应该可以load进内存里，但是执行过程中出现了out of memory文件。

> 想咨询一下如何拓展NEMU的RAM大小？

尝试更改了riscv-pk中的dts/noop.dtsi中memory部分的值如下，没有效果。
```
    L11: memory@100000000 {
        device_type = "memory";
        reg = <0x0 0x80000000 0x0 0x100000000>;
    };

```

cebarobot, 2024-08-30:
NEMU 的 menuconfig 里面有内存大小的配置，您需要在那里进行修改。请运行 `make menuconfig`。

dts 中的配置影响的仅是软件，并不会影响 NEMU 本身的配置。

FoolgryGamer, 2024-08-30:
***@***.***

cebarobot, 2024-08-30:
> ***@***.***

？这是什么？

closed 2024-11-15 by FoolgryGamer