OpenXiangShan/NEMU #505: Compile Error

figuresudo, 2024-09-05:
I encountered an issue when trying to compile the NEMU source code from the master branch. I executed the following command:

make riscv64-xs_defconfig
However, I received the following error messages:

~/pathfuzz/NEMU$ make riscv64-xs_defconfig
正克隆到 'resource/softfloat/repo'...
remote: Enumerating objects: 451, done.
remote: Counting objects: 100% (451/451), done.
remote: Compressing objects: 100% (370/370), done.
remote: Total 451 (delta 412), reused 95 (delta 80), pack-reused 0 (from 0)
接收对象中: 100% (451/451), 203.65 KiB | 1.07 MiB/s, 完成.
处理 delta 中: 100% (412/412), 完成.
正克隆到 'resource/gcpt_restore'...
remote: Enumerating objects: 145, done.
remote: Counting objects: 100% (145/145), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 145 (delta 84), reused 135 (delta 79), pack-reused 0 (from 0)
接收对象中: 100% (145/145), 39.70 KiB | 677.00 KiB/s, 完成.
处理 delta 中: 100% (84/84), 完成.
+ AS src/restore.S
src/restore.S: Assembler messages:
src/restore.S:91: 错误： unrecognized opcode `vsetvl t2,t2,t1'
src/restore.S:91: 错误： unrecognized opcode `vl1re64.v v0,(sp)'
[...additional error messages...]
make[1]: *** [Makefile:57：/home/NEMU/resource/gcpt_restore/build/obj/src/restore.o] 错误 1
make: *** [/home/NEMU/scripts/config.mk:45：resource/gcpt_restore/build/gcpt.bin] 错误 2

Could you please advise on how to resolve this issue?

cebarobot, 2024-09-05:
Your riscv64 cross compiler on your system is a little old. Please follow [this document](https://docs.xiangshan.cc/zh-cn/latest/workloads/toolchain/) to install newer toolchain.

figuresudo, 2024-09-05:
I have updated the riscv64-unknown-linux-gnu toolchain to version 13.2.0, and the make riscv64-xs_defconfig command now runs successfully. However, I still encounter errors when executing make:

+ CC src/isa/riscv64/instr/rvv/vldst_impl.c
src/isa/riscv64/instr/rvv/vldst_impl.c: In function ‘vld’:
src/isa/riscv64/instr/rvv/vldst_impl.c:265:3: error: ‘__attribute_maybe_unused__’ undeclared (first use in this function)
  265 |   __attribute_maybe_unused__ bool cross_page = last_addr / PAGE_SIZE != start_addr / PAGE_SIZE;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/isa/riscv64/instr/rvv/vldst_impl.c:265:3: note: each undeclared identifier is reported only once for each function it appears in
src/isa/riscv64/instr/rvv/vldst_impl.c:265:29: error: expected ‘;’ before ‘_Bool’
  265 |   __attribute_maybe_unused__ bool cross_page = last_addr / PAGE_SIZE != start_addr / PAGE_SIZE;
      |                             ^
      |                             ;
src/isa/riscv64/instr/rvv/vldst_impl.c:271:81: error: ‘cross_page’ undeclared (first use in this function)
  271 |   if (is_unit_stride && nf == 1 && vl_val > vstart->val && vtype->vlmul < 4 && !cross_page) {
      |                                                                                 ^~~~~~~~~~
src/isa/riscv64/instr/rvv/vldst_impl.c:284:33: error: expected ‘;’ before ‘unsigned’
  284 |       __attribute_maybe_unused__ unsigned count = gen_mask_for_unit_stride(s, eew, vstart, vl_val, masks);
      |                                 ^~~~~~~~~
      |                                 ;
src/isa/riscv64/instr/rvv/vldst_impl.c: In function ‘vst’:
src/isa/riscv64/instr/rvv/vldst_impl.c:501:3: error: ‘__attribute_maybe_unused__’ undeclared (first use in this function)
  501 |   __attribute_maybe_unused__ bool cross_page = last_addr / PAGE_SIZE != start_addr / PAGE_SIZE;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/isa/riscv64/instr/rvv/vldst_impl.c:501:29: error: expected ‘;’ before ‘_Bool’
  501 |   __attribute_maybe_unused__ bool cross_page = last_addr / PAGE_SIZE != start_addr / PAGE_SIZE;
      |                             ^
      |                             ;
src/isa/riscv64/instr/rvv/vldst_impl.c:504:81: error: ‘cross_page’ undeclared (first use in this function)
  504 |   if (is_unit_stride && nf == 1 && vl_val > vstart->val && vtype->vlmul < 4 && !cross_page) {
      |                                                                                 ^~~~~~~~~~
src/isa/riscv64/instr/rvv/vldst_impl.c: In function ‘vldff’:
src/isa/riscv64/instr/rvv/vldst_impl.c:841:3: error: ‘__attribute_maybe_unused__’ undeclared (first use in this function)
  841 |   __attribute_maybe_unused__ bool cross_page = last_addr / PAGE_SIZE != start_addr / PAGE_SIZE;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/isa/riscv64/instr/rvv/vldst_impl.c:841:29: error: expected ‘;’ before ‘_Bool’
  841 |   __attribute_maybe_unused__ bool cross_page = last_addr / PAGE_SIZE != start_addr / PAGE_SIZE;
      |                             ^
      |                             ;
src/isa/riscv64/instr/rvv/vldst_impl.c:847:81: error: ‘cross_page’ undeclared (first use in this function)
  847 |   if (is_unit_stride && nf == 1 && vl_val > vstart->val && vtype->vlmul < 4 && !cross_page) {
      |                                                                                 ^~~~~~~~~~
src/isa/riscv64/instr/rvv/vldst_impl.c:860:33: error: expected ‘;’ before ‘unsigned’
  860 |       __attribute_maybe_unused__ unsigned count = gen_mask_for_unit_stride(s, eew, vstart, vl_val, masks);
      |                                 ^~~~~~~~~
      |                                 ;
make: *** [/home/NEMU/scripts/build.mk:60：/home/NEMU/build/obj-riscv64-nemu-interpreter/src/isa/riscv64/instr/rvv/vldst_impl.o] 错误 1

It appears that __attribute_maybe_unused__ is not defined. Could you please advise on how to resolve this issue?

cebarobot, 2024-09-05:
What is the version of your host gcc compiler and your OS? Please run `gcc --version` and paste the result.

figuresudo, 2024-09-05:
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cebarobot, 2024-09-05:
The key problem is that `__attribute_maybe_unused__` is used in C, which is a feature from C++17. That's our problem, we will fix it soon.

By the way, Your host OS and gcc may also be too old. We now recommend using Ubuntu 22.04 or later as well as gcc-11 or later.

figuresudo, 2024-09-05:
ok, thanks a lot.

cebarobot, 2024-09-05:
fixed in #506 , you could try again.

figuresudo, 2024-09-06:
It successfully works. Thanks.

closed 2024-09-06 by figuresudo