OpenXiangShan/NEMU #517: [Bug Report] `mie` register not implemented correctly

Donkey-Hao, 2024-09-11:
According to the privileged [specification](https://riscv.org/technical/specifications/) in chapter **3.1.9. Machine Interrupt Registers**, some bits in MIE are set to zero. 

[image]

However, if we write `0xffffffff` to `mie`, the NEMU will set it to `0xffffffff`. 

[image]

The bug was found when I tested Xiangshan. This is the test program [mie_bug.zip](https://github.com/user-attachments/files/16960329/mie_bug.zip).

Here are some env information:

**NEMU commit:** 3033a69f,
**xs-env commit:** master,
**Xiangshan commit:** 8fae59bba.

huxuan0307, 2024-09-12:
Thank you for your bug reporting. I'm handling this.

Donkey-Hao, 2024-10-10:
Hi @huxuan0307,
I have tested this bug in the newest version NEMU, commit id is `821ea961`. I changed some code in order to run `./build/riscv64-nemu-interpreter -d ./ready-to-run/riscv64-spike-so -b mie.bin`.
[image]

Actually, I doubt the reference spike is not correct because I ran the code in spike, the 1.1.1-dev version, the right value is `mie = 0x0000000000000aaa` after we write it. 
[image]

Is there a way to run NEMU like spike step by step? I have not found it. I want to help to improve NEMU.

sinceforYy, 2024-10-11:
In `src/isa/riscv64/system/priv.c`, you can look at the source code for writing the mie register. We now support the `RV_H` and `RV_SSCOFPMF` extensions, so the corresponding bits of mie are readable and writable. 

We made a modification based on Spike, you can get it in https://github.com/OpenXiangShan/riscv-isa-sim. If you want to modify and generate spike-so that diff with NEMU, you can read `README` in `riscv-isa-sim/difftest/`

Besides, You can remove the -b option and run `./build/riscv64-nemu-interpreter -d ./ready-to-run/riscv64-spike-so mie.bin`. Execute `help` to view options.

Donkey-Hao, 2024-10-29:
Hi @sinceforYy, thanks for your reply.
I am sorry for confused you, this problem is not about `the corresponding bits of mie are readable and writable`. No matter what extensions Xiangshan supports, some bits in `mie` must be zero, such as `mie.{0, 2, 4, 6, 8}`. 
[image]
As you can see in my initial comment: `mie = 0xffffffff`, which means  `mie.{0, 2, 4, 6, 8}` is not zero. Here is my question,  `mie.{0, 2, 4, 6, 8}` === 0, is that right? Please correct me if I am wrong. 
Sincerely.

Donkey-Hao, 2024-10-31:
Thanks for your reply @sinceforYy.

I have RTFM in **chapter 17 & 18.4**, NEMU `821ea961` version is right, but [3033a69](https://github.com/OpenXiangShan/NEMU/commit/3033a69fb8cdd9939ebb43315f86f38ab97346fe) was wrong, as we can see the picture in my first comment:
[image]
Yes, xiangshan supports the RV_H and RV_SSCOFPMF extensions, so NEMU `821ea961` version is right when we write 0xffffffff to `mie`:

>C17: Note that the new count overflow interrupt will be treated as a standard local interrupt that is assigned to bit 13 in the mip/mie/sip/sie registers.
[image]

This bug had been fixed.

closed 2024-10-31 by Donkey-Hao