OpenXiangShan/NEMU #555: nemu 在启动 gcpt.bin 进行 profile 时触发 riscv-pk 的断言后 Power off

rmxpf, 2024-09-27:
**我的环境**：

- NEMU：master 15a64d79885fd6038e862d8e96119b1f814ee765
- riscv-linux：nanshan 655055af981b490cb6a12353a5bb846d2be79c6f
- riscv-pk：noop a621e1620a1ab050d81f53743253e8eb5cf1b24e
- riscv-rootfs：checkpoint 7b6cf0ccdfdb103d2e5918712036634d4e973552
- riscv-gnu-toolchain：15.0.0

------

> 以前可用的 NEMU：f63a985f00b2c3f310717468455147b2c5ab7bdb
> 现在遇到 error 的 NEMU：15a64d79885fd6038e862d8e96119b1f814ee765

**我遇到的问题**：nemu 在启动 gcpt.bin 进行 profile 时触发 riscv-pk 的断言后 Power off

**我想要的**：我想用最新的 NEMU 生成 checkpoint（rv64gcbv）

**我观察到的现象**：同一个 gcpt.bin 以前的 NEMU（15a64d79885fd6038e862d8e96119b1f814ee765） 可以正常启动，但是最新的 NEMU（f63a985f00b2c3f310717468455147b2c5ab7bdb） 会触发 riscv-pk 的断言（assert((~read_csr(medeleg) & mandatorily_delegable_exceptions) == 0);）具体如下所示：

```bash
(base) # hebo @ node028 in ~/test [15:50:13]
$ $NEMU_HOME/build/riscv64-nemu-interpreter $NEMU_HOME/resource/gcpt_restore/build/gcpt.bin -D test_gcpt.bin -w test.bbl -C profiling-0 -b --simpoint-profile --cpt-interval 20000000
[src/monitor/monitor.c:195,parse_args] Doing Simpoint Profiling
[src/checkpoint/path_manager.cpp:55,init] Cpt id: -1
[src/checkpoint/path_manager.cpp:73,setSimpointProfilingOutputDir] Created test_gcpt.bin/profiling-0/xalancbmk.bbl/

[src/checkpoint/simpoint.cpp:84,init] Doing simpoint profiling with interval 20000000
[src/checkpoint/serializer.cpp:378,next_index] set next index 0
[src/checkpoint/path_manager.cpp:92,setCheckpointingOutputDir] donot set checkpoint path without Checkpoint mode
[src/isa/riscv64/init.c:218,init_isa] NEMU will start from pc 0x80000000
[src/monitor/image_loader.c:203,load_img] Loading image (checkpoint/bare metal app/bbl) form cmdline: /nfs/home/hebo/TOOL/Simpoint_Checkpoint/NEMU/resource/gcpt_restore/build/gcpt.bin

[src/monitor/image_loader.c:258,load_img] Read 19744408 bytes from file /nfs/home/hebo/TOOL/Simpoint_Checkpoint/NEMU/resource/gcpt_restore/build/gcpt.bin to 0x80000000
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'uartlite' at [0x00000000000003f8, 0x0000000000000404]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'rtc' at [0x0000000000000048, 0x000000000000004f]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'screen' at [0x0000000000000100, 0x0000000000000107]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'keyboard' at [0x0000000000000060, 0x0000000000000063]
[src/device/sdcard.c:137,init_sdcard] Can not find sdcard image: 
[src/monitor/monitor.c:58,welcome] Debug: OFF
[src/monitor/monitor.c:63,welcome] Build time: 19:50:45, Sep 26 2024
Welcome to riscv64-NEMU!
For help, type "help"
/nfs/home/hebo/TOOL/Simpoint_Checkpoint/riscv-pk/machine/minit.c:96: assertion failed: (~read_csr(medeleg) & mandatorily_delegable_exceptions) == 0
Power off
^C
(base) # hebo @ node028 in ~/test [15:51:53]
```

使用以前的 NEMU（f63a985f00b2c3f310717468455147b2c5ab7bdb）正常启动 gcpt.bin 的 log：

```bash
(base) # hebo @ node028 in ~/test [15:52:48]
$ /nfs/home/hebo/TOOL/Simpoint_Checkpoint-RVV/NEMU/build/riscv64-nemu-interpreter $NEMU_HOME/resource/gcpt_restore/build/gcpt.bin -D test_gcpt.bin -w test.bbl -C profiling-0 -b --simpoint-profile --cpt-interval 20000000
[src/monitor/monitor.c:195,parse_args] Doing Simpoint Profiling
[src/checkpoint/path_manager.cpp:54,init] Cpt id: -1
[src/checkpoint/path_manager.cpp:72,setSimpointProfilingOutputDir] Created test_gcpt.bin/profiling-0/test.bbl/

[src/checkpoint/simpoint.cpp:83,init] Doing simpoint profiling with interval 20000000
[src/checkpoint/serializer.cpp:378,next_index] set next index 0
[src/checkpoint/path_manager.cpp:91,setCheckpointingOutputDir] donot set checkpoint path without Checkpoint mode
[src/isa/riscv64/init.c:191,init_isa] NEMU will start from pc 0x80000000
[src/monitor/image_loader.c:203,load_img] Loading image (checkpoint/bare metal app/bbl) form cmdline: /nfs/home/hebo/TOOL/Simpoint_Checkpoint/NEMU/resource/gcpt_restore/build/gcpt.bin

[src/monitor/image_loader.c:258,load_img] Read 19744408 bytes from file /nfs/home/hebo/TOOL/Simpoint_Checkpoint/NEMU/resource/gcpt_restore/build/gcpt.bin to 0x80000000
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'uartlite' at [0x00000000000003f8, 0x0000000000000404]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'rtc' at [0x0000000000000048, 0x000000000000004f]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'screen' at [0x0000000000000100, 0x0000000000000107]
[src/device/io/port-io.c:30,add_pio_map] Add port-io map 'keyboard' at [0x0000000000000060, 0x0000000000000063]
[src/device/sdcard.c:137,init_sdcard] Can not find sdcard image: 
[src/monitor/monitor.c:57,welcome] Debug: OFF
[... 344 lines cut ...]
+file /spec/foreman_qcif.yuv /nfs/home/hebo/TOOL/Simpoint_Checkpoint-RVV/auto_checkpoint/cpu2006_run_dir/h264ref/foreman_qcif.yuv 755 0 0
diff --git a/rootfsimg/run.sh b/rootfsimg/run.sh
index 07ecc43..bedfa34 100644
--- a/rootfsimg/run.sh
+++ b/rootfsimg/run.sh
@@ -1,13 +1,13 @@
 #!/bin/sh
 echo '===== Start running SPEC2006 ====='
-echo '======== BEGIN mcf ========'
+echo '======== BEGIN h264ref_foreman.baseline ========'
 set -x
-md5sum /spec/astar_base
+md5sum /spec/h264ref_base.riscv64-linux-gnu-gcc12.2.0
 date -R
 /spec_common/before_workload
-cd /spec && ./astar_base BigLakes2048.cfg
+cd /spec && ./h264ref_base.riscv64-linux-gnu-gcc12.2.0 -d foreman_ref_encoder_baseline.cfg > null
+/spec_common/trap
 date -R
 set +x
-/spec_common/trap
-echo '======== END   mcf ========'
+echo '======== END   h264ref_foreman.baseline ========'
 echo '===== Finish running SPEC2006 ====='
```

cebarobot, 2024-10-14:
看起来是 riscv-pk 的 `medeleg` 处理可能存在一些问题，类似的问题在加入 H 扩展的时候也出现过。

考虑到 riscv-pk 已经逐渐被上游抛弃，我们强烈建议切换到 OpenSBI。请参照香山文档中的指引来构建基于 OpenSBI 的工作负载。

rmxpf, 2024-10-14:
> 看起来是 riscv-pk 的 `medeleg` 处理可能存在一些问题，类似的问题在加入 H 扩展的时候也出现过。
> 
> 考虑到 riscv-pk 已经逐渐被上游抛弃，我们强烈建议切换到 OpenSBI。请参照香山文档中的指引来构建基于 OpenSBI 的工作负载。

感谢您的解答，我会尝试用 OpenSBI 来构建我的工作负载。

closed 2024-10-14 by rmxpf