OpenXiangShan/NEMU #594: [Bug report] NEMU does not handle illegal instructions correctly

Donkey-Hao, 2024-10-22:
Run the bare metal code [stuck.zip](https://github.com/user-attachments/files/17472690/stuck.zip) in NEMU, NEMU will abort after running about 450k instructions. As shown in the picture, it aborted at PC = 0x800001c8, actually it is an illegal instruction, but NEMU did not handle this trap correctly.
`800001c8:	0000700b          	.insn	4, 0x700b`

Run nemu difftest with spike:
[image]

Run nemu only:
[image]

Actually, we have set the handler in the code, NEMU should handle the trap. The wrong `mtval` value (compare with spike) may be related to this issue #584.

NEMU: master, commit id: 821ea961
spike version: 1.1.1-dev
make cmd: 
* `make riscv64-xs-diff-spike_defconfig; make -j64`, 
* `make riscv64-xs_defconfig; make -j64`

NewPaulWalker, 2024-10-23:
This is because both of these xx_defconfig files have set `CONFIG_REPORT_ILLEGAL_INSTR=y`, which means that NEMU will report the issue when it encounters an illegal instruction caused by an invalid instruction, instead of raising an illegal instruction and jumping to the corresponding trap handler to continue execution.
You can take a look at this line of [code](https://github.com/OpenXiangShan/NEMU/blob/master/src/engine/interpreter/hostcall.c#L34)

Donkey-Hao, 2024-10-24:
Hi @NewPaulWalker,
Thanks for your reply. Actually, those configs do not work:
[image]
NEMU still ABORT when it encounters invalid opcodes. I have no idea why it still aborts.

Thank you again.

Tang-Haojin, 2024-10-24:
Have you run `make riscv64-xs_defconfig` again after commenting `CONFIG_REPORT_ILLEGAL_INSTR=y` out?

Donkey-Hao, 2024-10-24:
> Have you run `make riscv64-xs_defconfig` again after commenting `CONFIG_REPORT_ILLEGAL_INSTR=y` out?

I commented it in `riscv64-xs-diff-spike_defconfig` before. 
I commented it just now when I saw your reply, I tried it again, still ABORT.

cebarobot, 2024-10-24:
Please follow this procedure to try again: 
```
# 1. Apply default config
make riscv64-xs_defconfig

# 2. Using menuconfig to change the config
# Miscellaneous  --->  [ ] Let NEMU report illegal instruction
make menuconfig

# 3. Build NEMU again
make -j
```

Just commented the `CONFIG_REPORT_ILLEGAL_INSTR=y` make no sense. If you want to modify the default config file, you should:
```diff
- CONFIG_REPORT_ILLEGAL_INSTR=y
+ # CONFIG_REPORT_ILLEGAL_INSTR is not set
```

`is not set` is very important is you want to disable an config in Kconfig.

Donkey-Hao, 2024-10-24:
> Please follow this procedure to try again:
>
> ```
> # 1. Apply default config
> make riscv64-xs_defconfig
>
> # 2. Using menuconfig to change the config
> # Miscellaneous  --->  [ ] Let NEMU report illegal [...]

This works! Thank you!

cebarobot, 2024-10-24:
I have tried your workload. It seems that it get into an loop of trap. The workload traps at
- `0x800001c0` for illegal instruction: trying to access `mepc` in S mode.
- `0x800001c8` for illegal instruction: unknown instruction `.insn 4, 0x700b`
- `0x80000358` for Environment call from U-mode or VU-mode: `ecall` in `finish:`

closed 2024-10-24 by Donkey-Hao

NewPaulWalker, 2024-10-24:
> > Have you run `make riscv64-xs_defconfig` again after commenting `CONFIG_REPORT_ILLEGAL_INSTR=y` out?
> 
> I commented it in `riscv64-xs-diff-spike_defconfig` before. I commented it just now when I saw your reply, I tried it again, still ABORT.

The reason for the abort after commenting out `CONFIG_REPORT_ILLEGAL_INSTR=y` is that the configuration `REPORT_ILLEGAL_INSTR` has a default value of `y` in the [Kconfig ](https://github.com/OpenXiangShan/NEMU/blob/master/Kconfig#L351)file, so `CONFIG_REPORT_ILLEGAL_INSTR` is still defined.

In addition to setting it in `make menuconfig`, you can also directly modify the corresponding defconfig file.

``` diff
- CONFIG_REPORT_ILLEGAL_INSTR=y
+ # CONFIG_REPORT_ILLEGAL_INSTR is not set
```
or
```diff
- CONFIG_REPORT_ILLEGAL_INSTR=y
+ CONFIG_REPORT_ILLEGAL_INSTR=n
```

Either way works.

Donkey-Hao, 2024-10-25:
Hi @NewPaulWalker，
I really appreciate it, thanks for your patient guide.