OpenXiangShan/NEMU #611: NEMU error  ；riscv64-unknown-linux-gnu-gcc: No such file or directory

yaoyuexiaogege, 2024-10-29:
[image]
在香山核仿真程序上仿真运行 workload过程中
进行：
source ./env.sh
make clean
make riscv64-xs-ref_defconfig
出现以下报错：

AS src/restore.S
make[1]: riscv64-unknown-linux-gnu-gcc: No such file or directory
make[1]: *** [Makefile:57: /home/xiaoyuge/Desktop/test1/xs-env/NEMU/resource/gcpt_restore/build/obj/src/restore.o] Error 127
make: *** [/home/xiaoyuge/Desktop/test1/xs-env/NEMU/scripts/config.mk:45: resource/gcpt_restore/build/gcpt.bin] Error 2
是什么原因？

yaoyuexiaogege, 2024-10-29:
> 此前已经在 [OpenXiangShan/XiangShan#3797](https://github.com/OpenXiangShan/XiangShan/issues/3797) 作了回复，如下：
> 
> 对于您遇到的问题，您的机器上缺少 `riscv64-unknown-linux-gnu-gcc` 交叉编译器，您需要安装 RISC-V 交叉编译工具链，请参考这个文档：[编译工具链 toolchains - XiangShan 官方文档](https://docs.xiangshan.cc/zh-cn/latest/workloads/toolchain/)
> 
> By the way, for your issue, your machine lacks `riscv64-unknown-linux-gnu-gcc` cross compiler. You need install risc-v cross-compiling toolchain and refer to [toolchains - XiangShan Document](https://docs.xiangshan.cc/zh-cn/latest/workloads/toolchain.en/)

好的，谢谢

yaoyuexiaogege, 2024-10-29:
> 此前已经在 [OpenXiangShan/XiangShan#3797](https://github.com/OpenXiangShan/XiangShan/issues/3797) 作了回复，如下：
>
> 对于您遇到的问题，您的机器上缺少 `riscv64-unknown-linux-gnu-gcc` 交叉编译器，您需要安装 RISC-V 交叉编译工具链，请参考这个文档：[编译工具链 [...]

已经按照文档说明安装了 交叉编译器，但重新执行后也是同样的报错，在AM中也是如此，报错如下：
# Building coremark [riscv64-xs] with AM_HOME {/home/xiaoyuge/Desktop/test1/xs-env/nexus-am}
+ CC src/core_util.c
make: riscv64-unknown-linux-gnu-gcc: No such file or directory
make: *** [/home/xiaoyuge/Desktop/test1/xs-env/nexus-am/Makefile.compile:27: /home/xiaoyuge/Desktop/test1/xs-env/nexus-am/apps/coremark/build/riscv64-xs//./src/core_util.o] Error 127
如下图：
[image]

Donkey-Hao, 2024-10-29:
从图中判断你使用 `apt` 安装了 `gcc-riscv64-linux-gnu`，如果你想使用它进行 `make`，那么你应该更改 `makefile` 中的 `riscv64-unknown-linux-gnu-gcc` 为 `gcc-riscv64-linux-gnu`。如果不了解 `makefile` 或者想要求稳，建议您从头在本地构建 [toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain)。如果之后和 RISC-V 经常打交道，建议重头构建。

cebarobot, 2024-10-29:
请根据文档中“获取预编译好的夜间版本”一节进行安装。如果已经按照这一节安装完成了，请检查 PATH 是否设置合理。你可以在互联网搜索如何正确配置 linux 的 PATH。

closed 2024-11-12 by NewPaulWalker