OpenXiangShan/NEMU #617: spec2017 521用例陷入死循环

hongwuya, 2024-10-29:
现象：linux kernel日志中未打印 random: crng init done
我认为是crng未初始化成功，导致后续调用getrandom时陷入死循环。
请问要做哪些修改？
riscv-linux版本：https://github.com/OpenXiangShan/riscv-linux nanshan分支
risck-pk版本:https://github.com/OpenXiangShan/riscv-pk noop分支
riscv-rootfs版本:https://github.com/OpenXiangShan/riscv-rootfs master 分支
问题用例:spec2017 521wrf用例

cebarobot, 2024-10-29:
您提供的信息不足以进行判断，请按照 issue 模板提交 bug 信息。

hongwuya, 2024-10-30:
问题描述：我使用nemu 跑spec2017 521用例的workload，执行profiling的过程中，发现程序一直陷入死循环，无法终止。
workload的生成过程参考的XiangShan 官方文档，bbl+linux+application
riscv-linux：https://github.com/OpenXiangShan/riscv-linux nanshan分支
risck-pk版本:https://github.com/OpenXiangShan/riscv-pk noop分支
riscv-rootfs版本:https://github.com/OpenXiangShan/riscv-rootfs master 分支
用例的bin文件过大，暂不支持上传
生成的profiling log见附件521_err.txt
[521_err.txt](https://github.com/user-attachments/files/17565259/521_err.txt)

cebarobot, 2024-10-30:
根据初步判断，这不是 NEMU 或香山相关的硬件实现问题，而是系统软件的配置问题，我们无法提供技术支持。请使用搜索引擎或生成式大语言模型，亦或是向相关软件社区咨询。

closed 2024-10-30 by cebarobot