OpenXiangShan/NEMU #627: Some M mode registers error.

Donkey-Hao, 2024-11-02:
When I tested xiangshan with difftest, it reported error:
```
privilegeMode: 3
mstatus different at pc = 0x00800e9cc0, right= 0x8000040a00007900, wrong = 0x8000000a00007900
   mepc different at pc = 0x00800e9cc0, right= 0x00000000800e9cdc, wrong = 0x0000000000000000
  mtval different at pc = 0x00800e9cc0, right= 0x00000000a2532053, wrong = 0x0000000000000000
 mcause different at pc = 0x00800e9cc0, right= 0x0000000000000002, wrong = 0x0000000000000000
Core 0: ABORT at pc = 0x80012b2e
Core-0 instrCnt = 511, cycleCnt = 4,393, IPC = 0.116321
Seed=0 Guest cycle spent: 4,396 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 9,774ms
```
Registers `mstatus, mepc, mtval, mcause` got wrong value. I have checked it in spike, xiangshan is right and NEMU is wrong. Please check the log file for more infomation.

**workload**
[34.zip](https://github.com/user-attachments/files/17605409/34.zip)
I have a lot of workloads triggered this error, the 34 is one of them.

**env**
* xiangshan: d9c759412
* NEMU: 2f5e6ed7
* spike: 1.1.1-dev

Donkey-Hao, 2024-11-02:
BTW: run spike with: `spike --pc=0x80000000 --isa=rv64gc -d xxx.elf`
Here is another test case: [test.zip](https://github.com/user-attachments/files/17605542/test.zip)

NewPaulWalker, 2024-11-02:
This issue was caused by an incorrect change to the Zfa decoding logic in https://github.com/OpenXiangShan/NEMU/pull/601, and it has been fixed in https://github.com/OpenXiangShan/NEMU/pull/620.

By the way, when encountering issues, you can try rebase the latest master to check if the problem has already been fixed.

closed 2024-11-02 by Donkey-Hao