OpenXiangShan/NEMU #663: [Bug report] debug register `tinfo` is read-only.

ha0lyu, 2024-11-17:
When we access `tinfo` register, NEMU will return some different value,
According to the [debug manual](https://riscv.org/wp-content/uploads/2019/03/riscv-debug-release.pdf), `tinfo` is **read-only.**
[image]

**But if we write it, it changed in NEMU:**
[image]
_Before write tinfo, it's value is 0x40, same as XiangShan._

[image]

**Test case**:
```
    csrr t0, tinfo
    csrrwi  s3, tinfo, 17
```

**Version**:
    XiangShan: master, 011f1effd
    NEMU: b9507fbc

NewPaulWalker, 2024-11-18:
Thank you for your issue.
First, please update your debug manual version. In the new manual, `tinfo` is a read/write register, and writing to this read/write CSR has no effect. This is different because writing to a read-only CSR should raise an illegal instruction exception.
[image]

Secondly, this is indeed a bug in NEMU. When writing to tinfo, NEMU should retain the old value. I have already fixed this via a PR.

ha0lyu, 2024-11-18:
> please update your debug manual version.

Thanks for the reminder. I didn’t notice the new version before. I found it now. Thanks again.

closed 2024-11-22 by huxuan0307