OpenXiangShan/NEMU #681: Mismatch in Dependency Between `henvcfg` and `menvcfg` in NEMU and XiangShan

youzi27, 2024-11-25:
**Describe the bug**
Hi there,

I noticed a mismatch in the behavior of the dependency between `henvcfg` and `menvcfg` in NEMU and XiangShan (XS). 

Specifically:

- In NEMU: Bits 4 to 7 of `henvcfg` are aligned with `menvcfg`. Modifying `menvcfg` implicitly updates `henvcfg`.
- In XS: Bits 4 to 7 of `henvcfg` are independent of `menvcfg`. Modifying `menvcfg` does not affect `henvcfg`.

**To Reproduce**

**Initial State:** Both NEMU and XS initialize `menvcfg` to 0x80000000000000f0.
**After Modification:** When `menvcfg` is modified to 0x0:
In NEMU, `henvcfg` updates accordingly to reflect the change in `menvcfg`.
In XS, `henvcfg` remains unchanged, with bits 4 to 7 unaffected.

**Error log or Screenshots**

[image]

**Necessary information on versions**
- ready-to-run's NEMU commit id: [OpenXiangShan/ready-ro-run@3575e65](https://github.com/OpenXiangShan/ready-to-run/commit/3575e65) (Date: Fri Nov 22 17:06:53 2024 +0800) 
- XiangShan commit id: https://github.com/OpenXiangShan/XiangShan/commit/aecf601e803bfd2371667a3fb60bfcd83c333027 (Date: Tue Nov 19 16:16:35 2024 +0800)

NewPaulWalker, 2024-11-26:
Thank you for your issue! Would you mind sharing your test case?

youzi27, 2024-11-26:
Hi @NewPaulWalker ,

Thank you for your response. 

The test case to reproduce the issue is: [testcase.zip](https://github.com/user-attachments/files/17918988/testcase.zip)

NewPaulWalker, 2024-11-27:
Thank you for sharing your test. This is indeed a bug in NEMU: the **CBIE**, **CBCFE**, and **CBZE** fields of **henvcfg** should not depend on the values of these three fields in **menvcfg**.

closed 2024-11-27 by huxuan0307