OpenXiangShan/NEMU #691: `IMSIC` in `mstateen0` Does Not Control Access

youzi27, 2024-11-27:
**Describe the bug**

A bug has been identified in NEMU where, in mode=1, the `IMSIC` bit in mstateen0 does not correctly control access to the `stopei` and `vstopei` registers. These registers can still be accessed erroneously, even though such access should be restricted by the IMSIC bit.

**NOTE:** It is worth noting that other bits(e.g., `SEO`) have been tested and comply with the specification requirements.

**To Reproduce**
Steps to reproduce the behavior:

1. Clear the `IMSIC` bit in `mstateen0`.
2. Set to mode=1.
3. Attempt to access `xstopei`.

**Expected behavior**
When IMSIC is cleared, accessing xstopei should trigger an illegal instruction exception.

According to the RISC-V specification: 

> The IMSIC bit in mstateen0 controls access to the IMSIC state, including CSRs stopei and vstopei.

**Error log or Screenshots**
When access stopei:
[image]

When access vstopei:
[image]

**Necessary information on versions**
- ready-to-run's NEMU commit id: [OpenXiangShan/ready-ro-run@3575e65](https://github.com/OpenXiangShan/ready-to-run/commit/3575e65) (Date: Fri Nov 22 17:06:53 2024 +0800) 
- XiangShan commit id: https://github.com/OpenXiangShan/XiangShan/commit/aecf601e803bfd2371667a3fb60bfcd83c333027 (Date: Tue Nov 19 16:16:35 2024 +0800)

**Additional context**
None

NewPaulWalker, 2024-11-28:
Thank you for your issue. It seems that NEMU indeed does not control access to `IMSIC` and `xenvcfg` based on the corresponding values in `xstateen0`.

Also, could you provide tests for these two (`ENVCFG` and `IMSIC`)?

youzi27, 2024-11-28:
Hi @NewPaulWalker ,

Thanks for your patient reply!

Here’s the test case:

For the IMSIC tests (including `stopei` and `vstopei`): [test-IMSIC.zip](https://github.com/user-attachments/files/17944137/test-IMSIC.zip)

For the ENVCFG tests related to issue #690  (including `henvcfg` and `senvcfg`): [test-ENVCFG.zip](https://github.com/user-attachments/files/17944149/test-ENVCFG.zip)

If you need further assistance, feel free to let me know.

closed 2024-12-09 by Tang-Haojin