OpenXiangShan/NEMU #711: NEMU启动虚拟机时遇到CPU卡顿事件

PDSHan, 2024-12-08:
我在用NEMU启动虚拟机时，遇到guest os启动过程中信息打印不全和CPU卡顿事件，具体如下：

1、信息打印不全
run Guest Linux!
# lkvm run -k ./apps/Image -m 100 -c 1 --name guest-36
Info: (riscv/kvm.c) kvm__arch_load_kernel_image:178: Placing fdt at 0x81c00000 - 0x863fffff
  Info: (virtio/mmio.c) virtio_mmio_init:209: virtio-mmio.devices=0x200@0x10000000:5
  Info: (virtio/mmio.c) virtio_mmio_init:209: virtio-mmio.devices=0x200@0x10000200:6
  Info: (virtio/mmio.c) virtio_mmio_init:209: virtio-mmio.devices=0x200@0x10000400:7
  Info: (virtio/mmio.c) virtio_mmio_init:209: virtio-mmio.devices=0x200@0x10000600:8
[ 2158.176326] loop: module loaded
没有打印出OF：reserved mem等其他信息

2、CPU卡顿
[ 2169.650693] NET: Registered PF_PACKET protocol family
[ 2169.910268] 9pnet: Installing 9P2000 support
[ 2170.670207] Key type dns_resolver registered
[ 2213.650294] debug_vm_pgtable: [debug_vm_pgtable         ]: Validating architecture page table helpers
[ 2214.713116] Legacy PMU implementation is available
[ 2691.646370] watchdog: BUG: soft lockup - CPU#0 stuck for 2503s! [swapper/0:0]
[ 2691.799172] Modules linked in:
[ 2691.872538] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 6.6.21-00030-g8b68c1917beb-dirty #117
[ 2691.955291] Hardware name: linux,dummy-virt (DT)
[ 2692.005375] epc : default_idle_call+0x26/0x34
[ 2692.064746]  ra : default_idle_call+0x22/0x34
[ 2692.116287] epc : ffffffff808b0694 ra : ffffffff808b0690 sp : ffffffff81403dd0
[ 2692.242869]  gp : ffffffff81501200 tp : ffffffff8140dac0 t0 : ffffffff80e0c07c
[ 2692.357977]  t1 : 0000000000000005 t2 : 000000000000001f s0 : ffffffff81403de0
[ 2692.469498]  s1 : 0000000000000000 a0 : 0000000000000004 a1 : 0000000000000000
[ 2692.588185]  a2 : ffffaf80847c2000 a3 : 0000000000000034 a4 : ffffffff80c21290
[ 2692.709087]  a5 : 0000000000000000 a6 : 4000000000000000 a7 : 0000000000000000
[ 2692.799212]  s2 : ffffffff815031e8 s3 : 000000000000003f s4 : 0000000000000000
[ 2692.901505]  s5 : ffffffff815033a0 s6 : 0000000000000000 s7 : 0000000000000000
[ 2692.973363]  s8 : 0000000000000000 s9 : 0000000000000000 s10: 0000000000000000
[ 2693.028429]  s11: 0000000000000000 t3 : 0000000000000394 t4 : 00000000e5b906e6
[ 2693.086247]  t5 : 0000000500000000 t6 : 00000000e5b906e6
[ 2693.177123] status: 0000000200000120 badaddr: 0000000000000000 cause: 8000000000000005
[ 2693.343940] [<ffffffff808b0694>] default_idle_call+0x26/0x34
[ 2693.472426] [<ffffffff8004ea52>] do_idle+0x1e4/0x226
[ 2693.563369] [<ffffffff8004ec12>] cpu_startup_entry+0x26/0x28
[ 2693.738762] [<ffffffff808b09f8>] kernel_init+0x0/0x10a
[ 2693.894235] [<ffffffff80a0067a>] arch_post_acpi_subsys_init+0x0/0x18
[ 2694.164809] [<ffffffff80a00dca>] console_on_rootfs+0x0/0x68
[ 2694.503788] rcu: INFO: rcu_sched self-detected stall on CPU
[ 2694.634310] rcu:     0-...!: (1 ticks this GP) idle=0034/1/0x4000000000000002 softirq=7/7 fqs=0
[ 2694.825619] rcu:     (t=166688 jiffies g=-1195 q=126 ncpus=1)
[ 2694.942876] rcu: rcu_sched kthread timer wakeup didn't happen for 166687 jiffies! g-1195 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402
[ 2695.201617] rcu:     Possible timer handling issue on cpu=0 timer-softirq=2
[ 2695.359216] rcu: rcu_sched kthread starved for 166688 jiffies! g-1195 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=0
[ 2695.503693] rcu:     Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[ 2695.566814] rcu: RCU grace-period kthread stack dump:
[ 2695.611333] task:rcu_sched       state:I stack:0     pid:15    ppid:2      flags:0x00000000
[ 2695.741114] Call Trace:
[ 2695.791679] [<ffffffff808b2210>] __schedule+0x29a/0x806
[ 2695.928119] [<ffffffff808b27c2>] schedule+0x46/0xd0
[ 2696.063419] [<ffffffff808b79c0>] schedule_timeout+0x7a/0xee
[ 2696.207482] [<ffffffff80078952>] rcu_gp_fqs_loop+0x2de/0x3e2
[ 2696.339084] [<ffffffff80079d0e>] rcu_gp_kthread+0x106/0x126
[ 2696.512730] [<ffffffff80031014>] kthread+0xc4/0xe0
[ 2696.643659] [<ffffffff808b8b32>] ret_from_fork+0xe/0x1c
[ 2696.740392] rcu: Stack dump where RCU GP kthread last ran:
[ 2696.834812] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G             L     6.6.21-00030-g8b68c1917beb-dirty #117
[ 2697.121297] Hardware name: linux,dummy-virt (DT)
该事件一直无法得到解决，反复出现。但是最后guest os仍打印出了Run /virt/init as init process
[ 4083.726345] [<ffffffff80a0067a>] arch_post_acpi_subsys_init+0x0/0x18
[ 4083.790340] [<ffffffff80a00dca>] console_on_rootfs+0x0/0x68
[ 4084.129351] Run /virt/init as init process
Mounting...
/ # [   22.415677] rcu: INFO: rcu_sched self-detected stall on CPU
[   22.621987] rcu:     0-...!: (1 ticks this GP) idle=011c/1/0x4000000000000002 softirq=26/26 fqs=0
[   22.834965] rcu:     (t=84134 jiffies g=-1167 q=173 ncpus=1)
请问是哪里出了问题？

NewPaulWalker, 2024-12-09:
请参考我们提供的 [issue 模板](https://github.com/OpenXiangShan/NEMU/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=) 进行提问并提供详细的信息，比如你运行的二进制文件，反汇编文件，以及完整的 log 文件等。

[TRANSLATION]
Please refer to the [issue template](https://github.com/OpenXiangShan/NEMU/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=) we provided when asking questions and include detailed information, such as the binary file you ran, the disassembled file, and the complete log file, etc.

PDSHan, 2024-12-09:
**Describe the bug**
在NEMU上启动的os中，运行kvmtool启动虚拟机，但是遇到了CPU卡顿事件。

**Error log or Screenshots**
启动日志：
[log.txt](https://github.com/user-attachments/files/18058539/log.txt)

**Necessary information on versions**
具体的软件版本：
1、NEMU使用master分支，riscv64-xs_defconfig。
2、guest os和host os用的都是linux-6.6.21_xiangshan版本。其中，guest os的config用的是默认config，并且开启CONFIG_DMA_RESTRICTED_POOL=y。host os用的xiangshan_defconfig，并且参照https://bosc.yuque.com/yny0gi/fzefxm/mypmesdadu6l7g7q更改了config和根文件系统配置。
3、kvmtool用的版本号是e17d182ad3。
4、opensbi用的是202408_xiangshan a83569d platform: Add kunminghu support。
请问还有什么信息需要补充吗？