OpenXiangShan/NEMU #728: 对于同一workload进行simpoint，不同branch的NEMU分支profiling得到的总指令数不同

FoolgryGamer, 2024-12-18:
**Describe the bug**

我所使用的workload是bbl+Linux+application，两个不同branch的NEMU分支版本为，暂且称之为old_NEMU以及new_NEMU
old_NEMU:
```
commit fe61fbfb1b681fc11d3539b2cb8984344173093b (HEAD -> master, origin/master, origin/HEAD)
Merge: dc3492f6 0f0ab550
Author: shinezyy <shinezyy@qq.com>
Date:   Mon Feb 26 17:11:42 2024 +0800
    Merge pull request #266 from OpenXiangShan/checkpoint_zstd_decompress

    support using zstd to take and restore checkpoints
```
以及new_NEMU:
```
commit 9b0a013517a877df0f32af25af7e9d66eac4baf7 (origin/master, origin/HEAD)
Author: NewPaulWalker <88817570+NewPaulWalker@users.noreply.github.com>
Date:   Thu Nov 14 10:08:35 2024 +0800

    fix(lrsc): configure NEMU's reservation set size to 64B to match XiangShan. (#654)
```
使用的工作负载是SPECCPU的bin文件，主要用simpoint来制作切片
目前存在的问题是，使用两个不同的NEMU版本制作切片，在最终simpoint profiling中的NEMU Log中得到的指令数并不相同
其中new_NEMU的profiling的输出log如下：
```
^[[1;34m[src/cpu/cpu-exec.c:95,monitor_statistic] host time spent = 102,716,838,612 us^[[0m
^[[1;34m[src/cpu/cpu-exec.c:97,monitor_statistic] total guest instructions = 8,541,886,488,146^[[0m
^[[1;34m[src/cpu/cpu-exec.c:98,monitor_statistic] vst count = 30,108,604, vst unit count = 30,108,604, vst unit optimized count = 0^[[0m
^[[1;34m[src/cpu/cpu-exec.c:101,monitor_statistic] simulation frequency = 83,159,554 instr/s^[[0m
^[[1;34m[src/utils/state.c:30,is_exit_status_bad] NEMU exit with good state: 2, halt ret: 0^[[0m
```
old_NEMU的profiling log得到的指令数为5,160,311,321,186。
该数值与new_NEMU中得到的指令数不同，且差异较大

同时我们验证过old_NEMU和物理机上对bin文件采样的指令数的差距，得到的结论是old_NEMU采样得到的指令数与物理机perf采样得到的指令数几乎一致
所以目前有一些疑惑，可能是什么原因导致了NEMU不同branch采样得到的指令数的差异

补充信息：不同branch NEMU，我们采用的bbl方式也不同

- old_NEMU采用了riscv-pk的方式，new_NEMU采用了opensbi(version 1.5)的方式
- new_NEMU包含更多的拓展，old_NEMU仅包含gcb拓展（但两个工程的workload相同，仅包含rv64gc拓展）
- old_NEMU中的GCPT restorer是在NEMU中的，new_NEMU中的GCPT restorer是OpenXiangShan/LibCheckpointAlpha中的版本
- 采用的kernel版本均为6.12

Tang-Haojin, 2024-12-18:
Thanks for your nice reporting. This is a known issue because the instruction counting in some circumstances is not accurate. @cebarobot big god is working on this issue and let us praying for him.

Besides, the mtime of NEMU depends on host time, which also brings us uncertainty for the dynamic instruction trace even using a NEMU to run two times. @cebarobot is also working on this.

FoolgryGamer, 2024-12-18:
***@***.***

FoolgryGamer, 2024-12-20:
非常感谢！

closed 2024-12-23 by FoolgryGamer