OpenXiangShan/NEMU #787: make riscv64-xs_defconfig error

itcastcastcast, 2025-01-21:
make riscv64-xs_defconfig
/home/jinm/ics2024/nemu/scripts/config.mk:20: \033[1;31mWarning: .config does not exist!\033[0m
/home/jinm/ics2024/nemu/scripts/config.mk:21: \033[1;31mTo build the project, first run 'make menuconfig'.\033[0m
Makefile:147: *** target file '/home/jinm/xiangshan/xs-env/NEMU/build/nemu-interpreter' has both : and :: entries.  Stop.

当我尝试配置的时候，报错，不清楚这是什么问题

itcastcastcast, 2025-01-21:
之前xs-env的默认版本没有这个问题，但是直接参照这个xs-env给定的默认版本最后会在nemu与chisel联合仿真的时候出现abort,因此需要我更新到最新版本。但是当我更新到最新版本编译nemu就出现问题了。

NewPaulWalker, 2025-01-21:
看起来你似乎在两个不同的 nemu 仓库里进行操作？其中一个可能来自于 https://github.com/NJU-ProjectN/nemu/tree/ics2024 ，另一个则是来自于 xs-env 下的 https://github.com/OpenXiangShan/NEMU 。你是否确认在 xs-env/NEMU 目录下执行的 make riscv64-xs_defconfig 命令？

如果以上信息没有帮助你解决问题，请提供更加完整的信息，如:
* 你在什么目录下执行的什么命令?
* 这个仓库是从哪里 clone 的？
* 你是否做了什么修改？

你提供的信息越详细，我们才能知道你在做什么，以及发现问题可能出在哪里。

[TRANSLATION]
It seems like you are operating on two different NEMU repositories? One might be from https://github.com/NJU-ProjectN/nemu/tree/ics2024, and the other from the xs-env at https://github.com/OpenXiangShan/NEMU. Have you confirmed that you executed the `make riscv64-xs_defconfig` command under the `xs-env/NEMU` directory? 

If the above information does not help resolve your issue, please provide more detailed information, such as:
- What commands were executed and in which directory?
- Where was the repository cloned from?
- Did you make any modifications?

The more details you provide, the better we can understand what you're doing and identify where the issue might be.

cebarobot, 2025-01-21:
ICS 使用的 NEMU 和香山使用的 NEMU 已经完全不是一个 NEMU 了，请使用香山提供的 NEMU。

NewPaulWalker, 2025-01-21:
> Makefile:147: *** target file '/home/jinm/xiangshan/xs-env/NEMU/build/nemu-interpreter' has both : and :: entries. Stop.

这个是由于 Makefile 中的某个目标同时定义了两种规则格式 : 和 ::

cebarobot, 2025-01-21:
另外，请确认您有正确的配置 `NEMU_HOME` 环境变量。请运行 `echo $NEMU_HOME` 确认其是否指向当前路径。如果不对，则在 NEMU 文件夹下运行 `export NEMU_HOME=$(pwd)`。

itcastcastcast, 2025-01-21:
不不不，我实在xiangshan下，在ics下的是自己的一个测试，是看看问题出在哪里

itcastcastcast, 2025-01-21:
谢谢，我找到问题了，因为两边NEMU都在用并且都用到了NEMU_HOME导致的错误，感谢

closed 2025-01-21 by itcastcastcast