OpenXiangShan/NEMU #815: NEMU profiling能够看到"Hello, RISC-V World!"但是最后Hanging

KKK10001, 2025-02-21:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

使用OpenSBI构建生成gcpt.bin (bbl.bin)所需的所有部件，包括：
1. rootfs (initramfs-emu.txt中指定hello)
2. 设备树（单核）
[image]
3. Linux Kernel
生成了-rwxr-xr-x 1 root root **13M** Feb 21 15:40 arch/riscv/boot/Image
4. OpenSBI
make PLATFORM=generic FW_PAYLOAD_PATH=$RISCV_LINUX_HOME/arch/riscv/boot/Image FW_FDT_PATH=$WORKLOAD_BUILD_ENV_HOME/dts/build/xiangshan.dtb FW_PAYLOAD_OFFSET=0x200000
-rw-r--r-- 1 root root 14M Feb 21 15:56 /root/dev/checkpoint/opensbi/build/platform/generic/firmware/fw_payload.bin
5. 采用LibCheckpointAlpha链接fw_payload.bin，生成gcpt.bin (renamed as bbl.bin)

最后，采用NEMU对上述bin进行profiling

bbl-err.txt最后的打印信息是：
Hello, RISC-V World!

hanging

完整的运行时报告如下：

OpenSBI v1.6-31-gec09918
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name               : freechips,rocketchip-unknown
Platform Features           : medeleg
Platform HART Count         : 1
Platform IPI Device         : aclint-mswi
Platform Timer Device       : aclint-mtimer @ 1000000Hz
Platform Console Device     : xlnx-uartlite
Platform HSM Device         : ---
Platform PMU Device         : ---
Platform Reboot Device      : ---
Platform Shutdown Device    : ---
Platform Suspend Device     : ---
Platform CPPC Device        : ---
Firmware Base               : 0x80100000
Firmware Size               : 325 KB
Firmware RW Offset          : 0x40000
Firmware RW Size            : 69 KB
Firmware Heap Offset        : 0x48000
Firmware Heap Size          : 37 KB (total), 2 KB (reserved), 11 KB (used), 23 KB (free)
Firmware Scratch Size       : 4096 B (total), 416 B (used), 3680 B (free)
Runtime SBI Version         : 2.0
Standard SBI Extensions     : time,rfnc,ipi,base,hsm,pmu,dbcn,legacy
Experimental SBI Extensions : fwft,sse

Domain0 Name                : root
Domain0 Boot HART           : 0
Domain0 HARTs               : 0*
Domain0 Region00            : 0x0000000038000000-0x000000003800ffff M: (I,R,W) S/U: ()
Domain0 Region01            : 0x0000000080140000-0x000000008015ffff M: (R,W) S/U: ()
Domain0 Region02            : 0x0000000080100000-0x000000008013ffff M: (R,X) S/U: ()
Domain0 Region03            : 0x000000003c000000-0x000000003fffffff M: (I,R,W) S/U: (R,W)
Domain0 Region04            : 0x0000000000000000-0xffffffffffffffff M: () S/U: (R,W,X)
Domain0 Next Address        : 0x0000000080200000
[... 256 lines cut ...]

[    0.124972] printk: legacy bootconsole [sbi0] disabled
[    0.125205] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled

[    0.128012] 40600000.serial: ttyUL0 at MMIO 0x40600000 (irq = 12, base_baud = 0) is a uartlite

[    0.128339] uartlite 40600000.serial: Runtime PM usage count underflow!

[    0.129376] clk: Disabling unused clocks

[    0.133483] Freeing unused kernel image (initmem) memory: 2156K

[    0.133629] Run /init as init process

Hello, RISC-V World!

hanging

### Environment

- riscv-rootfs commit id: d925bac27a2c725488435b2d9015f1078b3c969f
- OpenSBI commit id: ec09918426a5ccff5de09cd41c3f36d0ad5443f1
- NEMU commit id: 1ff30304fa3abe663c3ca0c60b7e897f96ead2eb
- Operating System: Ubuntu 20.04 LTS
- gcc version: gcc (Ubuntu 13.1.0-8ubuntu1~20.04.2) 13.1.0

cebarobot, 2025-02-21:
This is not a XiangShan Problem. So I transferred this issue to NEMU repo.

I don't get what your actual problem is. Everything works perfectly, right? 

这不是一个香山相关的问题，我将此 Issue 转移到了 NEMU 仓库。

我没有理解您的问题是什么。看起来所有东西都正常工作，这不是很好吗？

cebarobot, 2025-02-21:
If you are confusing about the "Hanging", Please check the [source code of hello program](https://github.com/OpenXiangShan/riscv-rootfs/blob/master/apps/hello/hello.c) 

如果你对 Hanging 感到疑惑，请看您使用的 [hello 程序的源码](https://github.com/OpenXiangShan/riscv-rootfs/blob/master/apps/hello/hello.c) 。

closed 2025-02-22 by KKK10001