OpenXiangShan/NEMU #816: bbl-err.txt中已经出现"END gcc_scilab_run"的dump，但是NEMU profiling的进程不退出

KKK10001, 2025-02-24:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

bbl-err.txt已经停止更新，并且出现了END ${bm}_${input_pattern}_run的dump，如下所示：

[image]

并且，采样生成的文件simpoint_bbv.gz的大小也停止增长，如下所示：
[image]

但是，NEMU profiling的进程始终不退出

[image]

[image]

### What did you do before

bbl-err.txt已经停止更新，并且出现了END ${bm}_${input_pattern}_run的dump，如下所示：

[image]

并且，采样生成的文件simpoint_bbv.gz的大小也停止增长，如下所示：
[image]

但是，NEMU profiling的进程始终不退出

[image]

[image]

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

xyyy1420, 2025-02-24:
NEMU requires an exit signal, which can be referenced from the code shown in this file https://github.com/OpenXiangShan/riscv-rootfs/blob/master/apps/trap/trap.c.

KKK10001, 2025-02-24:
Exactly. Thank you for your advice.

closed 2025-02-24 by KKK10001