OpenXiangShan/NEMU #818: NEMU启动debian.img时无法正常登录用户

duiek, 2025-02-27:
根据文档：编译使用RISCV-pk的Linux kernel

> https://docs.xiangshan.cc/zh-cn/latest/workloads/linux-kernel-for-xs/

根据第4点，使用riscv-linux nanshan 分支、riscv-pk noop 分支得到了bbl.bin文件

[image]

NEMU按照make riscv64-xs-cpt_defconfig配置，仅修改最大内存，CONFIG_MSIZE

debian镜像参照：

> https://github.com/OpenXiangShan/NEMU/tree/master/resource/debian

按照文档制作了一个最简的debian镜像。

`debootstrap --arch=riscv64 --keyring /usr/share/keyrings/debian-archive-keyring.gpg --include=debian-archive-keyring unstable /mnt http://deb.debian.org/debian`

系统启动为多用户模式（systemctl set-default multi-user.target），并尝试了紧急模式与免密登录两种方法

通过 ./build/riscv64-nemu-interpreter -b /home/diann/riscv-pk/build/bbl.bin 启动后，控制台最后只看到：

> [FAILED] Failed to start systemd-logind.service - User Login Management.
See 'systemctl status systemd-logind.service' for details.
         Starting modprobe@drm.service - Load Kernel Module drm...
[  OK  ] Started dbus.service - D-Bus System Message Bus.
[  OK  ] Finished networking.service - Raise network interfaces.
[  OK  ] Reached target network.target - Network.
         Starting ssh.service - OpenBSD Secure Shell server...
         Starting systemd-user-sessions.service - Permit User Sessions...
[  OK  ] Finished systemd-user-sessions.service - Permit User Sessions.
[  OK  ] Started getty@tty1.service - Getty on tty1.
[  OK  ] Started serial-getty@hvc0.service - Serial Getty on hvc0.
[  OK  ] Reached target getty.target - Login Prompts.
[  OK  ] Started ssh.service - OpenBSD Secure Shell server.
[  OK  ] Finished modprobe@drm.service - Load Kernel Module drm.
         Starting systemd-logind.service - User Login Management...
[FAILED] Failed to start systemd-logind.service - User Login Management.
See 'systemctl status systemd-logind.service' for details.
         Starting modprobe@drm.service - Load Kernel Module drm...
[  OK  ] Finished modprobe@drm.service - Load Kernel Module drm.
         Starting systemd-logind.service - User Login Management...
[FAILED] Failed to start systemd-logind.service - User Login Management.
See 'systemctl status systemd-logind.service' for details.
         Starting modprobe@drm.service - Load Kernel Module drm...
[  OK  ] Finished modprobe@drm.service - Load Kernel Module drm.
         Starting systemd-logind.service - User Login Management...
[FAILED] Failed to start systemd-logind.service - User Login Management.
See 'systemctl status systemd-logind.service' for details.
         Starting modprobe@drm.service - Load Kernel Module drm...
[  OK  ] Finished modprobe@drm.service - Load Kernel Module drm.
         Starting systemd-logind.service - User Login Management...
[FAILED] Failed to start systemd-logind.service - User Login Management.
See 'systemctl status systemd-logind.service' for details.
         Starting modprobe@drm.service - Load Kernel Module drm...
[  OK  ] Finished modprobe@drm.service - Load Kernel Module drm.
[FAILED] Failed to start systemd-logind.service - User Login Management.
See 'systemctl status systemd-logind.service' for details.
[  OK  ] Reached target multi-user.target - Multi-User System.

> 

> ^[[44;1R^[[44;186R

希望的正常情况是出现登陆提示符或者免密登录并执行/root/.bashrc中的内容。

完整日志文件：

[nemu-debian.txt](https://github.com/user-attachments/files/19007533/nemu-debian.txt)

大家又遇到类似的问题吗？或者求大佬们给点意见，非常感谢

cebarobot, 2025-02-28:
感谢您的 Issue，但我们目前可能没有成员对在 NEMU 上启动发行版比较熟悉了。

另外，NEMU 默认没有输入功能，这是个关键限制，可能需要请您注意。

duiek, 2025-02-28:
非常感谢您的回复，我看到sdhost-nemu以r+状态打开img，但是挂载mmcblk0p1时又是只读，所以还是没有输入功能是吗？

如果你们能重新测试一下nemu启动debian的功能就最好了，也许我的使用方式或者操作有误。

closed 2025-03-20 by duiek