OpenXiangShan/NEMU #821: NEMU恢复检查点报错

KKK10001, 2025-03-03:
NEMU master编译生成的./build/riscv64-nemu-interpreter不支持--restore入参

根据 https://docs.xiangshan.cc/zh-cn/latest/tools/simpoint/ 的推荐切换到NEMU gcpt_new_mem_layout分支之后，进行常规配置如下：
make riscv64-xs-cpt_defconfig
make menuconfig
若上述缺省的配置，在编译NEMU时会报链接优化的错
file:///home/kuanbba/%E5%9B%BE%E7%89%87/%E5%8B%BE%E9%80%89NEMU%E9%85%8D%E7%BD%AE%E9%93%BE%E6%8E%A5%E4%BC%98%E5%8C%96%E6%97%B6%E7%BC%96%E8%AF%91%E6%8A%A5%E9%94%99.png

重新make menuconfig，取消勾选链接优化的选项，重新编译，可以正常生成riscv64-nemu-interpreter

……持续进行基于OpenSBI的构建流程，成功生成了19MB大小的gcpt.bin，并且将其复制到$NEMU_HOME/resource/gcpt_restore/build目录

随后，进行NEMU-Profiling，source的脚本如下：

[image]

最终，发生了如下assertion:

[image]

xyyy1420, 2025-03-13:
Please use the `master` branch and remove the `-r` option

KKK10001, 2025-04-10:
> Please use the `master` branch and remove the `-r` option

如我于本问题单开头所铺垫的那样，我之所以要切换到gcpt_new_mem_layout分支，是因为
"NEMU master编译生成的./build/riscv64-nemu-interpreter不支持--restore入参"，而我的根本目的是要restore checkpoint

"-c,--restore            restoring from CPT FILE"只在NEMU gcpt_new_mem_layout分支中有定义

xyyy1420, 2025-04-10:
> > Please use the `master` branch and remove the `-r` option
> 
> 如我于本问题单开头所铺垫的那样，我之所以要切换到gcpt_new_mem_layout分支，是因为 "NEMU master编译生成的./build/riscv64-nemu-interpreter不支持--restore入参"，而我的根本目的是要restore checkpoint
> 
> "-c,--restore restoring from CPT FILE"只在NEMU gcpt_new_mem_layout分支中有定义

I mean checkout to the master branch and directly passing the checkpoint path as a parameter. I removed the `-r` option, but the restore functionality is still kept.

KKK10001, 2025-04-10:
谢谢

closed 2025-04-10 by KKK10001