OpenXiangShan/NEMU #852: How to run llama.cpp on NEMU?

sploving1, 2025-04-17:
As we know, llama.cpp is a  C++ framework for efficient inference of LLaMA and other large language models (LLMs), featuring model quantization to enable AI applications on edge devices with limited resources. I want to run llama.cpp on NEMU with RISC-V 64 instructions. Without OS emulation, how can I run llama.cpp directly on NEMU? Thanks.

eastonman, 2025-04-17:
What do you mean by "without OS emulation"?
If you want a qemu-user-like "thing", use qemu-user.
If your workload runs on qemu-system, it should run on NEMU.

sploving1, 2025-04-17:
Thank you for your incredibly prompt response - this project is honestly one of the fastest issue replies I've ever seen in my years participating in open source projects, truly impressive!

You've pointed out two viable approaches:

1. Booting Linux to run the program

2. Using nexus-am for baremetal execution

After briefly reviewing both options, while nexus-am seems suitable for simpler workloads, booting Linux appears more appropriate for my case given the complex library dependencies in LLM inference (like llama.cpp). I'll start by experimenting with the Linux approach first.

Regarding your note about NEMU being an ISA-level emulator: I completely understand it won't reflect microarchitectural performance characteristics. My primary goal is actually functional validation of some custom ISA extensions I'm working on, rather than performance measurement at this stage.

I may come back with follow-up questions about the Linux boot process if needed - your guidance has been tremendously helpful already. Thanks again!

sploving1, 2025-04-18:
I've successfully run basic workloads like busybox after booting Linux in NEMU. It works as expected, although it doesn't seem to support user input. Now, I'm trying to run the llama.cpp workload, which depends on a large model file that exceeds the size limitations of the initramfs. Where is the appropriate location to place this model file so that it can be accessed after Linux has booted? I've reviewed the initramfs-emu.txt, initramfs-spec.txt, and initramfs-autorun.txt files within the rootfsimg directory, but none of them address this specific use case.

cebarobot, 2025-04-18:
> Now, I'm trying to run the llama.cpp workload, which depends on a large model file that exceeds the size limitations of the initramfs. Where is the appropriate location to place this model file so that it can be accessed after Linux has booted?

You may use the [SD card](https://github.com/OpenXiangShan/NEMU/tree/master/resource/sdcard) to store that. However, I'm not sure whether it works well.

sploving1, 2025-04-18:
In the above link, it is stated that disabling 'General setup -> Initial RAM filesystem and RAM disk (initramfs/initrd) support' in menuconfig is a prerequisite for enabling SD card support. This seems counterintuitive, as initramfs is generally considered important for the boot process,  as it allows for the loading of necessary modules and early user space setup.

Could you please clarify the reason behind this recommendation to disable initramfs in order to support the SD card in NEMU? Thanks.

cebarobot, 2025-04-18:
It seems that document tried to use the SD Card to replace initramfs as a root file system. I don't think that is necessary.

I think you could still use initramfs as root file system, while use SD Card as an external storage... However, there is no documents about this...

sploving1, 2025-04-21:
I have tried the [SD card](https://github.com/OpenXiangShan/NEMU/tree/master/resource/sdcard) but failed. It hangs when booting the Linux kernel.  Here is the booting log. It hangs at the last line.
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000000] console [hvc0] enabled
[    0.000000] console [hvc0] enabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.000000] pid_max: default: 4096 minimum: 301
[    0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.000000] futex hash table entries: 16 (order: -4, 384 bytes)
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.000000] Unpacking initramfs...
[    0.000000] workingset: timestamp_bits=62 max_order=19 bucket_order=0