OpenXiangShan/NEMU #868: 关于NEMU和NutShell的`pmpaddr`对齐问题

LuLuji04, 2025-05-07:
## Describe the bug

  当使用NEMU作为NutShell的参考时，由于NutShell未实现 `pmpaddr4-15`，NutShell抛出了非法指令异常，但是NEMU却没有触发异常。

这是NutShell的运行log
[image]
这是二者的差异，由于NEMU没有触发异常，其mepc没有更新
[image]

## Expected behavior
  当使用NEMU作为NutShell的参考时，由于NutShell未实现 `pmpaddr4-15`，NEMU应该在 M-mode 下访问该 CSR 应抛出非法指令异常。

## To Reproduce
在NEMU中编译
  ```bash
  make riscv64-nutshell-ref_defconfig
  make -j
```
在NutShell中运行以下程序[test.zip](https://github.com/user-attachments/files/20084659/test.zip)，以刚编译的NEMU为参照
  ```bash
./build/emu -i /<path-to>/test/build/test-riscv64-nutshell.bin  --diff /<path-to>/NEMU/build/riscv64-nemu-interpreter-so
```
## Environment
- NutShell branch: master  
- NutShell commit id: fc12171d929e7e589fab9f794ab63ce12e6c594e  
- NEMU commit id: 4a24b77a61505e34745667b1ad712a817b090cf8

NewPaulWalker, 2025-05-09:
> NEMU应该在 M-mode 下访问该 CSR 应抛出非法指令异常。

I disagree with this statement.

I found the following description in the RISC-V PMP specification: 

>  Implementations may implement zero, 16, or 64 PMP entries; the lowest-numbered PMP entries must be implemented first. All PMP CSR fields are WARL and may be read-only zero. 

Therefore, there is no such thing as an implementation that only implements 4 pmpaddr registers. NEMU actually does the same. In https://github.com/OpenXiangShan/NEMU/blob/master/configs/riscv64-nutshell-ref_defconfig#L29 , 16 pmpaddr registers are still implemented, but only 4 of them are active; the rest are read-only zeros: https://github.com/OpenXiangShan/NEMU/blob/master/src/isa/riscv64/system/priv.c#L1773

I believe this might be a bug in NutShell.

LuLuji04, 2025-05-10:
Got it! However, NutShell is no longer maintained, right? There are many mismatches between NutShell and the current version of NEMU.

NewPaulWalker, 2025-05-12:
> Got it! However, NutShell is no longer maintained, right? There are many mismatches between NutShell and the current version of NEMU.

I'm not very familiar with this, but currently OpenXiangShan/NEMU is mainly used for aligning with XiangShan, especially Kunminghu.