OpenXiangShan/NEMU #887: NEMU支持页大小配置吗

yujiewang22, 2025-05-19:
基于GEM5做架构探索时，需要使用NEMU进行打checkpoint以及difftest。想要测试pagesize的大小对跑分的影响，因此需要调整RISC-V架构下的基础页。请问NEMU是否有相应的宏配置，使得修改页大小可以直接指定？

cebarobot, 2025-05-19:
NEMU 支持 RISC-V 规定的 4KB、2MB、1GB、512GB（仅限 Sv48）的页大小，还支持 64KB 的页（Svnapot）。

基础的页的大小是 ISA 规定的，在 NEMU 中是写死的，很难进行变化。如果你希望在上述范围外改变页大小，推荐你从 Svnapot 的相关基础设施出发开展研究。

eastonman, 2025-05-19:
我记得 Linux 近期有 Svnapot Folio 相关的 Patch 合并，你可以看看。在此之上可以扩展 Svnapot 的能力，包括 ISA 和 Linux

yujiewang22, 2025-05-19:
> NEMU 支持 RISC-V 规定的 4KB、2MB、1GB、512GB（仅限 Sv48）的页大小，还支持 64KB 的页（Svnapot）。
> 
> 基础的页的大小是 ISA 规定的，在 NEMU 中是写死的，很难进行变化。如果你希望在上述范围外改变页大小，推荐你从 Svnapot 的相关基础设施出发开展研究。

感谢回复。由于基础页大小作为软硬件接口，需要同时修改 Linux 、Xs-GEM5、 NEMU，若没有便捷调整的宏会较为复杂。若NEMU支持 Svnapot 扩展，我将考虑从此方面入手。此外，目前的 Xs-GEM5 是否支持该扩展？谢谢！

cebarobot, 2025-05-19:
* 基础页大小与各种东西都紧密相关，包括但不限于地址位宽、各级页框号的位宽、PTE 的格式等等。牵一发而动全身，不可能有方便的宏进行调整的。
* Svnapot 原生只提供了 64KB 页的支持，但预留了其他页大小的可能。显然香山、NEMU 都只按照 RISC-V 规范进行了实现。
* XS-GEM5 应该是不支持 Svnapot 的，具体请向 XS-GEM5 仓库咨询。

祝您好运~

yujiewang22, 2025-05-19:
> * 基础页大小与各种东西都紧密相关，包括但不限于地址位宽、各级页框号的位宽、PTE 的格式等等。牵一发而动全身，不可能有方便的宏进行调整的。
> * Svnapot 原生只提供了 64KB 页的支持，但预留了其他页大小的可能。显然香山、NEMU 都只按照 RISC-V 规范进行了实现。
> * XS-GEM5 应该是不支持 Svnapot 的，具体请向 XS-GEM5 仓库咨询。
> 
> 祝您好运~

已获悉这些方面，会按照建议和意见进行有关工作。再次感谢，顺颂商祺~

qiao-huan, 2025-06-20:
NEMU支持内存大小的修改吗

csgxiong, 2025-10-30:
@cebarobot 你好，上面回答中提到的“Svnapot 原生只提供了 64KB 页的支持，但预留了其他页大小的可能。显然香山、NEMU 都只按照 RISC-V 规范进行了实现。”，这里64KB的页大小是指svnapot的页一定是64KB页大小的leaf-pte吗？根据最新的risc-v privilege文档svnapot章节的描述来看，并没有限定带有svnapot标志的pte必须是leaf-pte，也可以是中间级pte，不知道我这个说法是否存在问题呢？
谢谢！

DavidYu360, 2025-11-22:
`i=0` 代表 last stage。64K page 只能用在最後的 state。
(參照 privileged spec, Virtual Address Translation Process) 
[image]