OpenXiangShan/NEMU #922: NEMU如何实现串口输入？

zzzhang2001, 2025-08-20:
NEMU standalone运行linux，根据inittab，所有命令完成之后应该进入ash，即交互式shell，但是没有找到输入命令的方式；
`::sysinit:/bin/busybox mount -t proc proc /proc
::sysinit:/bin/busybox mount -t sysfs sysfs /sys
::sysinit:/bin/busybox mount -t tmpfs tmpfs /tmp
::sysinit:/bin/busybox mount -o remount,rw /dev/htifbd0 /
::sysinit:/bin/busybox --install -s
::sysinit:/bin/mkdir -p /dev/pts
::sysinit:/bin/echo /bin/mdev > /proc/sys/kernel/hotplug
::sysinit:/sbin/mdev -s
/dev/console::sysinit:-/bin/ash`
默认配置的是uartlite，在config中开启CONFIG_UARTLITE_INPUT_FIFO=y，编译nemu会出错
`+ CC src/device/uartlite.c
src/device/uartlite.c: In function ‘serial_rx_ready_flag’:
src/device/uartlite.c:83:24: error: ‘LSR_RX_READY’ undeclared (first use in this function)
   83 |   return (f == r ? 0 : LSR_RX_READY);
      |                        ^~~~~~~~~~~~
src/device/uartlite.c:83:24: note: each undeclared identifier is reported only once for each function it appears in
src/device/uartlite.c: At top level:
src/device/uartlite.c:53:13: error: ‘serial_dequeue’ defined but not used [-Werror=unused-function]
   53 | static char serial_dequeue() {
      |             ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors`
修掉之后通过nemu运行linux，确实会生成 /tmp/nemu-serial，但是echo命令没有任何反应

是否uartlite_input_fifo这一串口输入方式还受支持？如果不支持应该使用什么方式串口输入？

zzzhang2001, 2025-08-21:
好的，感谢，rcs我已经使用过了，希望测试交互式是因为想上fpga测试cpu；
我还有一个小问题，NEMU现在支持哪些串口方式？config里有serial，uartlite，uart_snps;
我尝试切换为UART_SNPS（这个应该是16550a？）但是执行linux没有输出，我在设备树里指定的是ns16550a

cyyself, 2025-12-15:
> 默认配置的是uartlite，在config中开启CONFIG_UARTLITE_INPUT_FIFO=y，编译nemu会出错

已在 #963 中解决。现在已经可以通过 `echo "your_command" > /tmp/nemu-serial` 这样的操作来完成输入。