OpenXiangShan/NEMU #925: Build Error with `clang20` Because Compiler flag `-c` Clashes with `-E`

scratch-er, 2025-09-13:
**Describe the bug**

When building riscv64-nemu-interpreter with clang20, there are compiler errors because `-c` and `-E` is specified in the same command line in https://github.com/OpenXiangShan/NEMU/blob/790edeb25df07d4739ca33ecca5d581c2e65b6ba/scripts/build.mk#L61

**To Reproduce**

The toolchain used to build NEMU is clang20 from the package manager of Arch Linux.

Configure NEMU with defualt configs, except architecture is set to riscv64 and the compiler is set to clang. And then run make.

**Expected behavior**

The project should build currently.

**Error log or Screenshots**

```
clang: error: argument unused during compilation: '-c' [-Werror,-Wunused-command-line-argument]
make: *** [/home/scratcher/Workspace/NEMU/scripts/build.mk:61: /home/scratcher/Workspace/NEMU/build/obj-riscv64-nemu-interpreter/src/isa/riscv64/instr/decode.o] Error 1
```

**Necessary information on versions**
- NEMU version: master branch

cebarobot, 2025-09-13:
Emmmm... Actually, the default config in menuconfig is not the real "default" config. This "default" config has never been tested, and is not promised to be good.

Please use the following command to setup the real default config:
```
make riscv64-xs_defconfig
```

Then, you can modify the compiler in menuconfig.

scratch-er, 2025-09-13:
> Emmmm... Actually, the default config in menuconfig is not the real "default" config. This "default" config has never been tested, and is not promised to be good.
>
> Please use the following command [...]

Actually, ANY configs under the `config/` directory are affected by this, as long as

- use clang20 as the compiler
- run `make xxx_defconfig` to select one config, for example, `riscv64-xs-clang_defconfig`
- run `make menuconfig` to set the compiler to `clang` if it is not
- run make

And the same error will be reproduced.