OpenXiangShan/NEMU #926: Unused `static inline` Functions in `priv.c` STILL cause Compiler Errors on Clang

scratch-er, 2025-09-13:
**Describe the bug**

When trying to build NEMU interpreter with clang20, there are compiler errors on unused functions.

**To Reproduce**

Configure the project from a clean state. Leave all the configurations as default except setting the architecture to riscv64 and the compiler to clang. Then run make.

**Error log or Screenshots**

```
+ CC src/isa/riscv64/system/priv.c
src/isa/riscv64/system/priv.c:301:20: error: unused function 'is_U_custom_csr' [-Werror,-Wunused-function]
  301 | static inline bool is_U_custom_csr(uint32_t addr) {
      |                    ^~~~~~~~~~~~~~~
src/isa/riscv64/system/priv.c:312:20: error: unused function 'is_H_custom_csr' [-Werror,-Wunused-function]
  312 | static inline bool is_H_custom_csr(uint32_t addr) {
      |                    ^~~~~~~~~~~~~~~
2 errors generated.
```

**Necessary information on versions**

- NEMU version: master branch

**Additional context**

I have noticed that there is already a commit trying to solve this problem. The commit ID is:

e5a6c99bef368478a20101fa18a09221fa71088a

But it turns out that some other static inline functions may be unused under some other configurations.

cebarobot, 2025-09-13:
Emmmm... Actually, the default config in menuconfig is not the real "default" config. This "default" config has never been tested, and is not promised to be good.

Please use the following command to setup the real default config:
```
make riscv64-xs_defconfig
```

Then, you can modify the compiler in menuconfig.

scratch-er, 2025-09-13:
> Emmmm... Actually, the default config in menuconfig is not the real "default" config. This "default" config has never been tested, and is not promised to be good.
>
> Please use the following command [...]

Yes, these unused functions will not cause any problem if using any of the configs under `configs/` (by invoking `make xxx_defconfig`). I ran CI in the `ghcr.io/openxiangshan/xs-env:latest container to check this.

If only those configs are supported, I will close this issue.

closed 2025-09-14 by scratch-er

scratch-er, 2025-09-14:
Based on the discussion, it appears that the issue occurs when using a manually configured setup rather than one of the predefined configurations (e.g., `riscv64-xs_defconfig`). Since the project currently only supports configurations tested in CI, I am closing this issue as the problem does not occur with supported configurations.

To prevent similar confusion in the future, it would be helpful to explicitly state in the documentation or README that users should start with a supported configuration (e.g., by running `make <arch>_defconfig`) and then customize settings as needed, rather than configuring everything from scratch. This will ensure compatibility and reduce the likelihood of encountering untested cases.