OpenXiangShan/NEMU #927: Not All `tcache` Related Function Declarations and Calls Controlled by the Macro `CONFIG_PERF_OPT`

scratch-er, 2025-09-13:
I have noticed that in `src/cpu/tcache.c`, there are two sets of functions. Which set is enabled is controlled by the macro definition `CONFIG_PERF_OPT`. The only place where these functions are involved is `src/cpu/cpu-exec.c`. These functions are related with decoding caching. 

However, not all of the function calls and declarations in `src/cpu/cpu-exec.c` are controlled by the macro definition. This can make the declaration, the function call and the actual definition inconsistent under some configurations. And this can cause compiler errors, especially when building the differential testing dynamic libraries using the configurations under:

https://github.com/OpenXiangShan/NEMU/tree/master/configs

scratch-er, 2025-09-13:
This also makes the CI fail in container `ghcr.io/openxiangshan/xs-env:latest`.

[full_act_log.txt](https://github.com/user-attachments/files/22311245/full_act_log.txt)

CI script:

```yml
  compilation-all:
    runs-on: ubuntu-latest
    container:
      image: ghcr.io/openxiangshan/xs-env:latest
    continue-on-error: false
    name: Compilation - all configs
    timeout-minutes: 60
    steps:
    - name: Checkout repository
      uses: actions/checkout@v4
    - name: Setup env
      run: |
        echo "NEMU_HOME=$GITHUB_WORKSPACE" >> $GITHUB_ENV
    - name: Try to compile
      id: list_configs
      run: |
        bash ./scripts/compilation-test.sh
```

log:

```
[CI/Compilation - all configs]   | src/cpu/cpu-exec.c:848:26: error: passing arguments to 'tcache_handle_flush' without a prototype is deprecated in all versions of C and is not supported in C23 [-Werror,-Wdeprecated-non-prototype]
[CI/Compilation - all configs]   |   848 |       tcache_handle_flush(cpu.pc);
[CI/Compilation - all configs]   |       |                          ^
[CI/Compilation - all configs]   | src/cpu/cpu-exec.c:877:28: error: passing arguments to 'tcache_handle_flush' without a prototype is deprecated in all versions of C and is not supported in C23 [-Werror,-Wdeprecated-non-prototype]
[CI/Compilation - all configs]   |   877 |         tcache_handle_flush(cpu.pc);
[CI/Compilation - all configs]   |       |                            ^
[CI/Compilation - all configs]   | 2 errors generated.
```

cebarobot, 2025-09-13:
Yes, there is two "tcache", one is for stand alone mode with PERF_OPT, one is for ref mode (imported in #908). For the ref mode tcache, there should be a switch option, but there isn't any now.

Actually, I didn't get what the problem is. I find that the recent CI passed the compilation-all test.

poemonsense, 2025-11-22:
I just have time to look into this issue and think #929 can be merged as is.

> Yes, there is two "tcache", one is for stand alone mode with PERF_OPT, one is for ref mode (imported in [#908](https://github.com/OpenXiangShan/NEMU/pull/908)).

This is partially correct. It's actually determined by the macro `CONFIG_PERF_OPT` instead of ref/standalone modes.

When `CONFIG_PERF_OPT` is enabled, the original, complicated trace cache is enabled. Otherwise, there is a simple trace cache introduce by #908.

> Actually, I didn't get what the problem is. I find that the recent CI passed the compilation-all test.

Originally, the `void cpu_exec(uint64_t n)` function in `cpu-exec.c` does use the `CONFIG_PERF_OPT` to control whether to call the `tcache_handle_flush` function. This is correct.

In #908, @xiaokamikami removes the `IFDEF` and simply calls `tcache_handle_flush(cpu.pc);` without considering that this function has two different prototypes when `CONFIG_PERF_OPT` is enabled/disabled. See:

https://github.com/OpenXiangShan/NEMU/pull/908/files#diff-1f9d263325b615223a888631b8e514d3b1b7d6e3c4ea5220d67a29e2acaa9155 (Line 848 and 877).

This is wrong and should be changed as in #929 (calling different functions when `CONFIG_PERF_OPT` is enabled/disabled)

>  For the ref mode tcache, there should be a switch option, but there isn't any now.

I believe we can view this as a future work.